zephyrproject-rtos/west #490: West saves/caches invalid board on build failure, causing future builds to use old board without appropriate error message

drojf, 2021-03-19:
I've noticed that west can get "stuck" where it's saved/cached an invalid board name, and continues to use that invalid name unless you do a pristine build, without giving an appropriate error message.

### Reproduction instructions

1. Build the blinky sample, following the ["Getting Started Guide"](https://docs.zephyrproject.org/latest/getting_started/index.html#build-the-blinky-sample), but type the wrong board name:

```
west build -p auto -b an_invalid_board_name samples/hello_world
```

2. Run the command again with a valid board name

```
west build -p auto -b reel_board samples/hello_world
```

West will then try to build `an_invalid_board_name` again, instead of `reel_board `, and doesn't give an appropriate error message (see below expected behavior)

```
drojf@drojf-virtual:~/zephyrproject/zephyr$ west build -p auto -b reel_board samples/hello_world
WARNING: This looks like a fresh build and BOARD is unknown; so it probably won't work. To fix, use --board=<your-board>.
Note: to silence the above message, run 'west config build.board_warn false'
-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): /home/drojf/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/drojf/zephyrproject/zephyr/samples/hello_world
-- Zephyr version: 2.5.0-rc1 (/home/drojf/zephyrproject/zephyr)
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: an_invalid_board_name
No board named 'an_invalid_board_name' found.

Please choose one of the following boards:
```

### Expected behavior

Normally, if you change boards and don't do a pristine build, you get a nice error message telling you:

`ERROR: Build directory /home/drojf/zephyrproject/zephyr/build targets board reel_board, but board nucleo_g431rb was specified. (Clean the directory, use --pristine, or use --build-dir to specify a different one.)`

I would expect the same message if you previously entered an invalid board name (it should always warn you when you switch boards, regardless of what failures have happened).

marc-hb, 2021-03-19:
west version 0.8.0 is pretty old and I think I remember reporting something similar, fixed by @mbolivar-nordic . Can you please upgrade and re-test?

marc-hb, 2021-03-19:
> I think I remember reporting something similar

That issue was in zephyr `west_commands`, not in west itself. Can you try `cd zephyr && git cherry-pick -x` https://github.com/zephyrproject-rtos/zephyr/commit/af5a79f6c13be3 and retest?

drojf, 2021-03-19:
My apologies, you're right...I had this bug the last time I used zephyr, but I didn't report it. Then yesterday when I got back into it, I updated zephyr, but I forgot that I had switched to a fork to test something, which was way behind.

closed 2021-03-19 by drojf