zephyrproject-rtos/west #492: Unable to run custom west commands before all imported repositories are cloned

mrrosen, 2021-03-23:
If a manifest imports from other projects, no custom west commands can be run from any project until all imported projects have been cloned. For example, if a manifest has the following form:

    projects:
      - name: internal1
        revision: some-branch
      - name: internal2
        revision some-branch

      - name: zephyr
        remote: zephyr
        revision: v2.5.0
        import:
          name-whitelist:
          - proj1
          - proj2
      - name: external1
        remote: external-repo
        revision: v1.1.0
        import:
          name-whitelist:
          - proja
    self:
      west-commands: scripts/west-commands.yml

Custom commands from the `scripts/west-commands.yml` will be unavailable until both `zephyr` and `external1` are cloned. This can be a problem if some of the custom commands do important environment setup or supplement `west update` in some way; forcing user to `west update` before running the custom commands that are required for setting up/updating their environment properly.

Suggestion would be to ignore non-cloned imports from the repository instead of causing them to cancel the additional of all custom commands entirely.

marc-hb, 2021-03-23:
Can you please share an error message or stack trace?

Is this problem really specific to `import`, I mean is it possible to use custom west commands before the very first `west update` when there is no import?

Does this affect absolutely every kind of west command, I mean can you reproduce with a fake west command that does nothing and just prints "hello"?

mrrosen, 2021-03-23:
Ive created a simple example for you to see. It does seem to be tied to import, from what Ive observed:
https://github.com/mrrosen/manifest

```
west init -m https://github.com/mrrosen/manifest.git
=== Initializing in C:\Users\mrrosen\identity\test\simple
--- no --manifest-rev was given; using remote's default branch: refs/heads/main
--- Cloning manifest repository from https://github.com/mrrosen/manifest.git, rev. refs/heads/main
Initialized empty Git repository in C:/Users/mrrosen/identity/test/simple/.west/manifest-tmp/.git/
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 6 (delta 0), reused 6 (delta 0), pack-reused 0
Unpacking objects: 100% (6/6), 1001 bytes | 27.00 KiB/s, done.
From https://github.com/mrrosen/manifest
 * branch            main       -> FETCH_HEAD
 * [new branch]      main       -> origin/main
Note: switching to 'FETCH_HEAD'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 96126e5 simple west test
--- setting manifest.path to manifest
=== Initialized. Now run "west update" inside C:\Users\mrrosen\identity\test\simple.

C:\Users\mrrosen\identity\test\simple>west hello
usage: west [-h] [-z ZEPHYR_BASE] [-v] [-V] <command> ...
west: error: argument <command>: invalid choice: 'hello' (choose from 'init', 'update', 'list', 'manifest', 'diff', 'status', 'forall', 'help', 'config', 'topdir', 'selfupdate')

C:\Users\mrrosen\identity\test\simple>west update
=== updating zephyr (zephyr):
--- zephyr: initializing
Initialized empty Git repository in C:/Users/mrrosen/identity/test/simple/zephyr/.git/
--- zephyr: fetching, need revision v2.5.0
remote: Enumerating objects: 2, done.
remote: Counting objects: 100% (2/2), done.
remote: Total 532644 (delta 1), reused 1 (delta 1), pack-reused 532642 eceiving objects: 100% (532644/532644), 365.25 MiB | 7.26 MiB/s
Receiving objects: 100% (532644/532644), 368.58 MiB | 7.59 MiB/s, done.
Resolving deltas: 100% (403942/403942), done.
From https://github.com/zephyrproject-rtos/zephyr
 * tag               v2.5.0         -> FETCH_HEAD
 * [new tag]         v1.0.0         -> v1.0.0
 * [new tag]         v1.1.0         -> v1.1.0
 * [new tag]         v1.1.0-rc1     -> v1.1.0-rc1
 * [new tag]         v1.10.0        -> v1.10.0
 * [new tag]         v1.10.0-rc1    -> v1.10.0-rc1
 * [new tag]         v1.10.0-rc2    -> v1.10.0-rc2
 * [new tag]         v1.10.0-rc3    -> v1.10.0-rc3
 * [new tag]         v1.11.0        -> v1.11.0
 * [new tag]         v1.11.0-rc1    -> v1.11.0-rc1
 * [new tag]         v1.11.0-rc2    -> v1.11.0-rc2
 * [new tag]         v1.11.0-rc3    -> v1.11.0-rc3
 * [new tag]         v1.12.0        -> v1.12.0
 * [new tag]         v1.12.0-rc1    -> v1.12.0-rc1
 * [new tag]         v1.12.0-rc2    -> v1.12.0-rc2
 * [new tag]         v1.12.0-rc3    -> v1.12.0-rc3
 * [new tag]         v1.13.0        -> v1.13.0
[... 152 lines cut ...]

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 96126e5 simple west test
--- setting manifest.path to manifest
=== Initialized. Now run "west update" inside C:\Users\mrrosen\identity\test\simple.

<<<<<<<<< Here I manually edited manifest/west.yml to remove just lines 18-20, the import of Zephyr

C:\Users\mrrosen\identity\test\simple>west hello
=== Hello!
```

mbolivar-nordic, 2021-03-29:
> Suggestion would be to ignore non-cloned imports from the repository instead of causing them to cancel the additional of all custom commands entirely.

Sorry, but no. That would lead to nondeterministic results in the case where multiple projects try to define extensions with the same name.

closed 2021-03-29 by mbolivar-nordic

marc-hb, 2021-03-29:
@mbolivar-nordic I understand the requirement to resolve all imports to avoid complex situations, strange corner cases and non-predictable behaviours but then shouldn't west communicate this expectation more clearly? This would save "the next @mrrosen" the considerable time it took him to file this very well detailed report. In other words, shouldn't west be aware that the workspace is not functional yet and in a half-initialized state? Emitting regular error messages like the one below  is confusing because it's the exact same message when the workspace is fully initialized and the "hello" command is actually missing:

```
C:\Users\mrrosen\identity\test\simple>west hello
usage: west [-h] [-z ZEPHYR_BASE] [-v] [-V] <command> ...
west: error: argument <command>: invalid choice: 'hello' (choose from 'init', 'update', 'list', 'manifest', 'diff', 'status', 'forall', 'help', 'config', 'topdir', 'selfupdate')
```

This message also lists `diff`, `status` and `forall` as valid commands when they're not really.

mbolivar-nordic, 2021-03-29:
Yes, I agree the current situation is not very good. We should not be relying on ArgumentParser.add_subparsers for this; it leads to usability problems such as these.

I have a half baked branch where I tried to fix this in main.py by not using that method anymore and validating the command directly against what we've loaded from the current manifest. That let us put together a better error message. However, this ended up being more work than I could finish in a weekend and I haven't gotten back to it.

I think we'll finally have time for this once we've got the workspace creation optimizations that are ongoing discussion in #319 resolved. That has to come first and I'll be working on it this week.

mrrosen, 2021-03-29:
> > Suggestion would be to ignore non-cloned imports from the repository instead of causing them to cancel the additional of all custom commands entirely.
> 
> Sorry, but no. That would lead to nondeterministic results in the case where multiple projects try to define extensions with the same name.

Wouldnt the behavior be deterministic, ie, it takes the command from where it can find it in the set of currently cloned projects? While it cant provide such a nice warning that the command is defined elsewhere, it can throw a warning saying that some imported repositories are not cloned and thus could not be imported.

marc-hb, 2021-03-30:
> Wouldnt the behavior be deterministic, ie, it takes the command from where it can find it in the set of currently cloned projects?

I think "currently cloned projects" basically means "non-deterministic". "Combinatorial explosion" also comes to mind.

Version control is all about this: "if you and I have the same versions of the source then we observe the same, 'deterministic' behaviors". Now the reality is of course not that rosy, in practice it's more like: "if you and I have the same versions of the source _and_ the same configuration _and_ the same toolchain _and_ no sneaky script is hidden in CI etc."  but do we really want to add yet another variable to the configuration mix, namely what repos are present versus not? I hope not. I hope the future is more like this:

- either everything is initialized and everyone reproduces the same bugs, same errors and same failures,
- or initialization is not complete and all bets are off. Every man for himself.

BTW that's one reason why I was wary of "group-filter inheritance", see #481 and interesting new #491

mrrosen, 2021-03-30:
I see what you mean; its just that in my case since we are replacing `west update` with a supplemented version via a custom command, its more likely people trying to clone our repo will have an incorrect environment if they just do a regular `west update` or will try our custom command without doing `west update` after some import has changed and be confused as to why it isnt working. But I guess it is specific to our project and not general enough to be considered for the public release (while we dont really have the issue of handling multiple custom commands of the same name as other projects might). Just wanted to bring it up.

marc-hb, 2021-03-30:
Thanks @mrrosen for the additional info. Have you considered leaving `west update` untouched and moving your custom steps to the `west build` "phase" instead? Either combining them with `west build` somehow or making `west build` run a sanity check that prints the exact custom command(s) to run when they haven't been run yet.

> That lets us put together a better error message. However, this ended up being more work than I could finish in a weekend and I haven't gotten back to it.

@mbolivar-nordic should this issue be re-opened and re-purposed to track this future, better error message for this situation?

mrrosen, 2021-03-30:
> Thanks @mrrosen for the additional info. Have you considered leaving `west update` untouched and moving your custom steps to the `west build` "phase" instead? Either combining them with `west build` [...]

While thats an interesting idea, the big thing we are replacing `west update` for to apply some patches to some repos after updating them since we rather get some of the larger repositories (like Zephyr) from the public git instead of maintaining our own clone (similar to what crosstools does, applying patch files to gcc). Combining this with `west update` is the most natural thing to do (update and checkout repositories at fixed revision, then apply patches onto them).

marc-hb, 2021-03-30:
> the big thing we are replacing west update for to apply some patches to some repos after updating them since we rather get some of the larger repositories (like Zephyr) from the public git instead of maintaining our own clone (similar to what crosstools does, applying patch files to gcc).

Thanks, this clarifies.

First, I think you should really maintain your own clones and rebase them regularly but I understand automating a rebase can be scary so let's leave that aside.

Now sticking to your dynamic patching approach there is a  tool much better than `patch` to implement _the exact same approach_: `git merge`! In other words store your patches into a git branch somewhere; you do NOT need to rebase this branch regularly. Then perform a one-line `git merge --no-edit my-custom-patches` at build time.

Unlike `patch`, `git merge` is smart enough to realize something has already been merged and do nothing if you run it multiple times (idempotence). So you can make it part of `west build` and run it as many time as you want. I agree making "dynamic patching" part of `west update` feels much more logical, on the other hand making `git merge` part of `west build` Just Works. This is exactly what github continuously does for every Pull Request BTW, find the result in every `pull/12345/merge`

Other huge benefits of `git merge` over `patch`:
- no massive pollution of `git status` and `git describe ---dirty`
- proper version control with ability to reproduce builds.
- no need to edit `.patch` files any more (the horror)
- ability to breakdown the custom patches into multiple commits, no "per-file" restriction.
- etc.

> (similar to what crosstools does, applying patch files to gcc)

It's not because many people are still stuck with obsolete `.patch` files that they are a good idea :-)