zephyrproject-rtos/west #501: New release with narrow update feature

temap, 2021-04-28:
HI, could you tell me when the new version will be released with the narrow update feature? 
We need a narrow update for west tool because we have ~30 minutes clones which is annoying.

Thanks

marc-hb, 2021-04-29:
I don't know the answer but I never experienced any problem running the latest commit:
```shell
pip3 uninstall west
git clone https://github.com/zephyrproject-rtos/west
cd west
pip3 install -e .
```
"Don't do this at home!" :-)

temap, 2021-04-29:
@marc-hb, thanks for the solution, but I would like to have it on the official release. Could you explain about release cycles. When to expect a new west version?

mbolivar-nordic, 2021-04-29:
Hi. I'm starting the release process tomorrow. It usually only takes a day or so. I therefore expect it no late than sometime early next week.

closed 2021-04-29 by mbolivar-nordic

temap, 2021-04-29:
@mbolivar-nordic thank you