zephyrproject-rtos/west #510: Add support for msys2

Staudi14, 2021-06-26:
I would propose to support the msys2 msys shell.
It is already possible to use west in the git bash (which is the msys2 msys shell with the windows environment imported - as far as I know) and therefore my ignorant self does not find any reason why it should not be supported.

At the current date when trying to install west following error message is outputted:
```
$ west --version
Traceback (most recent call last):
  File "/home/patri/.local/bin/west", line 8, in <module>
    sys.exit(main())
  File "/home/patri/.local/lib/python3.9/site-packages/west/app/main.py", line 779, in main
    app.run(argv or sys.argv[1:])
  File "/home/patri/.local/lib/python3.9/site-packages/west/app/main.py", line 95, in run
    config.read_config(topdir=self.topdir)
  File "/home/patri/.local/lib/python3.9/site-packages/west/configuration.py", line 102, in read_config
    config.read(_gather_configs(configfile, topdir), encoding='utf-8')
  File "/home/patri/.local/lib/python3.9/site-packages/west/configuration.py", line 267, in _gather_configs
    ret.append(_location(ConfigFile.SYSTEM, topdir=topdir))
  File "/home/patri/.local/lib/python3.9/site-packages/west/configuration.py", line 242, in _location
    raise ValueError('unsupported platform ' + plat)
ValueError: unsupported platform MSYS_NT-10.0-19043
```

Following packages were installed on msys2:
```
$ pacman -Qe
asciidoc 9.1.0-2
autoconf 2.71-1
autoconf2.13 2.13-3
autogen 5.18.16-1
automake-wrapper 11-1
automake1.10 1.10.3-4
automake1.11 1.11.6-4
automake1.12 1.12.6-4
automake1.13 1.13.4-5
automake1.14 1.14.1-4
automake1.15 1.15.1-2
automake1.16 1.16.3-1
automake1.6 1.6.3-3
automake1.7 1.7.9-3
automake1.8 1.8.5-4
automake1.9 1.9.6-3
base 2020.12-1
bison 3.7.6-1
btyacc 20210619-1
ccache 4.2-1
cmake 3.20.4-1
diffstat 1.64-1
diffutils 3.7-1
dos2unix 7.4.2-1
dtc 1.6.0-4
flex 2.6.4-1
gdb 10.2-3
gettext-devel 0.19.8.1-1
git 2.32.0-1
gperf 3.1-2
groff 1.22.4-1
help2man 1.48.3-1
intltool 0.51.0-2
libtool 2.4.6-11
libunrar 6.0.7-1
libunrar-devel 6.0.7-1
m4 1.4.19-1
make 4.3-1
man-db 2.9.4-1
mingw-w64-x86_64-dfu-util 0.10-4
ncurses-devel 6.2-1
ninja 1.10.2-1
patch 2.7.6-1
patchutils 0.4.2-1
pkgfile 21-1
python 3.9.5-1
python-pip 21.1.1-1
quilt 0.66-2
reflex 20210510-1
scons 3.1.2-6
swig 4.0.2-1
texinfo 6.7-3
texinfo-tex 6.7-3
ttyrec 1.0.8-2
unrar 6.0.7-1
xmlto 0.0.28-2

```

Staudi14, 2021-06-26:
Not Possible as dependencies neither have support for msys2

closed 2021-06-26 by Staudi14

mbolivar-nordic, 2021-06-28:
> Not Possible as dependencies don't support msys2

Can you clarify which ones aren't supported? Is it colorama?

thewh1teagle, 2024-01-06:
Try with
```console
SETUPTOOLS_USE_DISTUTILS=stdlib pip install west
```