zephyrproject-rtos/west #512: West init error "status 128: git fetch origin --tags -- master 'refs/heads/*:refs/remotes/origin/*"

nqduy35, 2021-07-21:
Hello, I use the following link to install zephyrproject on my Ubuntu machine.  
https://docs.zephyrproject.org/latest/getting_started/index.html#install-dependencies

The installation is ok until the command _$west init ~/zephyrproject_.  Here is the error: 

fatal: Couldn't find remote ref master
FATAL ERROR: command exited with status 128: git fetch origin --tags -- master 'refs/heads/*:refs/remotes/origin/*

Several months ago, I used the same command to install west and zephyrproject and everything was ok.

I check the repo of west and I found no "master" branch but only the branch "main".  I don't know if your development team make some modifications that causes this error? 

Thank you in advance for your answer.

marc-hb, 2021-07-21:
Which west version is this?

I suspect you need a version with at least #485

nqduy35, 2021-07-21:
I use the command _$pip3 install --user -U west_ to get the West. The version is **v0.11.0** as the result using the command _$west --version_.

mbolivar-nordic, 2021-07-21:
Hi, I can't reproduce this.

```
$ west --version
West version: v0.11.0
```

```
$ west init ~/zephyrproject
=== Initializing in /home/mbolivar/zephyrproject
--- no --manifest-rev was given; using remote's default branch: refs/heads/main
--- Cloning manifest repository from https://github.com/zephyrproject-rtos/zephyr, rev. refs/heads/main
[...]
```

I think something is wrong with your environment. Maybe try uninstalling west and reinstalling it?

nqduy35, 2021-07-21:
Finally, I success with the command : _$ west init --mr main_. I have tried to install and uninstall several times in two different machines (one debian and one ubuntu).  I will close the thread. Thank you all.

closed 2021-07-21 by nqduy35