zephyrproject-rtos/west #527: west update -jN fails on windows

mbolivar-nordic, 2021-08-05:
With west [v0.12.0a2](https://github.com/zephyrproject-rtos/west/commit/06b063556418312bb447a56ccf8a81b131d004f6), on my windows 10 box running python 3.9.0:

```
C:\Users\Marti\zephyrproject>west update -j5
multiprocessing.pool.RemoteTraceback:
"""
Traceback (most recent call last):
  File "c:\users\marti\v\west-0.12\lib\site-packages\west\app\project.py", line 874, in _update_concurrent_thread
    self.update(project)
  File "c:\users\marti\v\west-0.12\lib\site-packages\west\app\project.py", line 1132, in update
    log.banner(f'updating {project.name_and_path}:')
  File "c:\users\marti\v\west-0.12\lib\site-packages\west\log.py", line 91, in banner
    inf('===', *args, colorize=True)
  File "c:\users\marti\v\west-0.12\lib\site-packages\west\log.py", line 82, in inf
    print(*args)
OSError: [WinError 6] The handle is invalid

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Marti\AppData\Local\Programs\Python\Python39-32\lib\multiprocessing\pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "c:\users\marti\v\west-0.12\lib\site-packages\west\app\project.py", line 880, in _update_concurrent_thread
    sys.stdout.flush()
OSError: [WinError 6] The handle is invalid
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\Marti\AppData\Local\Programs\Python\Python39-32\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Marti\AppData\Local\Programs\Python\Python39-32\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Marti\v\west-0.12\Scripts\west.exe\__main__.py", line 7, in <module>
  File "c:\users\marti\v\west-0.12\lib\site-packages\west\app\main.py", line 779, in main
    app.run(argv or sys.argv[1:])
  File "c:\users\marti\v\west-0.12\lib\site-packages\west\app\main.py", line 106, in run
    self.run_command(argv)
  File "c:\users\marti\v\west-0.12\lib\site-packages\west\app\main.py", line 336, in run_command
    cmd.run(args, unknown, self.topdir, manifest=self.manifest)
  File "c:\users\marti\v\west-0.12\lib\site-packages\west\commands.py", line 140, in run
    self.do_run(args, unknown)
  File "c:\users\marti\v\west-0.12\lib\site-packages\west\app\project.py", line 814, in do_run
    self.update_all()
  File "c:\users\marti\v\west-0.12\lib\site-packages\west\app\project.py", line 980, in update_all
    (failed, updated) = self.update_project_list(projects)
  File "c:\users\marti\v\west-0.12\lib\site-packages\west\app\project.py", line 938, in update_project_list
    failed, updated = self.update_project_list_parallel(projects)
  File "c:\users\marti\v\west-0.12\lib\site-packages\west\app\project.py", line 909, in update_project_list_parallel
    for index, successful in enumerate(
  File "C:\Users\Marti\AppData\Local\Programs\Python\Python39-32\lib\multiprocessing\pool.py", line 870, in next
    raise value
OSError: [WinError 6] The handle is invalid
```

mbolivar-nordic, 2021-08-05:
@M1cha FYI. I'm opening another bug as well that will be a release blocker for 0.12. If it's an easy fix, please send a PR. I'll be looking at this in parallel.

M1cha, 2021-08-05:
I'll not be able to work on anything until at least Monday. Please keep that in mind for your release schedule

mbolivar-nordic, 2021-08-05:
> I'll not be able to work on anything until at least Monday. Please keep that in mind for your release schedule

Ack, thanks for letting me konw.

M1cha, 2021-08-11:
why does this not happen in the unit tests?
the exception sounds like stdout is not setup properly.

mbolivar-nordic, 2021-08-11:
> why does this not happen in the unit tests?

I don't know for sure but I would guess because sys.stdout is something different in CI.

> the exception sounds like stdout is not setup properly.

Yeah, I agree something in stdout looks fishy. It seems like we can't print from the other process. Do you have a Windows environment to repro on?

M1cha, 2021-08-11:
Yes I have a VM and will try to repro and debug it there.

marc-hb, 2021-10-14:
Huge revert of the whole feature submitted in #549

closed 2021-10-15 by mbolivar-nordic