zephyrproject-rtos/west #528: west update -jN is interleaving output without --unbuffered

mbolivar-nordic, 2021-08-05:
This is with west 0.12.0a2 as I just built and uploaded to PyPI.

On my Arch Linux box running Python 3.9.6, with similar results on my macOS box running python 3.8.2:

```
$ west update -j5
=== updating canopennode (modules/lib/canopennode):
HEAD is now at f167efe zephyr: move build configuration into zephyr repository
=== updating civetweb (modules/lib/civetweb):
=== updating cmsis (modules/hal/cmsis):
HEAD is now at c3bd209 DSP: fix compiler warning on inline attribute
=== updating edtt (tools/edtt):
HEAD is now at 7dd56fc Add FIXME note to handle incorrect LL/CON/MAS/BV-27-C implementation
=== updating fatfs (modules/fs/fatfs):
HEAD is now at 1d1fcc7 fat_fs: Expose _VOLUMES to Kconfig
=== updating hal_altera (modules/hal/altera):
HEAD is now at 23c1c1d hal_altera: Conditionally include the Altera HAL
=== updating hal_atmel (modules/hal/atmel):
=== updating hal_cypress (modules/hal/cypress):
HEAD is now at 81a059f Merge pull request #1 from nandojve/pdl_scb_spi_i2c
=== updating hal_espressif (modules/hal/espressif):
=== updating hal_infineon (modules/hal/infineon):
HEAD is now at f1fa824 hal: infineon: initial commit of XMCLib
=== updating hal_microchip (modules/hal/microchip):
HEAD is now at 870d05e modules: microchip: MEC15xx Add missing eSPI VW definitions
=== updating hal_nordic (modules/hal/nordic):
=== updating hal_nuvoton (modules/hal/nuvoton):
HEAD is now at b4d31f3 hal: nuvoton: add license to numicro m480 series hal
=== updating hal_nxp (modules/hal/nxp):
HEAD is now at bb97ba1 watchdog: Enable build of watchdog driver for imxrt6xx
=== updating hal_openisa (modules/hal/openisa):
HEAD is now at 40d049f openisa: Make openisa hal a zephyr_library
=== updating hal_quicklogic (modules/hal/quicklogic):
HEAD is now at b3a66fe Remove references to FPGA IP core drivers
=== updating hal_silabs (modules/hal/silabs):
=== updating hal_st (modules/hal/st):
HEAD is now at 575de9d sensor/stmemsc: Align stmemsc i/f to v2.00
=== updating hal_stm32 (modules/hal/stm32):
HEAD is now at 4200321e dts/st/u5: Generate stm32u5*-pinctrl.dtsi u5 addition
=== updating hal_telink (modules/hal/telink):
HEAD is now at ffcfd62 hal: telink: initial commit
=== updating hal_ti (modules/hal/ti):
=== updating hal_xtensa (modules/hal/xtensa):
HEAD is now at 6e1cf3c zephyr: add SoC overlay for NXP i.MX
=== updating libmetal (modules/hal/libmetal):
HEAD is now at 39d049d lib: update libmetal to release v2021.04.0 + VERSION path fix
=== updating littlefs (modules/fs/littlefs):
HEAD is now at 9e4498d lfs_util: fix log declaration when logging is disabled
=== updating loramac-node (modules/lib/loramac-node):
HEAD is now at 12019623 zephyr: radio: sx127x: override buffer read/write functions
=== updating lvgl (modules/lib/gui/lvgl):
=== updating lz4 (modules/lib/lz4):
HEAD is now at 8e303c2 zephyr: make this repo a zephyr module
=== updating mbedtls (modules/crypto/mbedtls):
=== updating mcuboot (bootloader/mcuboot):
HEAD is now at 62b78ff boot: zephyr: add integration platforms to the allow list
=== updating mcumgr (modules/lib/mcumgr):
HEAD is now at a15a953 snapshot: Update mcumgr to commit 4fa869142f from upstream
=== updating mipi-sys-t (modules/debug/mipi-sys-t):
HEAD is now at 75e6715 zephyr: move platform.h to zephyr tree
=== updating nanopb (modules/lib/nanopb):
HEAD is now at d148bd2 Add zephyr module support
=== updating net-tools (tools/net-tools):
=== updating nrf_hw_models (modules/bsim_hw_models/nrf_hw_models):
HEAD is now at a47e326 HW_models: NRF_RTC: Fix compare event missed when interrupt disabled
=== updating open-amp (modules/lib/open-amp):
HEAD is now at 6010f05 lib: update open-amp lib to release v2021.04.0
=== updating openthread (modules/lib/openthread):
HEAD is now at 3bcb1e8e5 Merge pull request #44 from canisLupus1313/zephyr
=== updating segger (modules/debug/segger):
[... 80 lines cut ...]
=== updating nanopb (modules/lib/nanopb):
HEAD is now at d148bd2 Add zephyr module support
=== updating net-tools (tools/net-tools):
HEAD is now at f49bd13 net-capture: Listen captured traffic and save to pcap file
=== updating nrf_hw_models (modules/bsim_hw_models/nrf_hw_models):
HEAD is now at a47e326 HW_models: NRF_RTC: Fix compare event missed when interrupt disabled
=== updating open-amp (modules/lib/open-amp):
HEAD is now at 6010f05 lib: update open-amp lib to release v2021.04.0
=== updating openthread (modules/lib/openthread):
HEAD is now at 3bcb1e8e5 Merge pull request #44 from canisLupus1313/zephyr
=== updating segger (modules/debug/segger):
HEAD is now at 3a52ab2 config: Use zephyr provided options
=== updating sof (modules/audio/sof):
HEAD is now at 779f28ed4 Merge pull request #11 from nashif/kconfig_main_menu
=== updating tensorflow (modules/lib/tensorflow):
HEAD is now at dc70a45a7cc tensorflow: add tensorflow 2.4.1 as a zephyr module
=== updating tinycbor (modules/lib/tinycbor):
HEAD is now at 40daca9 zephyr: Remove TINYCBOR from interface libraries
=== updating tinycrypt (modules/crypto/tinycrypt):
HEAD is now at 3e9a49d cmake: Fix conditional in root CMakeLists.txt
=== updating TraceRecorderSource (modules/debug/TraceRecorder):
HEAD is now at 5b5f8d7 Merge pull request #5 from nashif/update_003
=== updating trusted-firmware-m (modules/tee/tfm):
HEAD is now at 1ab3b62d trusted-firmware-m: platform: lpcxpresso55s69: check-out HAL dependencies
```

mbolivar-nordic, 2021-08-05:
@M1cha similar comments apply here as to #527 -- if it's an easy fix, please send a PR. This will block v0.12.

M1cha, 2021-08-11:
@mbolivar-nordic it's not interleaving, part of the log is simply missing.
Adding `logfile.flush()` and `os.fdatasync(logfile.fileno())` didn't help

Also, my attempts to debug this make it seem like `imap_unordered` returns before flushing which makes me think that I can't trust print timing.

commit 040dd57a (2021-08-11): project: print logs from the correct job

`enumerate` on `imap_unordered` does (obviously) not return the index of the
element that finished processing.
That resulted in printing the wrong log file which might not even be finished
and thus swallowing future logs of that job.

Instead, let each job return it's index.

Fixes: #528
closed 2021-08-11 by mbolivar-nordic