zephyrproject-rtos/west #531: `west init -m <repo> --mr <hash>` fails on version 0.11.1

mmajchrzycki, 2021-08-10:
I just found that newest west from `pip` (`0.11.1`) is failing on
```
west init -m https://github.com/nrfconnect/sdk-nrf --mr 7a076c22df511ae6758e30bc69e47fcd78be14a3
=== Initializing in /home/mmajchrzycki/ncs-test
--- Cloning manifest repository from https://github.com/nrfconnect/sdk-nrf, rev. 7a076c22df511ae6758e30bc69e47fcd78be14a3
Cloning into '/home/mmajchrzycki/ncs-test/.west/manifest-tmp'...
fatal: Remote branch 7a076c22df511ae6758e30bc69e47fcd78be14a3 not found in upstream origin
FATAL ERROR: command exited with status 128: git clone --branch 7a076c22df511ae6758e30bc69e47fcd78be14a3 https://github.com/nrfconnect/sdk-nrf /home/mmajchrzycki/ncs-test/.west/manifest-tm
```
The same command works fine on `0.11.0`

janjongboom, 2021-08-10:
Also, 0.11.1 is in pip, but is not actually tagged in this repository.

mbolivar-nordic, 2021-08-10:
Please refer to the release notes:

https://docs.zephyrproject.org/latest/guides/west/release-notes.html#v0-11-1

>  The fix for issue #522 mentioned above introduces a new restriction. The west init --manifest-rev option value, if given, must now be either a branch or a tag. In particular, “pseudo-branches” like GitHub’s pull/1234/head references which could previously be used to fetch a pull request can no longer be passed to --manifest-rev. Users must now fetch and check out such revisions manually after running west init.

mbolivar-nordic, 2021-08-10:
> Also, 0.11.1 is in pip, but is not actually tagged in this repository.

Sorry, I had forgotten to push. This is up now. I'm also using the github "releases" feature to mark this as the latest release. This is the first time west has used this feature, so I may have done something unexpected here. Please LMK if you notice anything else. Thanks!

closed 2021-08-10 by mbolivar-nordic