zephyrproject-rtos/west #538: west build -b qemu_x86 not building (unrecognized command-line option '-m32'; did you mean '-mbe32')

maksimdrachov, 2021-09-13:
Hi,

I'm trying to replicate [this](https://docs.zephyrproject.org/2.3.0/boards/x86/qemu_x86/doc/index.html).

However when I try to build, I'm getting an error:

```
arm-none-eabi-gcc: error: unrecognized command-line option '-m32'; did you mean '-mbe32'?
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /Users/maksim/zephyrproject/zephyr/samples/synchronization/build
```

The entire output:
[image]

I've tried to do `west update` and pulled the last commit from zephyr, this didn't help.

maksimdrachov, 2021-09-13:
After looking into this a bit more, I realise this is probably because the toolchain I'm using only supports ARM.

So then the question is, how to compile qemu on MacOS?

marc-hb, 2021-09-13:
@maksimdrachov this really does not like a west question. Try `west -v ...` then run directly without west the same `cmake` command that west runs and you should see the exact same error.

Try https://github.com/zephyrproject-rtos/zephyr/discussions or even better the new Discord channels.

closed 2021-09-13 by maksimdrachov