zephyrproject-rtos/west #539: west does not follow a moving tag

dkalowsk, 2021-09-16:
As requested on Discord.

Our local build system generates and updates a git tag on some regularly cadence, for example a "nightly" tag that moves forward when the daily CI runs pass.  Under previous versions of west (I believe 0.10.1 was what I was at), a `west update` would correctly follow the tag, and pull down the now latest version of that repository.

With west v 0.11.1 this pattern no longer works.  It appears that west determines it already has the tag and therefore does not need to run git fetch on it.

This can be worked around by running `west update -f always` (which can be made default with a `west config update.fetch always`} but that appears to pull down a lot more detail than needed.  

Example west.yml:
```
    - name: our_module
      remote: github
      repo-path: some/path/our_module
      revision: NIGHTLY
```

mbolivar-nordic, 2021-09-16:
I'm guessing this is a lightweight tag. If so, west used to treat those as commits, which would make this work, but that was due to a bug which was fixed in https://github.com/zephyrproject-rtos/west/commit/121dfd8ac41a333c0127c6f29c299e1bbe05240b. So I would expect -falways to be required for all sorts of tags now.

marc-hb, 2021-09-16:
>  for example a "nightly" tag that moves forward when the daily CI runs pass.

Why use a tag instead of a branch? The only difference between a (light weight) tag and a branch and the latter moves while the former does not.

https://git-scm.com/book/en/v2/Git-Basics-Tagging
> A lightweight tag is very much like a branch that **does not change** — it’s just a pointer to a specific commit.

https://git-scm.com/docs/git-tag#_on_re_tagging

> 1. The sane thing. Just admit you screwed up, and use a different name.
> 2. The INSANE thing. You really want to call the new version "X" too, even though others have already seen the old one.

dkalowsk, 2021-09-16:
@marc-hb diving into a debate about how to best use git is likely beyond the scope here.  I don't disagree with your general thought, but the system I have to work with is already established and not eligible for a change at this time.  The key point is that a behavior that did exist, worked "correctly" before, is likely used and depended upon by someone else (Hyrum's law, some xkcd, etc etc).  The release notes in west don't appear to highlight this behavioral change, which means it will come as a surprise to those using it.

@mbolivar-nordic thanks for the pointer to the commit.  Why was a lightweight tag being treated as a commit considered a bug?  Was there some larger issue attached to it beyond the `cat-file` output?

marc-hb, 2021-09-16:
> diving into a debate about how to best use git is likely beyond the scope here. [...]

West is for the most part a git wrapper, in other words west is for the most part "using git" and it has no choice but to make some assumptions on how users _use git concurrently with west_. So yes, west is all about git usage.

> west worked "correctly" before, is likely used and depended upon by someone else

As west evolves, the way it uses git necessarily evolves too and may unfortunately start assuming that the immense majority doesn't _use git concurrently with west_ in ways described by the official git documentation as "insane". That's quite a low bar.

As an example, here's another, concurrent and possibly conflicting use of git I ran into recently: deleting the clone/module initialized by west and replacing it with another one with other branches (for a very practical and real-world use case). I was amazed to discover that `west` can "recover" from this without any issue. However this is absolutely not something documented so I'm merely enjoying while it lasts and I'm not counting on this to be still available in future releases.

git is not a tool, it's a toolkit that provides an infinite amount of flexibility, complexity and rope to hang yourself with so it's simply impossible to apply Hyrum's law here. See #344 for some other counter-examples.

>  but the system I have to work with is already established and not eligible for a change at this time.

I obviously don't know your system but using a branch instead of a tag looks like a very small change since the only difference is that one moves and the other does not. A smaller change than upgrading west?

mbolivar-nordic, 2021-09-16:
> . Why was a lightweight tag being treated as a commit considered a bug? Was there some larger issue attached to it beyond the `cat-file` output?

I would say @marc-hb already answered this question:

> A lightweight tag is very much like a branch that does not change — it’s just a pointer to a specific commit.

The goal has always been to skip fetches for all tags that are already present on the system, since tags are not expected to change. 

>  The key point is that a behavior that did exist, worked "correctly" before, is likely used and depended upon by someone else (Hyrum's law, some xkcd, etc etc).

I know "correctly" is in quotes, but to be clear, it worked incorrectly before (without any quotes). I only noticed the bug because my own company's manifest started pointing at an upstream project which used lightweight tags for releases instead of annotated tags. That caused unnecessary fetches, which I fixed with that commit.

> The release notes in west don't appear to highlight this behavioral change, which means it will come as a surprise to those using it.

Hmm, you had me worried for a second that I forgot to write this in the release notes, but they do highlight the change. Right in the "bug fixes" section, it says:

"west update --fetch=smart (smart is the default) now correctly skips fetches for project revisions which are lightweight tags (it already worked correctly for annotated tags; only lightweight tags were unnecessarily fetched)."

See: https://docs.zephyrproject.org/latest/guides/west/release-notes.html#v0-11-1

mbolivar-nordic, 2021-09-16:
> some xkcd

Did you mean https://xkcd.com/1172/?

Given that this is documented in the release notes (see above), is the expected behavior (it's even documented right there in the `west update -h` output: `"smart" (default) skips fetching projects whose revisions are SHAs or tags available locally`), and there is a way to get the old behavior (`-f=always`), I'm not going to treat this issue as a bug.

Are you asking for an option to "reenable spacebar heating"?

closed 2021-09-16 by mbolivar-nordic

mbolivar-nordic, 2021-09-16:
> Are you asking for an option to "reenable spacebar heating"?

Though I decided to close this issue since it's not a bug as @carlescufi suggested it might have been on Discord, I did go ahead and add a new feature which should handle this use case:

https://github.com/zephyrproject-rtos/west/pull/541

However, it makes no distinction between lightweight and annotated tags.

dkalowsk, 2021-09-16:
That is the correct xkcd I was thinking of.  And spacebar heating is extremely underrated.

Thank you for the answers and closing the issue.  I was going to close it as well given the direction of answers provided.

marc-hb, 2021-09-16:
> This can be worked around by running `west update -f always` (which can be made default with a west config update.fetch always} but that appears to pull down a lot more detail than needed.

Could you try `west update --fetch=always --narrow` instead and let us know whether it works  too? I think it should and it could be much faster depending on your repos.

See also brand new #541

dkalowsk, 2021-09-17:
Adding the `--narrow` flag restores a behavior similar to pre-v0.11.1 update.  Thanks for that suggestion as I wouldn't have thought to use it.