zephyrproject-rtos/west #547: "--group-filter -backend" doesn't work

icy, 2021-10-08:
I want to exclude some group when run `west update`, and it seems the exclusion option doesn't work as the minus was considered as another (unknown?) option:

```
$ west update --group-filter -backend
usage: west update [-h] [--stats] [--name-cache NAME_CACHE] [--path-cache PATH_CACHE] [-f {always,smart}] [-o FETCH_OPT] [-n] [-k] [-r] [--group-filter FILTER] [-x] [PROJECT ...]
west update: error: argument --group-filter/--gf: expected one argument
```

icy, 2021-10-08:
The work-around is to use non-space version in arguments, i.e, `--gf=-backend` or `--group-filter=-backend`, but that is still a problem with our argument parsing. What is our expected behavior here? I can help to submit a patch once it's clear. 

btw, I'm using `west` for my own projects. I hope `west` is my last tool after many tries [1]  ;)

[1] https://github.com/icy/git_xy#meta-repository

mbolivar-nordic, 2021-10-08:
Right. This is an argparse thing, not a west issue. The right way to pass option values which begin with `-` is to use `=` as a separator instead of whitespace. There's nothing for west to do here. We can't rewrite the argument parsing to use something other than the standard `argparse` library because that's the interface that extensions use to register their arguments.

I don't see it as a problem but just a fact of life when using argparse, so I'm going to close this issue. Thanks for taking the time to file it and I hope the discussion is helpful to future readers.

closed 2021-10-08 by mbolivar-nordic