zephyrproject-rtos/west #548: [feature] allow west to report if any of the cloned repositories is out of sync with the manifest

xnlcasad, 2021-10-08:
In order to ensure that a Zephyr application is always built from the exact same sources it would be good if west could report if any of the west-handled repositories is out of sync with the manifest file. Much like a git superproject `git status` would report a dirty working tree if any of the submodules' HEADs point to a sha different from the one recorded in the superproject.

```
$ west forall -c 'git status'
=== running "git status" in manifest (nrf):
HEAD detached at v1.7.0
nothing to commit, working tree clean
=== running "git status" in zephyr (zephyr):
HEAD detached at v2.6.99-ncs1
$ cd -
~/work/zephyr-upstream/zephyr
$ git checkout HEAD^
$ cd -
~/work/zephyr-upstream/nrf
$ west forall -c 'git status'
=== running "git status" in manifest (nrf):
HEAD detached at v1.7.0
nothing to commit, working tree clean
=== running "git status" in zephyr (zephyr):
HEAD detached at 6bfc48ef0b
```
So basically I rolled back zephyr one commit, but west would not tell me that zephyr's HEAD is pointing to a different commit than whatever is on nrf/west.yml, which is:
```
- name: zephyr
        repo-path: sdk-zephyr
        revision: v2.6.99-ncs1
```

@carlescufi suggested that a new parameter to `west list` could be used. I don't have anything against that, I would only suggest the interface to be automation friendly, so that it could be invoked from a build script. If I take git as an analogy, `git diff --exit-code` would return 1 if anything has been changed from the working tree versus the HEAD revision.

@henrikbrixandersen @mbolivar

mbolivar-nordic, 2021-10-08:
`west status`, as a wrapper around `git status`, reports status relative to `HEAD`, not `manifest-rev`. So its output is not related to the manifest file, unless of course if you've just run `west update` and `HEAD` is the same commit as `manifest-rev`.

>  it would be good if west could report if any of the west-handled repositories is out of sync with the manifest file.

We get requests like this from time to time. The devil is in the details, as usual. For a manifest that does not use `import:` that only uses SHAs in its `revision:` values, like upstream zephyr/west.yml, the request seems clear. But it's less obvious what the request is when the manifest `import:` and/or non-SHA `revision:` values, especially when it contains both.

The issue is that `revision:` is really a revision on the *remote* project, not the local tree. And since branches are allowed, the remote commit can change without west knowing about it. And then when you mix imports in with non-SHAs, you can actually end up with a final resolved manifest with a totally different project list, because the remote got some new projects added in the manifest you're importing from.

The best I've come up with in such situations is to compare the working tree (or the HEAD commit) with `manifest-rev`.

But what is in `manifest-rev` for each project is **not** the same thing as "the manifest file" for reasons explained above.

Given this, can you refine your proposal a bit? Is comparing with `manifest-rev` OK?

0Grit, 2021-10-10:
The route I'm going to end up going:

* force west to manage all the dependencies of my applications to be imported into a directory somewhere within the root directory of my applications git repository

* create a plugin for west which manages a standard-ish file containing a signature for all non ignored files within the top level directory

xnlcasad, 2021-10-11:
>Given this, can you refine your proposal a bit? Is comparing with manifest-rev OK?

Even after having read https://docs.zephyrproject.org/latest/guides/west/workspaces.html#the-manifest-rev-branch I have to acknowledge that I don't completely understand how is this branch used by west, so please bear with me if I somehow state/ask what might be obvious to others.

What I can try to do is to expose our use case, so that it might trigger a possible discussion/solution.

We'd need to have many freestanding applications, with most of them using the exact same version of Zephyr and NCS, we'd use the freestanding model to avoid having n-copies of the exact same NCS on disk, and to reduce cloning time.

We need to make sure that when any developer builds a final image of the (app + NCS bundle) we can programatically verify that the sources used were exactly the expected ones, so that the final image can be bit by bit reproducible in any subsequent build.

We would expect the build system to inform the developer if something is being built that is not exactly what was originally committed to git. This is also valid when going back in history (or bisecting, etc)

On https://github.com/nrfconnect/sdk-nrf/pull/5797#issuecomment-937595909 @tejlmand wrote a proposal to use cmake's find_package( EXACT ) to achieve this purpose, but in our case this would not go all the way, essentially because package versions are manually updated. 

Maybe an analogy with git submodules could better explain what I would ideally expect: If any of the superprojects' submodules is not at the expected revision, or has modified files in the working tree, then running git diff --exit-code will let me know I'm not building with a clean tree.

carlescufi, 2021-10-11:
> We get requests like this from time to time. The devil is in the details, as usual. For a manifest that does not use `import:` that only uses SHAs in its `revision:` values, like upstream zephyr/west.yml, the request seems clear. But it's less obvious what the request is when the manifest contains `import:` and/or non-SHA `revision:` values, especially when it contains both.

Here's a suggestion from my side:

How about we add a new `status` formatting option in `west list` that is able to produce the following output:
- `dirty`: if the tree is dirty
- `synced`: if the revision matches the `manifest-rev` **and** the revision is not a branch
- `unsynced`: if the revision does not match the `manifest-rev`
- `unknown`: if the revision is a branch and not a tag or a SHA

0Grit, 2021-10-12:
Going out on a limb here, but I've always felt git should be doing more to solve this problem.
Why git modules are based on non human readable entries into the index versus human readable configuration files tracked like .gitignore I'll never understand.

I assume we are all under the assumption that getting git to change is not a better path?

0Grit, 2021-10-12:
I have not taken a look at the development community surrounding git but I'm always down to help make a case if one can be made.

marc-hb, 2021-10-12:
It's a digression but in interesting one (and no: git is not going to totally change how submodules work just for us, you can forget about that)

If usability is not a top concern[*], I can see the submodules appeal of not confusing data with metadata because that helps avoid some dilemma. Take for example surepos of subrepos, I mean three levels of git repos. Now imagine the manifest at the middle level has been edited. So there are now two different manifests at the middle level: the edited one and the committed one. Which one should git use? The edited one because you should never ignore user input? Or the committed one because that's what the upper level refers to? Do you know what west does here? With submodules no need to ask yourself this question because you cannot "edit" the pointer, there's no file where you can do that. That extra degree of freedom is not present.

[*] https://www.ted.com/talks/linus_torvalds_the_mind_behind_linux/transcript, search "island"

tejlmand, 2021-10-12:
> > We need to make sure that when any developer builds a final image of the (app + NCS bundle) we can programatically verify that the sources used were exactly the expected ones, so that the final image can be bit by bit reproducible in any subsequent build.
>
> In my opinion this (very reasonable) requirement indicates a solution in the build system. The existence of ZEPHYR_MODULES and ZEPHYR_EXTRA_MODULES could push additional code into a local build that stands outside of anything in the manifest file, for example.

but we need to remember, there might be projects in the `west` manifest that are not Zephyr modules, and thus they won't be part of  `build/zephyr_modules.txt`.

But I do think there could be value in the CMake build system to support generating a meta file of:
- Zephyr modules in use, and there revision if the module is also a git repository
- west projects available if `west list` is used as part of the build process.

I think I would prefer to have projects duplicated in both `west` and `zephyr module` if they are both in west but also a Zephyr module.
Cause Zephyr modules themselves can only specify the SHAs and `dirty`.
But `west` may contain more project, and also have an expected revision.

Let me try to prototype, and you can provide some feedback.

0Grit, 2021-10-12:
> It's a digression but in interesting one (and no: git is not going to totally change how submodules work just for us, you can forget about that)
>
> If usability is not a top concern[*], I can see the [...]

I debated taking this to a separate github discussion since yes it is a digression and maybe still will. In the case referenced I might actually prefer git throw up it's hands and say "sorry, you touched something that should only be touched in isolation" 

So essentially prevent resolving and pulling the dependency change until the change to the manifest is committed upstream, to a branch (which would force a change in the manifest of the top level, or stashed?

tejlmand, 2021-10-12:
Note, that PR is simply a snapshot of what SHAs was actually in use when building and no relation to branches, tags, or similar.

marc-hb, 2021-10-15:
> eff0e1b  wasn't advertised as "breaking" because it omits printing useless information instead of changing the information printed.

A clean git status on a repo _away_ from manifest-rev is:
- something that commit eff0e1b  removed from `west status` output, and
- exactly what is requested in the description of this issue.

If you think this was "useless information" then I'm not sure why we (and especially you) spent so much time and effort discussing here.

xnlcasad, 2021-10-15:
thanks @mbolivar for such a complete answer. I agree that given all the use cases that west has to deal with it seems non-trivial to specify a no-surprise, straightforward behaviour. I agree with your suggestion of letting the community experiment and come back with a proposal that has proved itself to work in the trenches.

Just fyi, at the moment, we are hardcoding the repos sha1s in the freestanding application's CMakeLists.txt, like this:

```
macro(use_ncs_sdk_repo_at_revision path sha1)
   set(repo_${path} ${path} ${sha1})
   list(APPEND repo_list repo_${path})
endmacro()

function(validate_ncs_sdk list_of_repos clean_sdk)
    set(clean_sdk YES PARENT_SCOPE)
    foreach(a ${list_of_repos})
        list(GET ${a} 0 mod_path)
        list(GET ${a} 1 expected_mod_sha)
        message(STATUS "Checking repo: ${mod_path}, expected rev: ${expected_mod_sha}")
        execute_process(COMMAND git -C ${NCS_DIR}/${mod_path} rev-parse HEAD
                        OUTPUT_VARIABLE mod_sha OUTPUT_STRIP_TRAILING_WHITESPACE)
        if (NOT mod_sha STREQUAL expected_mod_sha)
            message(WARNING " repo: ${NCS_DIR}/${mod_path} is not at the expected revision.\n"
                            " expected revision: ${expected_mod_sha}\n"
                            " actual revision  : ${mod_sha}\n")
            set(clean_sdk NO PARENT_SCOPE)
        endif()
        execute_process(COMMAND git -C ${NCS_DIR}/${mod_path} diff --exit-code HEAD
                        RESULT_VARIABLE mod_dirty OUTPUT_QUIET ERROR_QUIET)
        if (mod_dirty STREQUAL 1)
            message(WARNING "repo: ${NCS_DIR}/${mod_path} has modified local files")
            set(clean_sdk NO PARENT_SCOPE)
        endif()
    endforeach(a)
endfunction()

use_ncs_sdk_repo_at_revision(nrf d29f1ddec99b81a66af93b7f0882a7bd0dfa30ee)
use_ncs_sdk_repo_at_revision(zephyr f19ebe4b50405b2eb174199b3635f24d741c9bf4)
use_ncs_sdk_repo_at_revision(nrfxlib a79fdc3dc432e40a13d9c631582aa5f83ea93635)
use_ncs_sdk_repo_at_revision(bootloader/mcuboot b514a3190821b7721dd6ffcd9940474ebe7a8832)

set(clean_sdk NO)
validate_ncs_sdk("${repo_list}" ${clean_sdk})
if (NOT IGNORE_NCS_SDK_STATE AND NOT clean_sdk)
    message(FATAL_ERROR "you are using an sdk that is not synced to this application")
endif ()
```
This is enough for our current needs. It has the caveat that this runs only at configure time, so we will need to rework it (maybe reusing or getting inspired from https://stackoverflow.com/a/4318642/2748456) so that configure is (re)triggered on git project status change.

However, I'm would still be interested in understanding how would west achieve R4 in https://docs.zephyrproject.org/latest/guides/west/why.html#requirements because I dont't see how to link a freestanding application with a specific version (down to the sha1) of the ncs sdk and its dependencies. This lets me wondering how I would go about bisecting the app's history accurately.

Then, another part of the problem is that we usually include the sha1 and the dirty status information within the final binary itself, so as to be able to rebuild from the exact same sources any binary found on a final product any time (even years) after the binary was produced. In other words this sha1+dirty information acts like a signature over all the source code used to build the application. I would have expected that west could have provided a similar feature to what git submodules provide, but as it looks, it is a non-trivial endeavour, due to the many other use cases that west has to handle.

IMO, we've done a good effort on brainstorming around this, and we see that there are some custom solutions that can be used to achieve this purpose somehow. This is enough for our needs, so I would be OK if you'd decide to close this issue now. Thanks!

mbolivar-nordic, 2021-10-20:
> However, I would still be interested in understanding how would west achieve R4 in https://docs.zephyrproject.org/latest/guides/west/why.html#requirements because I dont't see how to link a freestanding application with a specific version (down to the sha1) of the ncs sdk and its dependencies. This lets me wondering how I would go about bisecting the app's history accurately.

This requirement is met by west only for 'workspace applications'. If you're using a freestanding application, west isn't going to help you as much and you have to use the build system. This is because by definition a freestanding application is part from the workspace, which is what west manages.

> IMO, we've done a good effort on brainstorming around this, and we see that there are some custom solutions that can be used to achieve this purpose somehow. This is enough for our needs, so I would be OK if you'd decide to close this issue now. Thanks!

OK, thanks for confirming! I will close this then, but please feel free to reopen if you (or anyone else) feels they have a general solution backed up by experience and testing.

marc-hb, 2021-10-20:
> The commit you are referencing did not remove the output this issue is asking about from anything.

By chance this is the output of `west status` I can see right now on one of the systems I don't use often:

```
...
=== status of tflite-micro (modules/lib/tflite-micro):
HEAD detached at refs/heads/manifest-rev
nothing to commit, working tree clean
=== status of tinycbor (modules/lib/tinycbor):
HEAD detached at refs/heads/manifest-rev
Untracked files:
  (use "git add <file>..." to include in what will be committed)

	foo

nothing added to commit but untracked files present (use "git add" to track)
=== status of tinycrypt (modules/crypto/tinycrypt):
HEAD detached at refs/heads/manifest-rev
nothing to commit, working tree clean
=== status of TraceRecorderSource (modules/debug/TraceRecorder):
HEAD detached at 36c577727642
nothing to commit, working tree clean
=== status of trusted-firmware-m (modules/tee/tfm):
HEAD detached at c74be3890c9d
...
```
This is with very recent west version 38e656b05ea8 and git version 2.7.11. 

> I think you are confused about what west status does and always did do.

I honestly don't know why in this workspace `git status` displays `manifest-rev` most of the time and other times the SHA1 instead. But for sure the output above looks very much like what this issue requested.

Even when `git status` does not show `manifest-rev`, it provides the same information indirectly. I mean comparing a full `west status` output with the manifest is/was probably the most efficient way to work around for the lack of the feature requested here. So for sure dropping repos from that list did not help.

mbolivar-nordic, 2021-11-04:
> I honestly don't know why in this workspace `git status` displays `manifest-rev` most of the time and other times the SHA1 instead.

... because git status (and therefore west status) is displaying status relative to HEAD. Which can be `manifest-rev`, or it can be any other random commit, as your own example shows.

> But for sure the output above looks very much like what this issue requested.

No, again, HEAD != manifest-rev

marc-hb, 2021-11-05:
You missed the point. In the example I gave `HEAD == manifest-ref` in ALL repos whether `git status` shows `manifest-ref` or not. I added some comments to make that clearer.

marc-hb, 2021-11-05:
> I honestly don't know why in this workspace git status displays manifest-rev most of the time and other times the manifest-rev SHA1 instead

I had a look at git's source code to get to the bottom of this. This depends on the reflog. I didn't expect "git status" to look at the reflog but strace confirmed that.

In `git/wt_status.c`, `wt_status_get_detached_from()` scans the reflog depending on what it finds, it makes `wt_status_state->detached_from` non-zero which changes  the output of both `git status` and `git branch`: it makes these two commands stop displaying SHA1s and show "Detached from..." or "Detached at..." instead, pointing to the branch it found in the reflog.

After an initial `west init` + `west update`,  `git status` shows `HEAD detached at refs/heads/manifest-rev` in every repo. Every time the manifest changes, the corresponding repos will for some reflog reason switch to displaying a SHA instead.

Tests and analysis based on git 2.7.1, reproduced with git version 2.31.1

mbolivar-nordic, 2021-11-10:
> You missed the point. In the example I gave `HEAD == manifest-ref` in ALL repos whether `git status` shows `manifest-ref` or not.

OK, I see, in this particular situation you are saying they are the same; I get it. Thanks for explaining. But my point remains that they are not the same in general, though.

marc-hb, 2021-12-05:
> But my point remains that they are not the same in general, though.

?

Of course they are not in general, that's exactly why this issue was filed: to show differently the repos where they are not.

marc-hb, 2023-03-10:
New! `west compare` command proposed in #642

marc-hb, 2024-10-08:
> New! west compare command proposed in https://github.com/zephyrproject-rtos/west/issues/642

Now that `west compare` is available, people are still confused why their `west.yml` changes are ignored (not automatically propagated to `manifest-rev`). I think something like `west compare --fetch` would help, discuss in recent #747