zephyrproject-rtos/west #558: west init completes but leaves everything in zephyr/manifest-tmp

jefffhaynes, 2021-11-15:
I have no idea what causes this or how to reliably reproduce it, but running `west init` on windows occasionally leaves everything in the `zephyr/manifest-tmp` folder upon "successful" completion.  `west update` subsequently fails because everything is in the wrong place.

marc-hb, 2021-11-15:
Can you remember any error message, even vaguely?

Going forward, can you use `west -v -v init`?

jefffhaynes, 2021-11-15:
There is no error message until I run update as the init command doesn't fail, it just leaves everything in the wrong place. I'll try the new command.

marc-hb, 2021-11-16:
It could fail without a visible error message. It's extremely rare with `Python` and even rarer with `west` but it's theoretically possible. Next time this happens please use `echo %errorlevel%` immediately

jefffhaynes, 2021-11-16:
Running `west -v -v init` seems to have worked, although I don't know if that was the intention or if it just happened to work that time.  If it happens again I'll try to get the error level.  Thanks.

mbolivar-nordic, 2021-11-16:
`-v -v` should just increase the verbosity level without otherwise changing behavior (other than precise timing details of when things happen, of course), but the verbose logs would be very useful for diagnosing this. I haven't seen this happen myself, but I believe you, and it'd be good to get more details from the verbose logs.

congnguyenhuu, 2021-12-08:
Hi everyone, I have same problem when run `west init` on windows. All everything in the folder `manifest-tmp` moved folder `zephyr`, but The folder `manifest-tmp` is not removed. That make  can not run `west update`.
This is error display, Anyone have solution ? Thanks !

```
west.manifest: loaded C:\Users\congnh\zephyrproject\.west\manifest-tmp\zephyrbsp.yml
Traceback (most recent call last):
  File "c:\python38\lib\shutil.py", line 780, in move
    os.rename(src, real_dst)
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\congnh\\zephyrproject\\.west\\manifest-tmp' -> 'C:\\Users\\congnh\\zephyrproject\\zephyr'

```

marc-hb, 2021-12-08:
Can you reproduce every time? If yes then please follow the instructions above.

congnguyenhuu, 2021-12-08:
yes, it can reproduce. you mean use `west -v -v init` but I try and can not

jefffhaynes, 2021-12-08:
I think he's asking you to post what it says when you do.

congnguyenhuu, 2021-12-09:
This is all display when you did, it relates permission access folder when run os.rename
```
$ west -v -v init -m https://bitbucket.com/xx/xx/zephyr.git --mr xxxx.xx.xx.x --mf xxxx.yml
=== Initializing in C:\Users\congnh\zephyrproject
--- Cloning manifest repository from https://bitbucket.com/xx/xx/zephyr.git
running 'git clone --branch  https://bitbucket.com/'C:\Users\congnh\zephyrproject\.west\manifest-tmp'' in C:\Users\congnh\zephyrproject
Cloning into 'C:\Users\congnh\zephyrproject\.west\manifest-tmp'...
remote: Enumerating objects: 578417, done.
remote: Counting objects: 100% (578417/578417), done.
remote: Compressing objects: 100% (126962/126962), done.
remote: Total 578417 (delta 440454), reused 577231 (delta 439607)
Receiving objects: 100% (578417/578417), 383.54 MiB | 2.33 MiB/s, done.
Resolving deltas: 100% (440454/440454), done.
Note: switching to 'khsvt32h567586t8k'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Updating files: 100% (13543/13543), done.
west.manifest: loading C:\Users\congnh\zephyrproject\.west\manifest-tmp\zephyrbsp.yml
west.manifest: group-filter: unset
west.manifest: added project cmsis path modules/hal/cmsis revision from C:\Users\congnh\zephyrproject\.west\manifest-tmp\zephyrbsp.yml
west.manifest: added project hal_nxp path modules/hal/nxp revision from C:\Users\congnh\zephyrproject\.west\manifest-tmp\zephyrbsp.yml
west.manifest: loaded C:\Users\congnh8\Desktop\New folder\.west\manifest-tmp\zephyrbsp.yml
Traceback (most recent call last):
  File "c:\python38\lib\shutil.py", line 780, in move
    os.rename(src, real_dst)
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\congnh\\zephyrproject\\.west\\manifest-tmp' -> 'C:\\Users\\congnh\\zephyrproject\\zephyr'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\python38\lib\runpy.py", line 192, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\python38\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python38\Scripts\west.exe\__main__.py", line 9, in <module>
  File "c:\python38\lib\site-packages\west\app\main.py", line 781, in main
    app.run(argv or sys.argv[1:])
  File "c:\python38\lib\site-packages\west\app\main.py", line 106, in run
    self.run_command(argv)
  File "c:\python38\lib\site-packages\west\app\main.py", line 336, in run_command
    cmd.run(args, unknown, self.topdir, manifest=self.manifest)
  File "c:\python38\lib\site-packages\west\commands.py", line 140, in run
    self.do_run(args, unknown)
  File "c:\python38\lib\site-packages\west\app\project.py", line 198, in do_run
    topdir = self.bootstrap(args)
  File "c:\python38\lib\site-packages\west\app\project.py", line 305, in bootstrap
    shutil.move(os.fspath(tempdir), os.fspath(manifest_abspath))
  File "c:\python38\lib\shutil.py", line 792, in move
    rmtree(src)
  File "c:\python38\lib\shutil.py", line 730, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "c:\python38\lib\shutil.py", line 603, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "c:\python38\lib\shutil.py", line 603, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "c:\python38\lib\shutil.py", line 603, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "c:\python38\lib\shutil.py", line 608, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "c:\python38\lib\shutil.py", line 606, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\congnh\\zephyrproject\\.west\\manifest-tmp\\.git\\objects\\pack\\pack-2535647586gfhf56757.idx'
```

marc-hb, 2021-12-09:
> PermissionError: [WinError 5] Access is denied: 'C:\\Users\\congnh\\zephyrproject\\.west\\manifest-tmp' -> 'C:\\Users\\congnh\\zephyrproject\\zephyr'

What happens if you try to rename this yourself from a command window?

> west.manifest: loaded C:\Users\congnh8\Desktop\New folder\.west\manifest-tmp\zephyrbsp.yml

Even stranger: is this imported from another manifest?

congnguyenhuu, 2021-12-09:
`west.manifest: loaded C:\Users\congnh8\Desktop\New folder.west\manifest-tmp\zephyrbsp.yml`
sorry, I rename it , it must be this
`west.manifest: loaded C:\Users\congnh\zephyrproject\manifest-tmp\zephyrbsp.yml`

mbolivar-nordic, 2021-12-09:
> sorry, I rename it , it must be this

I don't understand what you mean, sorry. It looks like you have edited the logs. This makes it hard to understand them since I cannot know for sure how the output comes from the input, or even if the output is accurate.

Maybe you are using a private repository. I understand if you cannot share that.

But do you maybe have some way to make a manifest you can share which reproduces the issue for you? That would be really helpful, thank you.

congnguyenhuu, 2021-12-10:
yep. because some private information, i cannot post, so edited somethings in logs. but basically it is also same

marc-hb, 2021-12-10:
If you cannot reproduce with the open-source Zephyr manifest (did you try?), it means something in your manifest(s) is triggering the problem. To identify what triggers the issue, remove the private elements one by one until the problem disappears. Now you found what triggers the issue. Finally, to reproduce the issue without any confidential information, add something similar (and not confidential) to the open source manifest. Now you have a bug report that you can fully share and something in common we can all work on.

congnguyenhuu, 2021-12-13:
With community source  Zephyr, I have same problem. I try with other machines, some can work but some cannt
```
C:\Users\datnd14>west init -m https://github.com/zephyrproject-rtos/zephyr.git --mr zephyr-v2.7.0 --mf west.yml
=== Initializing in C:\Users\datnd14
--- Cloning manifest repository from https://github.com/zephyrproject-rtos/zephyr.git, rev. zephyr-v2.7.0
Initialized empty Git repository in C:/Users/datnd14/.west/manifest-tmp/.git/
remote: Enumerating objects: 642618, done.
remote: Counting objects: 100% (2/2), done.
remote: Total 642618 (delta 1), reused 1 (delta 1), pack-reused 642616 eceiving objects: 100% (642618/642618), 4
Receiving objects: 100% (642618/642618), 433.69 MiB | 25.00 KiB/s, done.
Resolving deltas: 100% (488975/488975), done.

* [new tag] zephyr-v2.7.0 -> zephyr-v2.7.0
503c7138df2b7b0afa0f649c783b33e8d971f4d0 refs/tags/zephyr-v2.7.0
Updating files: 100% (17693/17693), done.
Note: switching to 'zephyr-v2.7.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

git switch -c <new-branch-name>

Or undo this operation with:

git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 3f826560aa release: Zephyr 2.7.0
Traceback (most recent call last):
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\shutil.py", line 806, in move
os.rename(src, real_dst)
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\datnd14\\.west\\manifest-tmp' -> 'C:\\Users\\datnd14\\zephyr'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\runpy.py", line 197, in _run_module_as_main
return _run_code(code, main_globals, None,
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\runpy.py", line 87, in _run_code
exec(code, run_globals)
File "C:\Users\datnd14\AppData\Local\Programs\Python\Python39\Scripts\west.exe\__main__.py", line 7, in <module>
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\site-packages\west\app\main.py", line 779, in main
app.run(argv or sys.argv[1:])
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\site-packages\west\app\main.py", line 106, in run
self.run_command(argv)
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\site-packages\west\app\main.py", line 336, in run_command
cmd.run(args, unknown, self.topdir, manifest=self.manifest)
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\site-packages\west\commands.py", line 132, in run
self.do_run(args, unknown)
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\site-packages\west\app\project.py", line 198, in do_run
topdir = self.bootstrap(args)
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\site-packages\west\app\project.py", line 295, in bootstrap
shutil.move(os.fspath(tempdir), os.fspath(manifest_abspath))
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\shutil.py", line 824, in move
rmtree(src)
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\shutil.py", line 740, in rmtree
return _rmtree_unsafe(path, onerror)
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\shutil.py", line 613, in _rmtree_unsafe
_rmtree_unsafe(fullname, onerror)
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\shutil.py", line 613, in _rmtree_unsafe
_rmtree_unsafe(fullname, onerror)
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\shutil.py", line 613, in _rmtree_unsafe
_rmtree_unsafe(fullname, onerror)
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\shutil.py", line 618, in _rmtree_unsafe
onerror(os.unlink, fullname, sys.exc_info())
File "c:\users\datnd14\appdata\local\programs\python\python39\lib\shutil.py", line 616, in _rmtree_unsafe
os.unlink(fullname)
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\datnd14\\.west\\manifest-tmp\\.git\\objects\\pack\\pack-92b9de99850183d98ff55ccee114517db6eeae5c.idx'
```

marc-hb, 2021-12-13:
What happens when you try this after the failure:
```DOS
del C:\Users\datnd14\.west\manifest-tmp\.git\objects\pack\pack-92b9de99850183d98ff55ccee114517db6eeae5c.idx
```
Please also share the output of `west init -v -v...` and edit your comment above.

congnguyenhuu, 2021-12-14:
I tried delete that, but have access error although I grant permission
```
del C:\Users\datnd14\.west\manifest-tmp\.git\objects\pack\pack-5fe6ccedee5ad97c70bfdbc8b9476c8c056f7a98.idx
C:\Users\datnd14\.west\manifest-tmp\.git\objects\pack\pack-5fe6ccedee5ad97c70bfdbc8b9476c8c056f7a98.idx
Access is denied.
```
when run `west init -v -v`, log is quite long, so you can read in this file

[log.txt](https://github.com/zephyrproject-rtos/west/files/7710729/log.txt)

marc-hb, 2021-12-14:
Thanks! Can you please try this and report any error?

```DOS
cd C:\Users\datnd14\
git init w558
cd w558
git remote add origin -- https://github.com/zephyrproject-rtos/zephyr.git
git fetch origin --tags -- zephyr-v2.7.0 'refs/heads/*:refs/remotes/origin/*'
del .git\objects\pack\*
```

congnguyenhuu, 2021-12-15:
here is result when I run that
```
C:\Users\datnd14\w588>git remote add origin -- https://github.com/zephyrproject-rtos/zephyr.git

C:\Users\datnd14\w588>git fetch origin --tags -- zephyr-v2.7.0 'refs/heads/*:refs/remotes/origin/*'
remote: Enumerating objects: 607149, done.
remote: Counting objects: 100% (2/2), done.
remote: Total 607149 (delta 1), reused 1 (delta 1), pack-reused 607147
Receiving objects: 100% (607149/607149), 417.88 MiB | 596.00 KiB/s, done.
Resolving deltas: 100% (461118/461118), done.
From https://github.com/zephyrproject-rtos/zephyr
* tag zephyr-v2.7.0 -> FETCH_HEAD
* [new tag] v1.0.0 -> v1.0.0
* [new tag] v1.1.0 -> v1.1.0
* [new tag] v1.1.0-rc1 -> v1.1.0-rc1
* [new tag] v1.10.0 -> v1.10.0
* [new tag] v1.10.0-rc1 -> v1.10.0-rc1
* [new tag] v1.10.0-rc2 -> v1.10.0-rc2
* [new tag] v1.10.0-rc3 -> v1.10.0-rc3
* [new tag] v1.11.0 -> v1.11.0
* [new tag] v1.11.0-rc1 -> v1.11.0-rc1
* [new tag] v1.11.0-rc2 -> v1.11.0-rc2
* [new tag] v1.11.0-rc3 -> v1.11.0-rc3
* [new tag] v1.12.0 -> v1.12.0
* [new tag] v1.12.0-rc1 -> v1.12.0-rc1
* [new tag] v1.12.0-rc2 -> v1.12.0-rc2
* [new tag] v1.12.0-rc3 -> v1.12.0-rc3
* [new tag] v1.13.0 -> v1.13.0
* [new tag] v1.13.0-rc1 -> v1.13.0-rc1
* [new tag] v1.13.0-rc2 -> v1.13.0-rc2
* [new tag] v1.13.0-rc3 -> v1.13.0-rc3
* [new tag] v1.14.0 -> v1.14.0
* [new tag] v1.14.0-rc1 -> v1.14.0-rc1
* [new tag] v1.14.0-rc2 -> v1.14.0-rc2
* [new tag] v1.14.0-rc3 -> v1.14.0-rc3
* [new tag] v1.14.1 -> v1.14.1
* [new tag] v1.14.1-rc1 -> v1.14.1-rc1
* [new tag] v1.14.1-rc2 -> v1.14.1-rc2
* [new tag] v1.14.1-rc3 -> v1.14.1-rc3
* [new tag] v1.14.2 -> v1.14.2
* [new tag] v1.14.3 -> v1.14.3
* [new tag] v1.14.3-rc1 -> v1.14.3-rc1
* [new tag] v1.14.3-rc2 -> v1.14.3-rc2
* [new tag] v1.2.0 -> v1.2.0
* [new tag] v1.2.0-rc1 -> v1.2.0-rc1
* [new tag] v1.2.0-rc2 -> v1.2.0-rc2
* [new tag] v1.3.0 -> v1.3.0
* [new tag] v1.3.0-rc1 -> v1.3.0-rc1
* [new tag] v1.3.0-rc2 -> v1.3.0-rc2
* [new tag] v1.4.0 -> v1.4.0
* [new tag] v1.4.0-rc1 -> v1.4.0-rc1
* [new tag] v1.4.0-rc2 -> v1.4.0-rc2
* [new tag] v1.4.0-rc3 -> v1.4.0-rc3
* [new tag] v1.5.0 -> v1.5.0
* [new tag] v1.5.0-rc0 -> v1.5.0-rc0
* [new tag] v1.5.0-rc1 -> v1.5.0-rc1
* [new tag] v1.5.0-rc2 -> v1.5.0-rc2
* [new tag] v1.5.0-rc3 -> v1.5.0-rc3
* [new tag] v1.5.0-rc4 -> v1.5.0-rc4
* [new tag] v1.6.0 -> v1.6.0
* [new tag] v1.6.0-rc1 -> v1.6.0-rc1
* [new tag] v1.6.0-rc2 -> v1.6.0-rc2
* [new tag] v1.6.0-rc3 -> v1.6.0-rc3
* [new tag] v1.6.0-rc4 -> v1.6.0-rc4
* [new tag] v1.6.1 -> v1.6.1
* [new tag] v1.6.1-rc -> v1.6.1-rc
* [new tag] v1.6.99 -> v1.6.99
* [new tag] v1.7.0 -> v1.7.0
* [new tag] v1.7.0-rc1 -> v1.7.0-rc1
* [new tag] v1.7.0-rc2 -> v1.7.0-rc2
[... 68 lines cut ...]
* [new tag] zephyr-v1.6.0 -> zephyr-v1.6.0
* [new tag] zephyr-v1.6.1 -> zephyr-v1.6.1
* [new tag] zephyr-v1.7.0 -> zephyr-v1.7.0
* [new tag] zephyr-v1.7.1 -> zephyr-v1.7.1
* [new tag] zephyr-v1.8.0 -> zephyr-v1.8.0
* [new tag] zephyr-v1.9.0 -> zephyr-v1.9.0
* [new tag] zephyr-v1.9.1 -> zephyr-v1.9.1
* [new tag] zephyr-v1.9.2 -> zephyr-v1.9.2
* [new tag] zephyr-v2.0.0 -> zephyr-v2.0.0
* [new tag] zephyr-v2.1.0 -> zephyr-v2.1.0
* [new tag] zephyr-v2.2.0 -> zephyr-v2.2.0
* [new tag] zephyr-v2.2.1 -> zephyr-v2.2.1
* [new tag] zephyr-v2.3.0 -> zephyr-v2.3.0
* [new tag] zephyr-v2.4.0 -> zephyr-v2.4.0
* [new tag] zephyr-v2.5.0 -> zephyr-v2.5.0
* [new tag] zephyr-v2.6.0 -> zephyr-v2.6.0
* [new tag] zephyr-v2.7.0 -> zephyr-v2.7.0

C:\Users\datnd14\w588>del .git\objects\pack\*
C:\Users\datnd14\w588\.git\objects\pack\*, Are you sure (Y/N)? Y
C:\Users\datnd14\w588\.git\objects\pack\pack-9554992bbd0bdd6f346a8acf37ca89822648585f.idx
Access is denied.
C:\Users\datnd14\w588\.git\objects\pack\pack-9554992bbd0bdd6f346a8acf37ca89822648585f.pack
Access is denied.
```

marc-hb, 2021-12-15:
Great, so now we know you don't have a west problem. I think you have a problem with your git client and/or the way Windows permissions are set up in your home directory. Can you try the same thing with a USB stick or just in a different directory?

It seems normal for git to create read only pack files  but somehow this is not a problem for most people. Windows permissions can be very complicated and I'm no expert https://www.ntfs.com/ntfs-permissions-file-effective.htm

jaredwolff, 2022-01-05:
I'm getting the same error invoking `west` via `cmd.exe` though a VSCode extension. 

Full command is:
```
C:\Windows\System32\cmd.exe /d /c west -v -v init -m https://github.com/circuitdojo/nrf9160-feather-examples-and-drivers.git
```

Here's the output. 

```
> Executing task: west -v -v init -m https://github.com/circuitdojo/nrf9160-feather-examples-and-drivers.git <

=== Initializing in C:\Zephyr
--- Cloning manifest repository from https://github.com/circuitdojo/nrf9160-feather-examples-and-drivers.git
running 'git clone https://github.com/circuitdojo/nrf9160-feather-examples-and-drivers.git 'C:\Zephyr\.west\manifest-tmp'' in C:\Zephyr
Cloning into 'C:\Zephyr\.west\manifest-tmp'...
remote: Enumerating objects: 899, done.
remote: Counting objects: 100% (899/899), done.
remote: Compressing objects: 100% (521/521), done.
remote: Total 899 (delta 472), reused 750 (delta 327), pack-reused 0
Receiving objects: 100% (899/899), 6.71 MiB | 9.22 MiB/s, done.     
Resolving deltas: 100% (472/472), done.
west.manifest: loading C:\Zephyr\.west\manifest-tmp\west.yml
west.manifest: group-filter: unset
west.manifest: added project nrf path nrf revision v1.7.1 from C:\Zephyr\.west\manifest-tmp\west.yml
west.manifest: project <Project nrf ('C:\\Zephyr\\nrf') at v1.7.1>: ignored import (True) 
west.manifest: added project golioth path modules/lib/golioth revision main from C:\Zephyr\.west\manifest-tmp\west.yml
west.manifest: project <Project golioth ('C:\\Zephyr\\modules\\lib\\golioth') at main>: ignored import ({'name-allowlist': ['qcbor']})
west.manifest: added project pyrinas path pyrinas revision v1.7.x from C:\Zephyr\.west\manifest-tmp\west.yml
west.manifest: project <Project pyrinas ('C:\\Zephyr\\pyrinas') at v1.7.x>: ignored import (True)
west.manifest: loaded C:\Zephyr\.west\manifest-tmp\west.yml
Traceback (most recent call last):
  File "C:\Python39\lib\shutil.py", line 803, in move
    os.rename(src, real_dst)
PermissionError: [WinError 5] Access is denied: 'C:\\Zephyr\\.west\\manifest-tmp' -> 'C:\\Zephyr\\nfed'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Jared Wolff\.zephyrtools\env\Scripts\west.exe\__main__.py", line 7, in <module>
  File "C:\Users\Jared Wolff\.zephyrtools\env\lib\site-packages\west\app\main.py", line 781, in main
    app.run(argv or sys.argv[1:])
  File "C:\Users\Jared Wolff\.zephyrtools\env\lib\site-packages\west\app\main.py", line 106, in run
    self.run_command(argv)
  File "C:\Users\Jared Wolff\.zephyrtools\env\lib\site-packages\west\app\main.py", line 336, in run_command
    cmd.run(args, unknown, self.topdir, manifest=self.manifest)
  File "C:\Users\Jared Wolff\.zephyrtools\env\lib\site-packages\west\commands.py", line 140, in run
    self.do_run(args, unknown)
  File "C:\Users\Jared Wolff\.zephyrtools\env\lib\site-packages\west\app\project.py", line 198, in do_run
    topdir = self.bootstrap(args)
  File "C:\Users\Jared Wolff\.zephyrtools\env\lib\site-packages\west\app\project.py", line 305, in bootstrap
    shutil.move(os.fspath(tempdir), os.fspath(manifest_abspath))
  File "C:\Python39\lib\shutil.py", line 815, in move
    rmtree(src)
  File "C:\Python39\lib\shutil.py", line 737, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\Python39\lib\shutil.py", line 610, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python39\lib\shutil.py", line 610, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python39\lib\shutil.py", line 610, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python39\lib\shutil.py", line 615, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Python39\lib\shutil.py", line 613, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 5] Access is denied: 'C:\\Zephyr\\.west\\manifest-tmp\\.git\\objects\\pack\\pack-ae179110ddff9c4a119793934f0f44c69e9e0976.idx'
The terminal process "C:\Windows\System32\cmd.exe /d /c west -v -v init -m https://github.com/circuitdojo/nrf9160-feather-examples-and-drivers.git" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

jaredwolff, 2022-01-05:
Diff of the environment. Works in a stand alone cmd but not within VSCode

```diff
1d0
< !::=::\
3d1
< !ExitCode=00000000
9a8
> CHROME_CRASHPAD_PIPE_NAME=\\.\pipe\crashpad_9232_XCMYRGDVZKPZBDFN
24a24
> ORIGINAL_XDG_CURRENT_DESKTOP=undefined
26c26,27
< PATH=/cygdrive/c/Python39/Scripts:/cygdrive/c/Windows/system32:/cygdrive/c/Windows:/cygdrive/c/Windows/System32/Wbem:/cygdrive/c/Windows/System32/WindowsPowerShell/v1.0:/usr/bin:/cygdrive/c/Windows/System32/OpenSSH:/cygdrive/c/Program Files/Wireshark:/cygdrive/c/Program Files/PuTTY:/cygdrive/c/ProgramData/chocolatey/bin:/cygdrive/c/Program Files/Git/cmd:/cygdrive/c/bin:/cygdrive/c/Python39:/cygdrive/c/Program Files (x86)/Nordic Semiconductor/nrf-command-line-tools/bin:/cygdrive/c/Program Files (x86)/SEGGER/JLink:/cygdrive/c/Program Files (x86)/ZeroTier/One:/cygdrive/c/Program Files/Microsoft SQL Server/120/Tools/Binn:/cygdrive/c/Program Files/Microsoft SQL Server/Client SDK/ODBC/110/Tools/Binn:/cygdrive/c/Program Files (x86)/Microsoft SQL Server/120/Tools/Binn:/cygdrive/c/Program Files/Microsoft SQL Server/120/DTS/Binn:/cygdrive/c/Program Files (x86)/nodejs:/cygdrive/c/Users/Jared Wolff/AppData/Local/Microsoft/WindowsApps:/cygdrive/c/Users/Jared Wolff/AppData/Local/Programs/Microsoft VS Code/bin:/cygdrive/c/bin:/cygdrive/c/Python39:/cygdrive/c/Users/Jared Wolff/AppData/Roaming/npm
---
> PATH=/cygdrive/c/Python39/Scripts:/cygdrive/c/Windows/system32:/cygdrive/c/Windows:/cygdrive/c/Windows/System32/Wbem:/cygdrive/c/Windows/System32/WindowsPowerShell/v1.0:/usr/bin:/cygdrive/c/Windows/System32/OpenSSH:/cygdrive/c/Program Files/Wireshark:/cygdrive/c/Program Files/PuTTY:/cygdrive/c/ProgramData/chocolatey/bin:/cygdrive/c/Program Files/Git/cmd:/cygdrive/c/bin:/cygdrive/c/Python39:/cygdrive/c/Program Files (x86)/Nordic Semiconductor/nrf-command-line-tools/bin:/cygdrive/c/Program Files (x86)/SEGGER/JLink:/cygdrive/c/Program Files (x86)/ZeroTier/One:/cygdrive/c/Program Files/Microsoft SQL Server/120/Tools/Binn:/cygdrive/c/Program Files/Microsoft SQL Server/Client SDK/ODBC/110/Tools/Binn:/cygdrive/c/Program Files (x86)/Microsoft SQL 
> Server/120/Tools/Binn:/cygdrive/c/Program Files/Microsoft SQL Server/120/DTS/Binn:/cygdrive/c/Program Files (x86)/nodejs:/cygdrive/c/Users/Jared Wolff/AppData/Local/Microsoft/WindowsApps:/cygdrive/c/Users/Jared Wolff/AppData/Local/Programs/Microsoft VS Code/bin:/cygdrive/c/bin:/cygdrive/c/Python39:/cygdrive/c/Users/Jared Wolff/AppData/Roaming/npm
49a51,59
> TERM_PROGRAM=vscode
> TERM_PROGRAM_VERSION=1.63.2
> LANG=en_US.UTF-8
> COLORTERM=truecolor
> VSCODE_GIT_IPC_HANDLE=\\.\pipe\vscode-git-35da2a6a1e-sock
> GIT_ASKPASS=c:\Users\Jared Wolff\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\askpass.sh
> VSCODE_GIT_ASKPASS_NODE=C:\Users\Jared Wolff\AppData\Local\Programs\Microsoft VS Code\Code.exe
> VSCODE_GIT_ASKPASS_EXTRA_ARGS=--ms-enable-electron-run-as-node
> VSCODE_GIT_ASKPASS_MAIN=c:\Users\Jared Wolff\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\askpass-main.js
```

Cygwin looks like the culprit

jaredwolff, 2022-01-05:
Looks like it's because I'm running `west` in a virtual environment. Without the virtual environment, works ok. With the path to the virtual environment set, it breaks.

mbolivar-nordic, 2022-03-11:
> After **lots** of testing. I've narrowed it down to whether or not VSCode has the folder open. It appears to be scanning the files inside the `.west` folder. When `west` is ready to transfer the repo to it's final resting place, it's prevented from doing so because VSCode is indexing the files.

@wbober @trond-snekvik I was thinking about the VS code extension and I wanted to let you know about this to see if you've ever observed random issues like this.

Some of the above discussion looks like random permissions problems that unfortunately I have no idea how to triage right now, but this particular issue reported by @jaredwolff might be something you are running into more often.

marc-hb, 2022-03-11:
> it's prevented from doing so because VSCode is indexing the files.

It would help if Windows could delete files in use like Linux filesystems can (it would also save a lot of Windows reboots, I digress)

However similar issues could happen on Linux too: a few years ago, it came as a big surprise to me that some operations like "git diff" appear to be read-only but in fact they have some side effects because they can cause git to update some of its internal data: https://public-inbox.org/git/20160331140515.GA31116@sigill.intra.peff.net/
So in theory we could have a similar race where some Linux IDE triggers some not-so-read-only git commands on a half-moved repo and ends up corrupting it. Very unlikely but possible?

So, @mbolivar-nordic should west switch to some OS-dependent and "standard" /tmp locations to escape this sort of concurrency? Even if the interactions with IDEs ends up being harmless, it's not "nice" to confuse them.

mbolivar-nordic, 2022-03-11:
> So, @mbolivar-nordic should west switch to some OS-dependent and "standard" /tmp locations to escape this sort of concurrency? Even if the interactions with IDEs ends up being harmless, it's not "nice" to confuse them.

I thought about that, but then you're inviting issues with cross-filesystem moves, which are a can of worms of their own. If you have a strategy for accomplishing this ask, I'm open to ideas, though.

marc-hb, 2022-03-11:
> but then you're inviting issues with cross-filesystem moves, 

OMG, I referenced something from 2016 but now you're bringing back memories from 2010!!
https://bugs.eclipse.org/bugs/show_bug.cgi?id=319233 "Cloning repository fails if destination is on network drive because File.renameTo doesn't succeed"

How about `zephyr/.manifest-tmp` instead of `zephyr/manifest-tmp`? Or some "hidden" Windows file attribute? Both? @jaredwolff would you be able to test these on Windows?

jaredwolff, 2022-03-11:
My work around with this, from what I remember, was disabling the Git functionality in VSCode temporarily while `west` did its thing. It was trying to cache or open the repository in the `manifest-tmp` folder and thus locking up the files. Next time I'm working on my VSCode plugin I'll have to see if changing the manifest-tmp directory name fixes it.

mbolivar-nordic, 2022-03-11:
> How about `zephyr/.manifest-tmp` instead of `zephyr/manifest-tmp`? Or some "hidden" Windows file attribute? Both?

I don't know what you mean, sorry. The temporary manifest repository already goes under `.west`, not `zephyr`. I don't know a lot about hidden files on windows or what vs code does. If any body has any resources they can point me at, I'd be grateful.

marc-hb, 2022-03-12:
> The temporary manifest repository already goes under .west, not zephyr

Sorry I blindly trusted the description and title at the very top. At the moment they are:

> west init completes but leaves everything in zephyr/manifest-tmp
>
> but running west init on windows occasionally leaves everything in the zephyr/manifest-tmp folder upon "successful" completion

marc-hb, 2022-03-14:
In a totally unrelated discussion I just heard that VS2022 is automatically generating some `c/Users/me/.gitconfig` file. Sounds nasty, what else does it do?

jefffhaynes, 2022-09-04:
I just rebuilt a new machine and this is still happening.  VSCode was not running at any point.  Running ```west -v -v init``` results in:

```
PS C:\Users\Jeff\Source\Repos\Test> west -v -v init
=== Initializing in C:\Users\Jeff\Source\Repos\Test
--- Cloning manifest repository from https://github.com/zephyrproject-rtos/zephyr
running 'git clone https://github.com/zephyrproject-rtos/zephyr 'C:\Users\Jeff\Source\Repos\Test\.west\manifest-tmp'' in C:\Users\Jeff\Source\Repos\Test
Cloning into 'C:\Users\Jeff\Source\Repos\Test\.west\manifest-tmp'...
remote: Enumerating objects: 765677, done.
remote: Counting objects: 100% (2/2), done.
remote: Total 765677 (delta 1), reused 1 (delta 1), pack-reused 765675
Receiving objects: 100% (765677/765677), 534.13 MiB | 931.00 KiB/s, done.

Resolving deltas: 100% (582305/582305), done.
Updating files: 100% (21810/21810), done.
west.manifest: loading data (no file)
west.manifest: resolving self import submanifests
west.manifest: ignored self import
west.manifest: resolved self import
west.manifest: group-filter: ['-ci']
west.manifest: added project canopennode path modules/lib/canopennode revision 53d3415c14d60f8f4bfca54bfbc5d5a667d7e724
west.manifest: added project chre path modules/lib/chre revision ef76d3456db07e4959df555047d6962279528c8d
west.manifest: added project cmsis path modules/hal/cmsis revision 093de61c2a7d12dc9253daf8692f61f793a9254a groups ['hal']
west.manifest: added project edtt path tools/edtt revision 1ea61a390d2bfcf3b2ecdba8f8b0b98dfdffbd11 groups ['tools']
west.manifest: added project fatfs path modules/fs/fatfs revision a30531af3a95a9a3ea7d771ea8a578ebfed45514 groups ['fs']
west.manifest: added project fff path modules/lib/fff revision 6ce5ba26486e93d5b7696a3e23f0585932c14b16 groups ['ci']
west.manifest: added project hal_altera path modules/hal/altera revision 0d225ddd314379b32355a00fb669eacf911e750d groups ['hal']
west.manifest: added project hal_atmel path modules/hal/atmel revision 1d237f2e2f262751975b6da6e03af569b2b49b2b groups ['hal']
west.manifest: added project hal_espressif path modules/hal/espressif revision 1e29548e4492807eb18cbaa77d3e78801cc7d07e groups ['hal']
west.manifest: added project hal_gigadevice path modules/hal/gigadevice revision dd0e0322474462b58059e6fedaf1d67d2a0864d0 groups ['hal']
west.manifest: added project hal_infineon path modules/hal/infineon revision 4af06965f57ba1e7d170e6a97d24c33785543a8c groups ['hal']
west.manifest: added project hal_microchip path modules/hal/microchip revision 5d079f1683a00b801373bbbbf5d181d4e33b30d5 groups ['hal']
west.manifest: added project hal_nordic path modules/hal/nordic revision 249199ec5a5c31d170659921048764e96d05cc0e groups ['hal']
west.manifest: added project hal_nuvoton path modules/hal/nuvoton revision b4d31f33238713a568e23618845702fadd67386f groups ['hal']
west.manifest: added project hal_nxp path modules/hal/nxp revision 3516aa5f722eb70b969b3bc57bcf9389ad901057 groups ['hal']
west.manifest: added project hal_openisa path modules/hal/openisa revision 40d049f69c50b58ea20473bee14cf93f518bf262 groups ['hal']
west.manifest: added project hal_quicklogic path modules/hal/quicklogic revision b3a66fe6d04d87fd1533a5c8de51d0599fcd08d0 groups ['hal']
west.manifest: added project hal_renesas path modules/hal/renesas revision 468d3f2146d18c7f86a4640fc641cc1d20a4a100 groups ['hal']
west.manifest: added project hal_rpi_pico path modules/hal/rpi_pico revision 191f5ba46fda49523cdaaef27583d1c875ba2c36 groups ['hal']
west.manifest: added project hal_silabs path modules/hal/silabs revision 1ec8dd99aa4ac3e8632d2aa28a7438049bb27102 groups ['hal']
west.manifest: added project hal_st path modules/hal/st revision cccbc24c14decfd3f93959f7b14514536af973c7 groups ['hal']
west.manifest: added project hal_stm32 path modules/hal/stm32 revision 5b1610c3cf46589d50e2634ba240b7b38bc59c53 groups ['hal']
west.manifest: added project hal_telink path modules/hal/telink revision 38573af589173259801ae6c2b34b7d4c9e626746 groups ['hal']
west.manifest: added project hal_ti path modules/hal/ti revision 29af528900d03af02a17b4b7f4fc93f19a81da2a groups ['hal']
west.manifest: added project hal_wurthelektronik path modules/hal/wurthelektronik revision 24ca9873c3d608fad1fea0431836bc8f144c132e groups ['hal']
west.manifest: added project hal_xtensa path modules/hal/xtensa revision 63f655362423aa49507da7977a2d37142e8debeb groups ['hal']
west.manifest: added project libmetal path modules/hal/libmetal revision 2f586b4f1276fb075ee145421bdf6cbe5403aa41 groups ['hal']
west.manifest: added project liblc3 path modules/lib/liblc3 revision 448f3de31f49a838988a162ef1e23a89ddf2d2ed
west.manifest: added project littlefs path modules/fs/littlefs revision 652f2c5646e79b881e6f3099686ad3b7af9e216c groups ['fs']
west.manifest: added project loramac-node path modules/lib/loramac-node revision 0257b50905695192d095667b1c3abb80346db1a1
west.manifest: added project lvgl path modules/lib/gui/lvgl revision af95bdfcf6784edd958ea08139c713e2d3dee7af
west.manifest: added project lz4 path modules/lib/lz4 revision 8e303c264fc21c2116dc612658003a22e933124d
west.manifest: added project mbedtls path modules/crypto/mbedtls revision 7fed49c9b9f983ad6416986661ef637459723bcb groups ['crypto']
west.manifest: added project mcuboot path bootloader/mcuboot revision 21e56a1bd033af263e808940779a1adaf4d5465e
west.manifest: added project mipi-sys-t path modules/debug/mipi-sys-t revision 0d521d8055f3b2b4842f728b0365d3f0ece9c37f groups ['debug']
[... cut ...]
west.manifest: added project zscilib path modules/lib/zscilib revision a54986aa98db4082ac56b582843bb5b5435208a6
west.manifest: loaded data (no file)
FATAL ERROR: Destination path 'C:\Users\Jeff\Source\Repos\Test\zephyr\manifest-tmp' already exists
```

After running the files once again appear to be in ```.west/manifest-tmp``` and at this point I can't init or update.  It would be great if we could get a bug label on this.  This is a pristine copy of Windows 11 and I've followed the Zephyr setup instructions to a T.  There is obviously an issue of some sort.

jefffhaynes, 2022-09-05:
Ok I misread the error and it looks like I still had a zephyr/manifest-tmp directory, possibly left over from the "vs code" issue.  Still, the whole process seems quite fragile.

marc-hb, 2022-09-06:
If you don't wipe out `C:\Users\Jeff\Source\Repos\Test` before attempting again then many things can happen; I wouldn't call that "fragile".

jefffhaynes, 2022-09-06:
I have a repo at that location so I can't wipe it out. Checking out zephyr to a different project and then running the environment scripts has never worked for me so I have zephyr in my project directly, the same way many projects do.

marc-hb, 2022-09-06:
Understood. Makes sense. You should at least delete `.west/` completely:
- https://github.com/zephyrproject-rtos/west/issues/593

jefffhaynes, 2022-09-06:
I had but I missed the zephyr directory, thanks

timonsku, 2023-01-12:
Had something related to this general issue happen too today. I used `west init` and it failed when it tried to link the cloned manifest (using defaults, it freshly cloned). Python threw a permission error when trying to link the manifest directory to the zephyr directory `PermissionError: [WinError 5] Access is denied`
Unfortunately I lost the log history after executing another command.
Bottom line though is that it worked fine after I deleted the `.west` and `zephyr` folder and just ran `west init` again.
So possible that the issue came from an issue during cloning. The project was definitely "clean" before attempting the first init and had no old files laying around.

attie-argentum, 2023-08-03:
I can reliably reproduce this on Linux with the following steps:

```bash
git clone https://github.com/zephyrproject-rtos/zephyr.git
west -vv init ./ -m https://github.com/zephyrproject-rtos/zephyr.git
```
<details><summary>Console Output</summary>

```
user@c3a266ec741a:~/zephyrproject$ west --version
West version: v1.1.0
user@c3a266ec741a:~/zephyrproject$ mkdir ./zephyr/
user@c3a266ec741a:~/zephyrproject$ west -vv init ./ -m https://github.com/zephyrproject-rtos/zephyr.git
=== Initializing in /home/user/zephyrproject
--- Cloning manifest repository from https://github.com/zephyrproject-rtos/zephyr.git
running 'git clone https://github.com/zephyrproject-rtos/zephyr.git /home/user/zephyrproject/.west/manifest-tmp' in /home/user/zephyrproject
Cloning into '/home/user/zephyrproject/.west/manifest-tmp'...
remote: Enumerating objects: 903441, done.
remote: Counting objects: 100% (17018/17018), done.
remote: Compressing objects: 100% (889/889), done.
remote: Total 903441 (delta 16322), reused 16130 (delta 16129), pack-reused 886423
Receiving objects: 100% (903441/903441), 584.57 MiB | 15.85 MiB/s, done.
Resolving deltas: 100% (685965/685965), done.
west.manifest: DEBUG: loading data (no file)
west.manifest: DEBUG: resolving self import submanifests
west.manifest: DEBUG: ignored self import
west.manifest: DEBUG: resolved self import
west.manifest: DEBUG: group-filter: ['-babblesim']
west.manifest: DEBUG: added project acpica path modules/lib/acpica revision f16a0b4d0f0edd7b78a332fcf507be2187fac21e
west.manifest: DEBUG: added project bsim path tools/bsim revision 384a091445c57b44ac8cbd18ebd245b47c71db94 groups ['babblesim']
west.manifest: DEBUG: added project babblesim_base path tools/bsim/components revision 19d62424c0802c6c9fc15528febe666e40f372a1 groups ['babblesim']
west.manifest: DEBUG: added project babblesim_ext_2G4_libPhyComv1 path tools/bsim/components/ext_2G4_libPhyComv1 revision 9018113a362fa6c9e8f4b9cab9e5a8f12cc46b94 groups ['babblesim']
west.manifest: DEBUG: added project babblesim_ext_2G4_phy_v1 path tools/bsim/components/ext_2G4_phy_v1 revision d47c6dd90035b41b14f6921785ccb7b8484868e2 groups ['babblesim']
west.manifest: DEBUG: added project babblesim_ext_2G4_channel_NtNcable path tools/bsim/components/ext_2G4_channel_NtNcable revision 20a38c997f507b0aa53817aab3d73a462fff7af1 groups ['babblesim']
west.manifest: DEBUG: added project babblesim_ext_2G4_channel_multiatt path tools/bsim/components/ext_2G4_channel_multiatt revision bde72a57384dde7a4310bcf3843469401be93074 groups ['babblesim']
west.manifest: DEBUG: added project babblesim_ext_2G4_modem_magic path tools/bsim/components/ext_2G4_modem_magic revision cb70771794f0bf6f262aa474848611c68ae8f1ed groups ['babblesim']
west.manifest: DEBUG: added project babblesim_ext_2G4_modem_BLE_simple path tools/bsim/components/ext_2G4_modem_BLE_simple revision 809ab073159c9ab6686c2fea5749b0702e0909f7 groups ['babblesim']
west.manifest: DEBUG: added project babblesim_ext_2G4_device_burst_interferer path tools/bsim/components/ext_2G4_device_burst_interferer revision 5b5339351d6e6a2368c686c734dc8b2fc65698fc groups ['babblesim']
west.manifest: DEBUG: added project babblesim_ext_2G4_device_WLAN_actmod path tools/bsim/components/ext_2G4_device_WLAN_actmod revision 9cb6d8e72695f6b785e57443f0629a18069d6ce4 groups ['babblesim']
west.manifest: DEBUG: added project babblesim_ext_2G4_device_playback path tools/bsim/components/ext_2G4_device_playback revision 85c645929cf1ce995d8537107d9dcbd12ed64036 groups ['babblesim']
west.manifest: DEBUG: added project babblesim_ext_libCryptov1 path tools/bsim/components/ext_libCryptov1 revision eed6d7038e839153e340bd333bc43541cb90ba64 groups ['babblesim']
west.manifest: DEBUG: added project canopennode path modules/lib/canopennode revision dec12fa3f0d790cafa8414a4c2930ea71ab72ffd
west.manifest: DEBUG: added project chre path modules/lib/chre revision b7955c27e50485b7dafdc3888d7d6afdc2ac6d96
west.manifest: DEBUG: added project cmsis path modules/hal/cmsis revision 74981bf893e8b10931464b9945e2143d99a3f0a3 groups ['hal']
west.manifest: DEBUG: added project edtt path tools/edtt revision 64e5105ad82390164fb73fc654be3f73a608209a groups ['tools']
west.manifest: DEBUG: added project fatfs path modules/fs/fatfs revision 427159bf95ea49b7680facffaa29ad506b42709b groups ['fs']
west.manifest: DEBUG: added project hal_altera path modules/hal/altera revision 0d225ddd314379b32355a00fb669eacf911e750d groups ['hal']
west.manifest: DEBUG: added project hal_atmel path modules/hal/atmel revision 5ab43007eda3f380c125f957f03638d2e8d1144d groups ['hal']
west.manifest: DEBUG: added project hal_espressif path modules/hal/espressif revision ae1bd648a1ac701672c46b6ff4eadfa716937ce3 groups ['hal']
west.manifest: DEBUG: added project hal_ethos_u path modules/hal/ethos_u revision 90ada2ea5681b2a2722a10d2898eac34c2510791 groups ['hal']
west.manifest: DEBUG: added project hal_gigadevice path modules/hal/gigadevice revision 2994b7dde8b0b0fa9b9c0ccb13474b6a486cddc3 groups ['hal']
west.manifest: DEBUG: added project hal_infineon path modules/hal/infineon revision 0bebc14d8bd1a249ee7fbc70b37db6f01f72544f groups ['hal']
west.manifest: DEBUG: added project hal_intel path modules/hal/intel revision 89a0db5d9155484624447d62a5211bcf6f6f0387 groups ['hal']
west.manifest: DEBUG: added project hal_microchip path modules/hal/microchip revision 5d079f1683a00b801373bbbbf5d181d4e33b30d5 groups ['hal']
[... cut ...]
I'm reading into it / making some assumptions... but in this case the issue appears to be that the target `./zephyr/` directory isn't empty - so after `west init` clones into `./.west/manifest-tmp/`, it performs a "_dumb_" `mv ./.west/manifest-tmp/ ./zephyr/`, which results placing `./.west/manifest-tmp/` into the existing `./zephyr/` directory (final path `./zephyr/manifest-tmp/`), _**not**_ renaming `./.west/manifest-tmp/` to `./zephyr/`... i.e: no check is performed for `./zephyr/` existing already.

Other modules are cloned directly into their final destination, so `git` is responsible for checking the state of things.

This behavior is present back to at least v0.12.0.

attie-argentum, 2023-08-03:
PS: `west` uses [`shutil.move()`](https://docs.python.org/3/library/shutil.html#shutil.move) to perform this move.

> If the destination is an existing directory, then src is moved inside that directory. If the destination already exists but is not a directory, it may be overwritten depending on os.rename() semantics.

https://github.com/zephyrproject-rtos/west/blob/9e8f5002f2ee86b7150c039cb69aa653ff454162/src/west/app/project.py#L330

attie-argentum, 2023-08-03:
I believe that the issue reported by @congnguyenhuu / @jaredwolff / @timonsku is different to the original issue, and is certainly different to mine.

The crux is:
```
PermissionError: [WinError 5] Access is denied: 'C:\\Users\\congnh\\zephyrproject\\.west\\manifest-tmp' -> 'C:\\Users\\congnh\\zephyrproject\\zephyr'
```

@jaredwolff may be on to something with this comment:

> After lots of testing. I've narrowed it down to whether or not VSCode has the folder open. It appears to be scanning the files inside the .west folder. When west is ready to transfer the repo to it's final resting place, it's prevented from doing so because VSCode is indexing the files.

... if any of you are still experiencing this, please could you open a new ticket to discuss further? (or perhaps this issue should be renamed accordingly, and I should remove the 'Fixes' tag from my PR)

jaredwolff, 2023-08-03:
@attie-argentum the issue, in my case, seemed to be caused by the integrated Git instance of VSCode. When I check out and init a repository with [my plugin](https://github.com/circuitdojo/zephyr-tools) it was wreaking all types of havoc. My fix? I simply turn off the Git pane during init and turn it on afterwords. Problem solved. :)

attie-argentum, 2023-08-03:
Very true, though others have reported similar a `PermissionError` and claim no VSCode... 🙈

Perhaps this is just "_locking on Windows_", and it cannot be worked around - VSCode isn't the only thing that may have a peek while `west` is busy after all.

mbolivar-ampere, 2023-08-31:
I'm going to close this since it seems like many of the issues here have been diagnosed as environment related problems on Windows that are outside of west's control. If someone is still having an issue that they believe can be attributed to west and which matches the symptom in this issue description ('west init completes but leaves everything in zephyr/manifest-tmp'), feel free to reopen. Otherwise, please open new issues for separate problems to keep this tracker clean.

thewh1teagle, 2024-01-18:
I had this issue today too.
Looks like vscode indexed things related to git and that's locked the folders while west init tried to delete it.
A workround:
1. I closed vscode completly and opened cmd.exe in that path
2.
```console
rmdir /s .west\
rmdir /s nrf\manifest-tmp\
west init -m https://github.com/nrfconnect/sdk-nrf --mr v2.5.1
=== Initialized. Now run "west update" inside C:\files\projects\side_projects\setup-nrf
```

marc-hb, 2024-01-18:
This can't be "fixed" but it looks frequent enough to deserve a warning and recommended workaround in `west init -h`. Everyone please review #703 that I just submitted, thanks!

trond-snekvik, 2024-01-19:
Although it wouldn't fix this in all scenarios, abandoning the `manifest-tmp` directory mechanism would resolve this for a lot of users. If the manifest has a `self.path`, we'd still need to move the directory after cloning (which could result in the same issue), but I'd argue that most manifest repositories don't have this field. Notably, Zephyr itself doesn't.

I understand that `manifest-tmp` offers a cleaner recovery mechanism on repeated calls to `west init`, but it also randomly breaks the second step of the getting started procedure on the most popular OS. 

It can't be fixed for all scenarios, but it can be fixed for the most common one.

marc-hb, 2024-01-20:
> I understand that manifest-tmp offers a cleaner recovery mechanism on repeated calls to west init,

There is worse that "repeated calls": consecutive `west` commands believing that `west init` was successful when it wasn't. Practically all filesystems guarantee atomicity of renames. This makes rename an extremely useful tool.

BTW the core issue is not just with `west` and not just with `init`: git can repack and delete files too.

Most popular or not, users of a filesystem that does not support concurrency should learn to avoid concurrency. This is good practice and important education that's valid way beyond west and VS code. Maybe the newer ReFS is better? I tried to find information about it but the name is unfortunately unsearchable.

If anything, _VS Code_ and other IDEs should be aware of the limitations of the filesystem they're using and refrain from interfering with random files they don't own. Avoiding background use of files in `.git/` and `.west/` would be good. They could also pause when they don't have focus; it wouldn't be perfect but it would avoid some failures.

amil13, 2024-03-25:
> ~Looks like it's because I'm running `west` in a virtual environment. Without the virtual environment, works ok. With the path to the virtual environment set, it breaks.~
> 
> The plot thickens.
> 
> After **lots** of testing. I've narrowed it down to whether or not VSCode has the folder open. It appears to be scanning the files inside the `.west` folder. When `west` is ready to transfer the repo to it's final resting place, it's prevented from doing so because VSCode is indexing the files.
> 
> When I _close_ VSCode and run it in a terminal with same environment, it runs perfectly fine.
> 
> Hope this helps people that may come across the same problem!

So this helped me too. When I closed my VScode, the west init zephyrproject worked fine from inside the venv.

marc-hb, 2024-03-28:
Thanks for the confirmation @amil3. Please report this issue to the VSCode project. VScode should go easy on NTFS filesystems that do not support concurrency and it should not scan as aggressively as it does right now because this requires exclusive ownership. This is not specific to west, it could happen every time VScode reads while anything else writes at the same time.

marc-hb, 2024-03-28:
>  VScode should go easy on NTFS filesystems that do not support concurrency

https://en.wikipedia.org/wiki/ReFS supports copy-on-write so it's likely not affected. However it's missing many NTFS features which is probably why its availability is very limited.

timonsku, 2024-03-28:
Windows 11 now has support for what they call "Dev Drives" which uses ReFS. Maybe worth a note in the setup guide that this could be useful to use for a Zephyr installation/project on Windows.
https://learn.microsoft.com/en-us/windows/dev-drive/

marc-hb, 2024-03-28:
> Windows 11 now has support for what they call "Dev Drives" which uses ReFS.

Very interesting, thanks for sharing!

> Maybe worth a note in the setup guide that this could be useful to use for a Zephyr installation/project on Windows.

It very likely would be worth it, cc: @tejlmand , @carlescufi , @stephanosio ,...

However, from the narrow/selfish perspective of this very specific `west init` failure it would be fantastic if a couple of the Windows users who reported the issue and who can still reproduce could confirm that this "Dev Drive" solves it.

BettaSharma, 2024-10-17:
Trying a clean install of Zephyr per https://docs.zephyrproject.org/latest/develop/getting_started/index.html

But getting this: (Complete log, unedited. No Visual Studio running or anything but browsers, email)

C:\>cd %HOMEPATH%

C:\>python -m venv zephyrproject\.venv

C:\>zephyrproject\.venv\Scripts\activate.bat

(.venv) C:\>pip install west
Collecting west
  Downloading west-1.2.0-py3-none-any.whl.metadata (5.2 kB)
Collecting colorama (from west)
  Downloading colorama-0.4.6-py2.py3-none-any.whl.metadata (17 kB)
Collecting PyYAML>=5.1 (from west)
  Downloading PyYAML-6.0.2-cp311-cp311-win_amd64.whl.metadata (2.1 kB)
Collecting pykwalify (from west)
  Downloading pykwalify-1.8.0-py2.py3-none-any.whl.metadata (5.3 kB)
Requirement already satisfied: setuptools in c:\zephyrproject\.venv\lib\site-packages (from west) (65.5.0)
Collecting packaging (from west)
  Downloading packaging-24.1-py3-none-any.whl.metadata (3.2 kB)
Collecting docopt>=0.6.2 (from pykwalify->west)
  Downloading docopt-0.6.2.tar.gz (25 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting python-dateutil>=2.8.0 (from pykwalify->west)
  Downloading python_dateutil-2.9.0.post0-py2.py3-none-any.whl.metadata (8.4 kB)
Collecting ruamel.yaml>=0.16.0 (from pykwalify->west)
  Downloading ruamel.yaml-0.18.6-py3-none-any.whl.metadata (23 kB)
Collecting six>=1.5 (from python-dateutil>=2.8.0->pykwalify->west)
  Downloading six-1.16.0-py2.py3-none-any.whl.metadata (1.8 kB)
Collecting ruamel.yaml.clib>=0.2.7 (from ruamel.yaml>=0.16.0->pykwalify->west)
  Downloading ruamel.yaml.clib-0.2.8-cp311-cp311-win_amd64.whl.metadata (2.3 kB)
Downloading west-1.2.0-py3-none-any.whl (95 kB)
   ---------------------------------------- 96.0/96.0 kB ? eta 0:00:00
Downloading PyYAML-6.0.2-cp311-cp311-win_amd64.whl (161 kB)
   ---------------------------------------- 162.0/162.0 kB 9.5 MB/s eta 0:00:00
Downloading colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Downloading packaging-24.1-py3-none-any.whl (53 kB)
   ---------------------------------------- 54.0/54.0 kB 2.9 MB/s eta 0:00:00
Downloading pykwalify-1.8.0-py2.py3-none-any.whl (24 kB)
Downloading python_dateutil-2.9.0.post0-py2.py3-none-any.whl (229 kB)
   ---------------------------------------- 229.9/229.9 kB 7.1 MB/s eta 0:00:00
Downloading ruamel.yaml-0.18.6-py3-none-any.whl (117 kB)
   ---------------------------------------- 117.8/117.8 kB 7.2 MB/s eta 0:00:00
Downloading ruamel.yaml.clib-0.2.8-cp311-cp311-win_amd64.whl (118 kB)
   ---------------------------------------- 118.0/118.0 kB 6.7 MB/s eta 0:00:00
Downloading six-1.16.0-py2.py3-none-any.whl (11 kB)
Building wheels for collected packages: docopt
  Building wheel for docopt (pyproject.toml) ... done
  Created wheel for docopt: filename=docopt-0.6.2-py2.py3-none-any.whl size=13774 sha256=8a5bb1f659d5e0615d7bc1a9d8989247c8ae330049791c6aba87af941134c69c
  Stored in directory: c:\users\bsharma\appdata\local\pip\cache\wheels\1a\b0\8c\4b75c4116c31f83c8f9f047231251e13cc74481cca4a78a9ce
Successfully built docopt
Installing collected packages: docopt, six, ruamel.yaml.clib, PyYAML, packaging, colorama, ruamel.yaml, python-dateutil, pykwalify, west
Successfully installed PyYAML-6.0.2 colorama-0.4.6 docopt-0.6.2 packaging-24.1 pykwalify-1.8.0 python-dateutil-2.9.0.post0 ruamel.yaml-0.18.6 ruamel.yaml.clib-0.2.8 six-1.16.0 west-1.2.0

[notice] A new release of pip is available: 24.0 -> 24.2
[notice] To update, run: python.exe -m pip install --upgrade pip

(.venv) C:\>west init zephyrproject
=== Initializing in C:\zephyrproject
--- Cloning manifest repository from https://github.com/zephyrproject-rtos/zephyr
Cloning into 'C:\zephyrproject\.west\manifest-tmp'...
remote: Enumerating objects: 1136079, done.
remote: Counting objects: 100% (932/932), done.
remote: Compressing objects: 100% (464/464), done.
remote: Total 1136079 (delta 535), reused 617 (delta 450), pack-reused 1135147 (from 1)
Receiving objects: 100% (1136079/1136079), 670.97 MiB | 6.14 MiB/s, done.
[... 24 lines cut ...]
    topdir = self.bootstrap(args)
             ^^^^^^^^^^^^^^^^^^^^
  File "C:\zephyrproject\.venv\Lib\site-packages\west\app\project.py", line 339, in bootstrap
    shutil.move(os.fspath(tempdir), os.fspath(manifest_abspath))
  File "C:\Python311\Lib\shutil.py", line 871, in move
    rmtree(src)
  File "C:\Python311\Lib\shutil.py", line 787, in rmtree
    return _rmtree_unsafe(path, onerror)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\shutil.py", line 629, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python311\Lib\shutil.py", line 629, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python311\Lib\shutil.py", line 629, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "C:\Python311\Lib\shutil.py", line 634, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Python311\Lib\shutil.py", line 632, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 5] Access is denied: 'C:\\zephyrproject\\.west\\manifest-tmp\\.git\\objects\\pack\\pack-c6e0d6ac99c8c9d17618c5b443b295412f5897dd.idx'

(.venv) C:\>
---------------------------------------------------------------------------------------------------------------------

I think I am stuck now as repeating the command reproduces the error.

timonsku, 2024-10-17:
@BettaSharma you are having a permission error accessing the root of the C: drive. You either need to run this from an Administrator shell or install to a different directory on your computer that does not require elevated permissions to write to.

BettaSharma, 2024-10-17:
@timonsku Thank you for the suggestion. I tried as admin, still get the same behavior. Will try on another directory as suggested.

marc-hb, 2024-10-17:
> you are having a permission error accessing the root of the C: drive.

I don't think this is the problem because `C:\zephyrproject\` would not have been created in the first place otherwise. Things would fail much earlier.

```
File "C:\Python311\Lib\shutil.py", line 853, in move
os.rename(src, real_dst)
PermissionError: [WinError 5] Access is denied: 'C:\zephyrproject\.west\manifest-tmp' -> 'C:\zephyrproject\zephyr'`
```

> No Visual Studio running or anything but browsers, email

It really looks like something concurrent is holding a reference at the same time because this looks exactly like the (solved) issues with VS above... some unusual antivirus maybe? An unusual one because renaming directories would fail for all Windows users all the time otherwise...

As already mentioned briefly before, a workaround:
```
cd zephyrproject
git clone zephyr
west init -l zephyr
```

BettaSharma, 2024-10-17:
@marc-hb, you may be right. Trying in an alternate path did not help. This is what I get:

Updating files: 100% (38630/38630), done.
Traceback (most recent call last):
  File "C:\Python311\Lib\shutil.py", line 853, in move
    os.rename(src, real_dst)
PermissionError: [WinError 5] Access is denied: 'C:\\Workspace\\zephyrproject\\.west\\manifest-tmp' -> 'C:\\Workspace\\zephyrproject\\zephyr' 

---------------------------------------------------

I will try the workaround. Thank You.

marc-hb, 2024-10-17:
> some unusual antivirus maybe? 

Or some backup program? Any background scanner really.

Also, could you please make a backup of the following file, make the following change and try again:

C:\zephyrproject.venv\Lib\site-packages\west\app\project.py

```diff
@@ -339,6 +339,10 @@ finalize the deletion until there is no concurrent user left.

         manifest_abspath = topdir / manifest_path

+        self.wrn("wait 5 minutes for any file scanner to complete")
+        import time
+        time.sleep(5 * 60)
+
         self.dbg('moving', tempdir, 'to', manifest_abspath,
                  level=Verbosity.DBG_EXTREME)

```

[18 more replies]

closed 2024-10-18 by marc-hb