zephyrproject-rtos/west #559: Selecting commit hash in west init is broken

Mucka, 2021-11-17:
Until now i have used: `west init -m git@remote --mr 756a3401d6df97604e0ac28da8cf5e76cc755989` to select certain commit while initialising project, but after i bumped to `0.12.0` this stopped working. Last version where this works properly is `0.11.0`. The error is probably related to `git init` -> `git clone` transition in `0.11.1` but this is my guess. How should I initialise project with certain commit hash now?

```
=== Initializing in /temporary/west-temp
--- Cloning manifest repository from git@remote, rev. 756a3401d6df97604e0ac28da8cf5e76cc755989
Cloning into '/temporary/west-temp/.west/manifest-tmp'...
fatal: Remote branch 756a3401d6df97604e0ac28da8cf5e76cc755989 not found in upstream origin
FATAL ERROR: command exited with status 128: git clone --branch 756a3401d6df97604e0ac28da8cf5e76cc755989 git@remote /temporary/west-temp/.west/manifest-tmp
```

marc-hb, 2021-11-18:
Looks like a duplicate of (wontfix) #531

mbolivar-nordic, 2021-11-18:
> Looks like a duplicate of (wontfix) #531

Correct

closed 2021-11-18 by mbolivar-nordic