zephyrproject-rtos/west #560: ZEPHYR_MODULES Is Being Ignored - Ref Sdk-Nrf

vindicatorr, 2021-12-13:
```
Configuration error:
There is a programmable error in your configuration file:

Traceback (most recent call last):
  File "/home/userName/.local/lib/python3.9/site-packages/sphinx/config.py", line 329, in eval_config_file
    exec(code, namespace)
  File "/<pathToTopDir>/ARM/Nordic/sdk-nrf/doc/nrfx/conf.py", line 24, in <module>
    conf = eval_config_file(str(NRFX_BASE / "doc" / "sphinx" / "conf.py"), tags)
  File "/home/userName/.local/lib/python3.9/site-packages/sphinx/config.py", line 324, in eval_config_file
    with cd(path.dirname(filename)):
  File "/usr/lib/python3.9/contextlib.py", line 119, in __enter__
    return next(self.gen)
  File "/home/userName/.local/lib/python3.9/site-packages/sphinx/util/osutil.py", line 168, in cd
    os.chdir(target_dir)
FileNotFoundError: [Errno 2] No such file or directory: '/<pathToTopDir>/modules/hal/nordic/nrfx/doc/sphinx'
```

I'm not sure if the issue belongs in the purview of west or sdk-nrf (https://devzone.nordicsemi.com/f/nordic-q-a/82668/accept-zephyr_modules-for-building-docs).
My intent was to build the docs in sdk-nrf (which builds seemingly a lot of (all?) projects docs, including zephyr).

A couple of things you may not care for:
First, I'm doing whatever I can to avoid west when I have more control with the underlying tool cmake.
Second, my project/file structure isn't going with the typical `<topDir>/modules`, but rather:
```
...
/.../sources/
  ARM/
    Nordic/
      hal_nordic/
      sdk-nrf/
  RISCV/
  zephyrProject
    ARM/ //ln -s from directory above
    RISCV //ln -s from directory above
    zephyr
    //no "modules" dir
...
```
Unfortunately, it appears `sdk-nrf` requires the use of west.

For ZEPHYR_MODULES, I've attempted to:
`$ export ZEPHYR_MODULES="/<pathToTopDir>/ARM/Nordic/hal_nordic/;/<pathToTopDir>/mcuboot/"`
`$ echo 'set(ZEPHYR_MODULES /<pathToTopDir>/mcuboot/ /<pathToTopDir>/ARM/Nordic/hal_nordic/
    CACHE STRING "pre-cached modules")' | tee /<pathTo>/ARM/Nordic/zephyr-modules.cmake`
with `cmake -C /<pathTo>/ARM/Nordic/zephyr-modules.cmake ...`
Editing `<pathToTopDir>/zephyr/west.yml` and changing the path for "hal_nordic".

With every attempt I've made, I still got the error message showing the "hal_nordic" path never reflected my changes.

After a lot of "print" statements, I narrowed it down to `manifest.py` `_manifest_content_at` where I see the actual `.git` blobs are searched through, so no change in 'west.yml' was going to be read.

Should west be caring about ZEPHYR_MODULES? Should it care about changes made to west.yml that aren't committed?
It was interesting to see some aspects of west recognized my changes in west.yml like `west list`, yet that manifest function does not.

marc-hb, 2021-12-13:
> I'm not sure if the issue belongs in the purview of west or sdk-nrf

The word "west" is nowhere to be found in your stack trace and there isn't any west command in the entire description either.

marc-hb, 2021-12-13:
> Should west be caring about ZEPHYR_MODULES?

From https://docs.zephyrproject.org/latest/guides/modules.html#integrate-modules-in-zephyr-build-system

> If you have west installed, you don’t need to worry about how this variable is defined unless you are adding a new module. The build system knows how to use west to set ZEPHYR_MODULES. You can add additional modules to this list by setting the ZEPHYR_EXTRA_MODULES CMake variable or by adding a ZEPHYR_EXTRA_MODULES line to .zephyrrc (See the section on Environment Variables for more details). This can be useful if you want to keep the list of modules found with west and also add your own.

> Should it care about changes made to west.yml that aren't committed?
It was interesting to see some aspects of west recognized my changes in west.yml like west list, yet that manifest function does not.

Are you using imports? It sounds like you are. From https://docs.zephyrproject.org/latest/guides/west/manifest.html#example-1-2-rolling-release-zephyr-downstream
> It’s also important to understand that west ignores your working tree’s zephyr/west.yml entirely when resolving imports. West always uses the contents of imported manifests as they were committed to the latest manifest-rev when importing from a project.

closed 2021-12-13 by vindicatorr