zephyrproject-rtos/west #561: west update crashes on `west update`

JohnVillalovos, 2021-12-21:
The `west.yml` file:
```
$ cat west.yml
manifest:
  remotes:
    - name: github
      url-base: https://github.com

  projects:
    - name: acpica
      remote: github
      repo-path: acpica/acpica
      path: tools/acpica
      revision: refs/tags/R01_05_21
      clone-depth: 1
```

Version and initialize
```
$ west --version
West version: v0.12.0
$ west init -l
=== Initializing from existing manifest repository temp
--- Creating /home/jlvillal/testing/.west and local configuration file
=== Initialized. Now run "west update" inside /home/jlvillal/testing.
```

Where things go wrong:
```

$ west --verbose update
west.manifest: loading /home/jlvillal/testing/temp/west.yml
west.manifest: group-filter: unset
west.manifest: added project acpica path tools/acpica revision refs/tags/R01_05_21 from /home/jlvillal/testing/temp/west.yml
west.manifest: loaded /home/jlvillal/testing/temp/west.yml
=== updating acpica (tools/acpica):
--- acpica: initializing
west.manifest: running 'git init --initial-branch=init_placeholder /home/jlvillal/testing/tools/acpica' in /home/jlvillal/testing
Initialized empty Git repository in /home/jlvillal/testing/tools/acpica/.git/
west.manifest: "git init --initial-branch=init_placeholder /home/jlvillal/testing/tools/acpica" exit code: 0 stdout: None stderr: None
west.manifest: running 'git remote add -- github https://github.com/acpica/acpica' in /home/jlvillal/testing/tools/acpica
west.manifest: "git remote add -- github https://github.com/acpica/acpica" exit code: 0 stdout: None stderr: None
west.manifest: running 'git cat-file -t refs/tags/R01_05_21' in /home/jlvillal/testing/tools/acpica
west.manifest: "git cat-file -t refs/tags/R01_05_21" exit code: 128 stdout: b'' stderr: b'fatal: Not a valid object name refs/tags/R01_05_21\n'
--- acpica: fetching, need revision refs/tags/R01_05_21
west.manifest: running 'git fetch -f --tags --depth 1 -- https://github.com/acpica/acpica refs/tags/R01_05_21' in /home/jlvillal/testing/tools/acpica
remote: Enumerating objects: 36897, done.
remote: Counting objects: 100% (36897/36897), done.
remote: Compressing objects: 100% (14091/14091), done.
remote: Total 36897 (delta 28345), reused 29660 (delta 22193), pack-reused 0
Receiving objects: 100% (36897/36897), 86.45 MiB | 14.42 MiB/s, done.
Resolving deltas: 100% (28345/28345), done.
From https://github.com/acpica/acpica
 * tag                 R01_05_21  -> FETCH_HEAD
 * [new tag]           HEAD       -> HEAD
 * [new tag]           R01_05_18  -> R01_05_18
 * [new tag]           R01_05_21  -> R01_05_21
 * [new tag]           R01_08_16  -> R01_08_16
 * [new tag]           R01_08_19  -> R01_08_19
 * [new tag]           R01_10_20  -> R01_10_20
 * [new tag]           R01_11_12  -> R01_11_12
 * [new tag]           R01_14_14  -> R01_14_14
 * [new tag]           R01_17_13  -> R01_17_13
 * [new tag]           R01_19_17  -> R01_19_17
 * [new tag]           R02_04_15  -> R02_04_15
 * [new tag]           R02_09_18  -> R02_09_18
 * [new tag]           R02_12_16  -> R02_12_16
 * [new tag]           R02_14_13  -> R02_14_13
 * [new tag]           R02_14_14  -> R02_14_14
 * [new tag]           R02_14_20  -> R02_14_20
[... 116 lines cut ...]
west.manifest: "git merge-base --is-ancestor 28cb42013541950cf378582a5a5a5587061498ca" exit code: 129 stdout: None stderr: None
Traceback (most recent call last):
  File "/home/jlvillal/.local/bin/west", line 8, in <module>
    sys.exit(main())
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/app/main.py", line 781, in main
    app.run(argv or sys.argv[1:])
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/app/main.py", line 106, in run
    self.run_command(argv)
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/app/main.py", line 336, in run_command
    cmd.run(args, unknown, self.topdir, manifest=self.manifest)
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/commands.py", line 140, in run
    self.do_run(args, unknown)
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/app/project.py", line 802, in do_run
    self.update_all()
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/app/project.py", line 869, in update_all
    self.update(project)
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/app/project.py", line 1048, in update
    current_branch, is_ancestor, try_rebase = self.decide_update_strategy(
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/app/project.py", line 1355, in decide_update_strategy
    is_ancestor = project.is_ancestor_of(sha, current_branch)
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/manifest.py", line 876, in is_ancestor_of
    raise RuntimeError(f'unexpected git merge-base result {rc}')
RuntimeError: unexpected git merge-base result 129

```

JohnVillalovos, 2021-12-21:
As a note I have also tried the `west.yml` file without the `clone-depth: 1` value and same issue.

marc-hb, 2021-12-21:
Can you please try this?
```diff
-      revision: refs/tags/R01_05_21
+      revision: R01_05_21
```

Thanks for the verbose trace, this can be very useful but what is the terser error message first?

PS: "crash" sounds a bit extreme and a bit "clickbaity" (which seems to have worked!)

JohnVillalovos, 2021-12-21:
> Can you please try this?
> 
> ```diff
> -      revision: refs/tags/R01_05_21
> +      revision: R01_05_21
> ```

```
$ cat west.yml
manifest:
  remotes:
    - name: github
      url-base: https://github.com

  projects:
    - name: acpica
      remote: github
      repo-path: acpica/acpica
      path: tools/acpica
      revision: R01_05_21
      # revision: refs/tags/R01_05_21
      clone-depth: 1
```

```
$ west update
=== updating acpica (tools/acpica):
--- acpica: initializing
Initialized empty Git repository in /home/jlvillal/testing/tools/acpica/.git/
--- acpica: fetching, need revision R01_05_21
remote: Enumerating objects: 36897, done.
remote: Counting objects: 100% (36897/36897), done.
remote: Compressing objects: 100% (14091/14091), done.
remote: Total 36897 (delta 28345), reused 29660 (delta 22193), pack-reused 0
Receiving objects: 100% (36897/36897), 86.45 MiB | 9.35 MiB/s, done.
Resolving deltas: 100% (28345/28345), done.
From https://github.com/acpica/acpica
 * tag                 R01_05_21  -> FETCH_HEAD
 * [new tag]           HEAD       -> HEAD
 * [new tag]           R01_05_18  -> R01_05_18
 * [new tag]           R01_05_21  -> R01_05_21
 * [new tag]           R01_08_16  -> R01_08_16
 * [new tag]           R01_08_19  -> R01_08_19
 * [new tag]           R01_10_20  -> R01_10_20
 * [new tag]           R01_11_12  -> R01_11_12
 * [new tag]           R01_14_14  -> R01_14_14
 * [new tag]           R01_17_13  -> R01_17_13
 * [new tag]           R01_19_17  -> R01_19_17
 * [new tag]           R02_04_15  -> R02_04_15
 * [new tag]           R02_09_18  -> R02_09_18
 * [new tag]           R02_12_16  -> R02_12_16
 * [new tag]           R02_14_13  -> R02_14_13
 * [new tag]           R02_14_14  -> R02_14_14
 * [new tag]           R02_14_20  -> R02_14_20
 * [new tag]           R02_15_12  -> R02_15_12
 * [new tag]           R02_15_19  -> R02_15_19
 * [new tag]           R02_24_17  -> R02_24_17
 * [new tag]           R03_03_17  -> R03_03_17
 * [new tag]           R03_13_18  -> R03_13_18
 * [new tag]           R03_18_16  -> R03_18_16
 * [new tag]           R03_20_12  -> R03_20_12
 * [new tag]           R03_25_14  -> R03_25_14
 * [new tag]           R03_26_20  -> R03_26_20
 * [new tag]           R03_28_13  -> R03_28_13
 * [new tag]           R03_29_19  -> R03_29_19
 * [new tag]           R03_31_21  -> R03_31_21
 * [new tag]           R04_05_19  -> R04_05_19
 * [new tag]           R04_08_15  -> R04_08_15
 * [new tag]           R04_18_13  -> R04_18_13
 * [new tag]           R04_20_12  -> R04_20_12
[... 86 lines cut ...]
    sys.exit(main())
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/app/main.py", line 781, in main
    app.run(argv or sys.argv[1:])
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/app/main.py", line 106, in run
    self.run_command(argv)
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/app/main.py", line 336, in run_command
    cmd.run(args, unknown, self.topdir, manifest=self.manifest)
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/commands.py", line 140, in run
    self.do_run(args, unknown)
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/app/project.py", line 802, in do_run
    self.update_all()
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/app/project.py", line 869, in update_all
    self.update(project)
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/app/project.py", line 1048, in update
    current_branch, is_ancestor, try_rebase = self.decide_update_strategy(
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/app/project.py", line 1355, in decide_update_strategy
    is_ancestor = project.is_ancestor_of(sha, current_branch)
  File "/home/jlvillal/.local/lib/python3.10/site-packages/west/manifest.py", line 876, in is_ancestor_of
    raise RuntimeError(f'unexpected git merge-base result {rc}')
RuntimeError: unexpected git merge-base result 129
```

> Thanks for the verbose trace, this can be very useful but what is the terser error message first?

Ran it this time without verbose

JohnVillalovos, 2021-12-21:
Think the issue maybe this: https://github.com/acpica/acpica/issues/739

marc-hb, 2021-12-21:
That acpica git repo is broken. @acpibob created a tag named `HEAD` 4 days ago: https://github.com/acpica/acpica/releases/tag/HEAD

```
git clone https://github.com/acpica/acpica/
git rev-parse HEAD
warning: refname 'HEAD' is ambiguous.
5c20304bc45b7954e1a4f7e536c167b901838cfb
```

JohnVillalovos, 2021-12-21:
I'm going to close this. Feel free to re-open if someone thinks west should work around repositories that use `HEAD` as a tag name.

closed 2021-12-21 by JohnVillalovos

acpibob, 2022-01-03:
Well, I can delete the tag if someone can tell me how to do it.
Thx,
Bob

marc-hb, 2022-01-03:
Thanks! Looks like it's gone now: https://github.com/acpica/acpica/releases/tag/HEAD => not found.

Make sure it's also removed from any local clone you have. Because tags are supposed to be immutable; they propagate very (too?) quickly and easily; think Hydra heads...

PR #562 (2022-01-21): Raise RuntimeError() if can't find rev for HEAD
If a project has created a tag or branch with the name HEAD (which
they really shouldn't do) it will cause the command:
'git rev-parse --abbrev-ref HEAD' to fail to provide a ref.

Check if 'git rev-parse --abbrev-ref HEAD' fails to return a value and
if so do a `log.die()` rather than failing at the 'git merge-base'
stage with a less useful error.

Closes #561
commit 1f21203f (2022-01-21): Raise RuntimeError() if can't find rev for HEAD

If a project has created a tag or branch with the name HEAD (which
they really shouldn't do) it will cause the command:
'git rev-parse --abbrev-ref HEAD' to fail to provide a ref.

Check if 'git rev-parse --abbrev-ref HEAD' fails to return a value and
if so do a `log.die()` rather than failing at the 'git merge-base'
stage with a less useful error.

Closes #561