zephyrproject-rtos/west #564: [FEATURE] Support flash with full-chip erase

jefffhaynes, 2022-01-11:
It would be great to be able to configure the flash command to do a whole chip erase rather than partial.  This would be helpful in cases where one wants to erase bonds, settings, local storage, etc. and is a much faster option.

marc-hb, 2022-01-11:
I'm not familiar with this topic but this looks like a runner/flasher specific question, not a generic west question. West is just a wrapper.

https://docs.zephyrproject.org/latest/guides/west/build-flash-debug.html#flash-and-debug-runners

> The flash and debug commands use Python wrappers around various Flash & Debug Host Tools. These wrappers are all defined in a Python library at scripts/west_commands/runners. Each wrapper is called a runner. Runners can flash and/or debug Zephyr programs.

I don't know which runner you're requesting this for but there are none in the repo of west itself. There are runners in https://github.com/zephyrproject-rtos/zephyr/tree/main/scripts/west_commands/runners so maybe you should file a (more specific) issue there?

tejlmand, 2022-01-11:
> It would be great to be able to configure the flash command to do a whole chip erase rather than partial.

Did you check the runner you are using ?

For example the `nrfjprog` runner already supports this:
```
$ west flash --context -r nrfjprog
...
nrfjprog options:
...
  --erase, --no-erase   mass erase flash before loading, or don't
```

so one can do `west flash --erase` when using the `nrfjprog` runner, which supports nRF kits.

So check the options for the specific runner you are using.

jefffhaynes, 2022-01-11:
ah, gotcha.  looks like it might already be supported!

closed 2022-01-11 by jefffhaynes