zephyrproject-rtos/west #573: west init FATAL ERROR: command exited with status 128

ljnash, 2022-03-21:
I'm brand new to zephyr (and west) and I've been trying to follow the [Getting Started Guide](https://docs.zephyrproject.org/latest/getting_started/index.html) using Ubuntu 20.04.4 LTS but when I get to the step `west init ~/zephyrproject` I receive the following failure:

```
$ west init ~/zephyrproject
=== Initializing in /home/user/zephyrproject
--- Cloning manifest repository from https://github.com/zephyrproject-rtos/zephyr
Cloning into '/home/user/zephyrproject/.west/manifest-tmp'...
remote: Enumerating objects: 678217, done.
remote: Counting objects: 100% (1332/1332), done.
remote: Compressing objects: 100% (618/618), done.
error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
FATAL ERROR: command exited with status 128: git clone https://github.com/zephyrproject-rtos/zephyr /home/user/zephyrproject/.west/manifest-tmp
```
The failure is occurring when "Receiving objects" at around 200 MiB.

```
Receiving objects:  30% (203494/678217), 206.43 MiB | 8.14 MiB/s
```

I have a fresh installation of Ubuntu (I've tried reinstalling the entire OS just in case) and have followed the guide linked above, with the following additions that I found necessary to avoid errors.  As I wanted to follow the "Install within a Virtual Environment" route I needed to `$ sudo apt install python3.8-venv`.  Once in the virtual environment I needed to `$ pip install wheel`.

I have the following versions of software installed.

```
$ west --version
West version: v0.12.0
$ cmake --version
cmake version 3.22.2
$ python3 --version
Python 3.8.10
$ dtc --version
Version: DTC 1.5.0
```

I'm confused why the "Getting Started Guide" doesn't seem to be working on a completely vanilla Ubuntu install.  Please help.

mbolivar-nordic, 2022-03-21:
Hi there, the issues you are reporting are from git itself, not west. I'm not sure why it's not working in your environment, but it looks like a networking error.

Can you clone the zephyr repository manually?

```
git clone https://github.com/zephyrproject-rtos/zephyr
```

ljnash, 2022-03-22:
Thank you for your reply.  It looks like running the git command alone produces the same error?

```
$ git clone https://github.com/zephyrproject-rtos/zephyr
Cloning into 'zephyr'...
remote: Enumerating objects: 678315, done.
remote: Counting objects: 100% (1397/1397), done.
remote: Compressing objects: 100% (715/715), done.
error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
```
```
$ git --version
git version 2.25.1
```

I hadn't expected the issue to be git itself, do you have any suggestions where I can seek further help?  As this is a vanilla Ubuntu LTS release I'm surprised to be having such an issue.

mbolivar-nordic, 2022-03-25:
I'm sorry, but I don't have any suggestions beyond getting everything up to date on your operating system and checking if you have any network problems. Since this is not a west issue, I'm closing.

closed 2022-03-25 by mbolivar-nordic

Pondidum, 2023-09-24:
I have also been seeing this error when building inside a docker container:
```
FROM zmkfirmware/zmk-build-arm:3.2-branch
WORKDIR /west

COPY config/west.yml west.yml
RUN west init -l "/west/config"
RUN west update
```

With the following `west.yml` file:

```yaml
manifest:
  remotes:
    - name: zmkfirmware
      url-base: https://github.com/zmkfirmware
  projects:
    - name: zmk
      remote: zmkfirmware
      revision: main
      import: app/west.yml
  self:
    path: config
```

The fix was changing `west update` to `west update --narrow` as for some reason fetching with `--tags` was causing the error.

marc-hb, 2023-09-25:
@Pondidum you may have seen the same error code but I doubt you're actually experiencing the same error because I don't see how some git tags/references can cause a TLS errors while others not.

In any case the very first step is as above to test with git alone. If git alone fails the same, then west is totally irrelevant and this is the wrong issue tracker.