zephyrproject-rtos/west #576: RuntimeError: local configuration file not found

WangQi14, 2022-04-18:
When I use the order to do something like "west init", an error is showed: "RuntimeError: local configuration file not found". The os is ubuntu20.04 and the version of python3 is 3.8.10 and only one in the os. So are some configuration files missing from the os?

File "/home/_user/.local/lib/python3.8/site-packages/west/configuration.py", line 229, in _whence
    raise RuntimeError('local configuration file not found')

marc-hb, 2022-04-19:
> something like "west init"

Please provide the exact command the complete backtrace.

WangQi14, 2022-04-20:
Step 1: Install west

`pip3 install west`

Collecting west
  Using cached west-0.13.0-py3-none-any.whl (88 kB)
Requirement already satisfied: colorama in /usr/lib/python3/dist-packages (from west) (0.4.3)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from west) (45.2.0)
Requirement already satisfied: packaging in /home/_user/.local/lib/python3.8/site-packages (from west) (21.3)
Requirement already satisfied: pykwalify in /home/_user/.local/lib/python3.8/site-packages (from west) (1.8.0)
Requirement already satisfied: PyYAML>=5.1 in /usr/lib/python3/dist-packages (from west) (5.3.1)
Requirement already satisfied: pyparsing!=3.0.5,>=2.0.2 in /home/_user/.local/lib/python3.8/site-packages (from packaging->west) (3.0.8)
Requirement already satisfied: ruamel.yaml>=0.16.0 in /home/_user/.local/lib/python3.8/site-packages (from pykwalify->west) (0.17.21)
Requirement already satisfied: python-dateutil>=2.8.0 in /home/_user/.local/lib/python3.8/site-packages (from pykwalify->west) (2.8.2)
Requirement already satisfied: docopt>=0.6.2 in /home/_user/.local/lib/python3.8/site-packages (from pykwalify->west) (0.6.2)
Requirement already satisfied: ruamel.yaml.clib>=0.2.6; platform_python_implementation == "CPython" and python_version < "3.11" in /home/_user/.local/lib/python3.8/site-packages (from ruamel.yaml>=0.16.0->pykwalify->west) (0.2.6)
Requirement already satisfied: six>=1.5 in /usr/lib/python3/dist-packages (from python-dateutil>=2.8.0->pykwalify->west) (1.14.0)
Installing collected packages: west
Successfully installed west-0.13.0

Step 2: Use west

`west init`

Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/_user/.local/lib/python3.8/site-packages/west/__main__.py", line 3, in <module>
    main()
  File "/home/_user/.local/lib/python3.8/site-packages/west/app/main.py", line 786, in main
    app.run(argv or sys.argv[1:])
  File "/home/_user/.local/lib/python3.8/site-packages/west/app/main.py", line 101, in run
    self.load_manifest()
  File "/home/_user/.local/lib/python3.8/site-packages/west/app/main.py", line 119, in load_manifest
    self.manifest = Manifest.from_topdir(topdir=self.topdir,
  File "/home/_user/.local/lib/python3.8/site-packages/west/manifest.py", line 1105, in from_topdir
    return Manifest(topdir=topdir, config=config,
  File "/home/_user/.local/lib/python3.8/site-packages/west/manifest.py", line 1370, in __init__
    self._ctx = self._top_level_init(source_data,
  File "/home/_user/.local/lib/python3.8/site-packages/west/manifest.py", line 1712, in _top_level_init
    manifest_path_option = get_option('manifest.path')
  File "/home/_user/.local/lib/python3.8/site-packages/west/manifest.py", line 1709, in get_option
    return config.get(option, default=default,
  File "/home/_user/.local/lib/python3.8/site-packages/west/configuration.py", line 169, in get
    return self._get(lambda cf: cf.get(option), default, configfile)
  File "/home/_user/.local/lib/python3.8/site-packages/west/configuration.py", line 208, in _get
    for cf in self._whence(configfile):
  File "/home/_user/.local/lib/python3.8/site-packages/west/configuration.py", line 229, in _whence
    raise RuntimeError('local configuration file not found')
RuntimeError: local configuration file not found

 Also, to do west/west update/west --version/python3 -m west --version or other orders  would get the same error.

> > something like "west init"
> 
> Please provide the exact command the complete backtrace.

marc-hb, 2022-04-20:
I've seen a number of manifest changes recently, @mbolivar-nordic could this be caused or triggered by one of them?

@WangQi14 can you please try to reproduce after this?
```
pip3 uninstall west
pip3 install west==0.12.0
```

WangQi14, 2022-04-20:
> I've seen a number of manifest changes recently, @mbolivar-nordic could this be caused or triggered by one of them?
> 
> @WangQi14 can you please try to reproduce after this?
> 
> ```
> pip3 uninstall west
> pip3 install west==0.12.0
> ```

Nice! The problem is sloved by the solution that you provided. Now I can use west to do anything. Thanks!
By the way, my another PC is installed python3.10, and no error returned when to use west. Maybe the error is related to the python version?

mbolivar-nordic, 2022-05-11:
@WangQi14 thank you very much for your detailed report.

I'm curious to know what happens if you try testing with this pull request: https://github.com/zephyrproject-rtos/west/pull/581

Does it solve your issue too? Or is the error message at least more descriptive? Thanks for your help!

eugene-nikolaev, 2022-05-26:
Hi!
I've just got the same on after I've installed the west on Linux Mint 20.3:

```
pip3 install --user U west
export PATH=~/.local/bin:"$PATH"' >> ~/.bashrc
source ~/.bashrc 
```

It was able to show the usage hints after typing just `west`.
And then I've tried to install the sdk:

`west init -m https://github.com/nrfconnect/sdk-nrf --mr v1.9.1`

and then I've terminated it by Ctrl-C as I've used wrong dir and then west suddenly has stopped to work and started to throw the same exception as above.

I've tried to uinstall and install with no effect but then I've just added a print into the west's code found that it tries to read `~/.west/config`. The `~/.west` dir was just empty. Maybe it was some race or kinda, but it seems my termination caused "west" to be in an inconsistent state. I've just removed the `~/.west` dir and unistalled the package and installed again and it worked.

Not sure if uninstall and install step was really necessary.

Best regards,
Eugen

mbolivar-nordic, 2022-07-20:
> I've tried to uinstall and install with no effect but then I've just added a print into the west's code found that it tries to read ~/.west/config. The ~/.west dir was just empty. Maybe it was some race condition or kinda, but it seems my termination caused "west" to be in an inconsistent state.

That is definitely an inconsistent state. You need a .west/config for west to do much of anything at all, because that's where west keeps track of where the manifest repository and file are located among other basic things.

I haven't heard anything about a95f530 as a fix or not, but I'm going to prepare a west v0.14.0 release as soon as practical with it included. I'm closing this for now as I believe the issue to be addressed. If anyone can show this issue persists on main, please reopen it.

closed 2022-07-20 by mbolivar-nordic