zephyrproject-rtos/west #577: feature:  west sync & west upload cmmand

altair-albert, 2022-04-19:
Anyone want this feature like google repo?

marc-hb, 2022-04-19:
For Github or Gerrit?

west supports extension, you could write one:
https://docs.zephyrproject.org/2.7.0/guides/west/extensions.html

BTW https://docs.zephyrproject.org/1.14.1/guides/west/why.html
> Existing tools were considered during west’s initial design and development. None were found suitable for Zephyr’s requirements. In particular, these were examined in detail:
> Google repo
> - Assumes Gerrit is used for code review
> - ...
> - ...

altair-albert, 2022-04-20:
I think both need.  or other git server.  because when I work on mulit project  I need smart push my projects.   when the manifest file updated, the command west update only downloads project code.I can write  extensions,   I tried before.  but I think it maybe have a good design.   I can give some help, if you want. -------- 原始邮件 --------发件人： Marc Herbert ***@***.***>日期： 2022年4月20日周三 02:01收件人： zephyrproject-rtos/west ***@***.***>抄送： Albert Lv ***@***.***>, Author ***@***.***>主    题： Re: [zephyrproject-rtos/west] feature:  west sync & west upload cmmand (Issue #577)
For Github or Gerrit?
west supports extension, you could write one:
https://docs.zephyrproject.org/2.7.0/guides/west/extensions.html
BTW https://docs.zephyrproject.org/1.14.1/guides/west/why.html

Existing tools were considered during west’s initial design and development. None were found suitable for Zephyr’s requirements. In particular, these were examined in detail:
Google repo
Assumes Gerrit is used for code review......

mbolivar-nordic, 2022-05-13:
The problem with 'west upload' is that west would need to know how each individual git server accepts uploads. I think this is impossible to do in general, though I'd be interested to see a design for it if you have one.

As for 'sync', I guess that would mean pulling the latest changes from the manifest repository's remote and then running west update, all in one step? How would that work?

- how would you configure the remote manifest repository? .west/config? the manifest repository's .git/config? somewhere else?
- should it be possible to configure the remote manifest repository and branch to pull from on a per-local-branch basis, or just have one global thing to pull from?
- what about workspaces initialized with `west init -l`?

altair-albert, 2022-09-21:
too many work need to do. :dog: 

this my scripts. https://gist.github.com/altair-albert/16c758626afda1fb50a13cad1491c4f2

maybe have another better design. :smile: