zephyrproject-rtos/west #580: west doesn't work with python 3.10

0xlee, 2022-05-06:
I have arch linux and have problems with python 3.10. And in arch linux, it's almost impossible to use python 3.9.

```
% west build -b nrf52840dongle_nrf52840
-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): /home/dongha/projects/third/zephyrproject-nrf-v1.9.1/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/dongha/projects/third/connectedhomeip/examples/lighting-app/nrfconnect
-- Zephyr version: 2.7.99 (/home/dongha/projects/third/zephyrproject-nrf-v1.9.1/zephyr), build: v2.7.99-ncs1-17-gc3208e7ff49d
CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3: Found unsuitable version "3.10.4", but required is
  exact version "3.9" (found /usr/bin/python3.10, found components:
  Interpreter)
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:592 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake/Modules/FindPython/Support.cmake:3181 (find_package_handle_standard_args)
  /usr/share/cmake/Modules/FindPython3.cmake:490 (include)
  /home/dongha/projects/third/zephyrproject-nrf-v1.9.1/zephyr/cmake/python.cmake:44 (find_package)
  /home/dongha/projects/third/zephyrproject-nrf-v1.9.1/zephyr/cmake/app/boilerplate.cmake:149 (include)
  /home/dongha/projects/third/zephyrproject-nrf-v1.9.1/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/dongha/projects/third/zephyrproject-nrf-v1.9.1/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:34 (find_package)
```

marc-hb, 2022-05-06:
As you can see this is not a west error, it's coming from CMake. Try this:
```
west -v build -b nrf52840dongle_nrf52840
```
This will print the `cmake` command at the very start. Delete the build directory and run that cmake command directory without involving `west` at all. Observe the same error and close this issue, thanks!

0xlee, 2022-05-06:
Oh. Sorry for not paying attention enough. Thanks!

closed 2022-05-06 by 0xlee