zephyrproject-rtos/west #584: west build always rebuilds if -DBOARD_ROOT is added

benediktibk, 2022-05-30:
The following works as expected, the second time west build is executed nothing has to be done:

```
huber@m42-minipc:~/Development/temp$ west build -d build/sample_test/ -p auto -b stm32h735g_disco zephyr/samples/hello_world/
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/huber/Development/temp/zephyr/samples/hello_world
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.4") found components: Interpreter 
-- Cache files will be written to: /home/huber/.cache/zephyr
-- Zephyr version: 3.1.0-rc1 (/home/huber/Development/temp/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: stm32h735g_disco
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.13.1 (/opt/zephyr-sdk-0.13.1_distcc)
-- Found dtc: /opt/zephyr-sdk-0.13.1_distcc/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.13.1 (/opt/zephyr-sdk-0.13.1_distcc)
-- Found BOARD.dts: /home/huber/Development/temp/zephyr/boards/arm/stm32h735g_disco/stm32h735g_disco.dts
-- Generated zephyr.dts: /home/huber/Development/temp/build/sample_test/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/huber/Development/temp/build/sample_test/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/huber/Development/temp/build/sample_test/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/huber/Development/temp/build/sample_test/zephyr/dts.cmake
/home/huber/Development/temp/build/sample_test/zephyr/zephyr.dts:494.7-520.5: Warning (simple_bus_reg): /soc/can: missing or empty reg/ranges property
Parsing /home/huber/Development/temp/zephyr/Kconfig
Loaded configuration '/home/huber/Development/temp/zephyr/boards/arm/stm32h735g_disco/stm32h735g_disco_defconfig'
Merged configuration '/home/huber/Development/temp/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/huber/Development/temp/build/sample_test/zephyr/.config'
Kconfig header saved to '/home/huber/Development/temp/build/sample_test/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk-0.13.1_distcc/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/huber/Development/temp/build/sample_test
-- west build: building application
[1/156] Generating include/generated/version.h
-- Zephyr version: 3.1.0-rc1 (/home/huber/Development/temp/zephyr), build: v3.1.0-rc1-60-g12442dd96b85
[146/156] Linking C executable zephyr/zephyr_pre0.elf

[150/156] Linking C executable zephyr/zephyr_pre1.elf

[156/156] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       15340 B         1 MB      1.46%
            SRAM:        4352 B       320 KB      1.33%
           SRAM1:          0 GB        16 KB      0.00%
           SRAM2:          0 GB        16 KB      0.00%
           SRAM4:          0 GB        16 KB      0.00%
            DTCM:          0 GB       128 KB      0.00%
            ITCM:          0 GB        64 KB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%
huber@m42-minipc:~/Development/temp$ west build -d build/sample_test/ -p auto -b stm32h735g_disco zephyr/samples/hello_world/
ninja: no work to do.
```
As soon as -DBOARD_ROOT is specified west builds everything from scratch, all the time:
```
huber@m42-minipc:~/Development/temp$ west build -d build/sample_test/ -p auto -b stm32h735g_disco zephyr/samples/hello_world/ -DBOARD_ROOT=../../
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/huber/Development/temp/zephyr/samples/hello_world
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.4") found components: Interpreter 
-- Cache files will be written to: /home/huber/.cache/zephyr
-- Zephyr version: 3.1.0-rc1 (/home/huber/Development/temp/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: stm32h735g_disco
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.13.1 (/opt/zephyr-sdk-0.13.1_distcc)
-- Found dtc: /opt/zephyr-sdk-0.13.1_distcc/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.13.1 (/opt/zephyr-sdk-0.13.1_distcc)
-- Found BOARD.dts: /home/huber/Development/temp/zephyr/boards/arm/stm32h735g_disco/stm32h735g_disco.dts
[... 55 lines cut ...]
-- Configuring done
-- Generating done
-- Build files have been written to: /home/huber/Development/temp/build/sample_test
-- west build: building application
[42/50] Linking C executable zephyr/zephyr_pre0.elf

[45/50] Linking C executable zephyr/zephyr_pre1.elf

[50/50] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       15340 B         1 MB      1.46%
            SRAM:        4352 B       320 KB      1.33%
           SRAM1:          0 GB        16 KB      0.00%
           SRAM2:          0 GB        16 KB      0.00%
           SRAM4:          0 GB        16 KB      0.00%
            DTCM:          0 GB       128 KB      0.00%
            ITCM:          0 GB        64 KB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%
```

As a workaround it is possible to specify the board root via CMakeLists.txt:

```
list(APPEND BOARD_ROOT "whatever_you_need")
```

marc-hb, 2022-05-31:
https://docs.zephyrproject.org/latest/develop/west/build-flash-debug.html#one-time-cmake-arguments

> To pass additional arguments to the CMake invocation performed by west build, pass them after a -- at the end of the command line.

> Important

> Passing additional CMake arguments like this forces west build to re-run CMake, even if a build system has already been generated. After using -- once to generate the build directory, use west build -d <build-dir> on subsequent runs to do incremental builds.

(previously discussed in https://github.com/zephyrproject-rtos/zephyr/pull/40431#pullrequestreview-808936807)

marc-hb, 2022-05-31:
CMake has two very distinct phases: 1. configuration time, 2. build time.

When you use CMake directly without an additional layer like west, you don't pass configuration options again at build time. `cmake --build` will not accept -Dstuff and I don't even know how you would pass them to `ninja` or `make`; you're simply not supposed to configure at build time (?) and no one even tries.

This 2-steps process is necessary but annoying and it is why so many projects add another layer of indirection on top of CMake and try to simplify / hide these 2 steps. `west` tries to hides these 2-steps too but this simplification has limitations; you just found one.

The limitation is: what is west supposed to do the second time you pass `-Dsomething` when the build directory already exists? It can't pass it to `cmake --build`, `ninja` or `make` cause none wants it. It could drop it on the floor but silently ignoring user input makes the most despicable user interfaces and what if you changed the value? So it does the safest thing: rebuild from scratch and pass the option to CMake. It should probably hint at why @mbolivar-nordic , @tejlmand , @SebastianBoe ?

benediktibk, 2022-05-31:
Thanks for pointing towards the manual, obviously I did not find this by myself. Hence, the bug is technically not a bug anymore as it is documented.

But as you already started thinking about it, maybe it would be a possibility to store the additional parameter in the build directory. Then it would be possible to compare them on a second execution and it could be detected if they have changed? Of course I am not sure if this would break something else.

mbolivar-nordic, 2022-06-06:
I am going to close this since it is neither a west issue nor a zephyr issue.

The request to save the cmake arguments and avoid rebuilding if they are unchanged has been raised many times before, but I don't have any plans to implement such a thing, as I think it would be fragile and not likely to work well in all cases. If you disagree and want to send a PR doing this to build.py in the zephyr repository, I would be happy to review it, though.

closed 2022-06-06 by mbolivar-nordic

marc-hb, 2022-06-06:
> The request to save the cmake arguments and avoid rebuilding if they are unchanged has been raised many times before,

Interesting, @mbolivar-nordic any links or even just search keywords and places? I mean Zephyr repo, west repo or both?

mbolivar-nordic, 2022-06-06:
This has been an FAQ on slack a bunch at least as far as I recall. I don't have receipts.