zephyrproject-rtos/west #585: Override local changes in manifest repository

Nano-HK, 2022-06-07:
Hi, is it possible to override local changes when running west update?

For instance these when running west update;
```
=== updating trusted-firmware-m (modules/tee/tfm):
HEAD is now at c74be389 trusted-firmware-m: platform: lpcxpresso55s69: Update SDK
M       trusted-firmware-m/bl2/ext/mcuboot/CMakeLists.txt
M       trusted-firmware-m/bl2/ext/mcuboot/bl2_main.c
M       trusted-firmware-m/bl2/ext/mcuboot/keys.c
M       trusted-firmware-m/bl2/ext/mcuboot/scripts/assemble.py
M       trusted-firmware-m/bl2/ext/mcuboot/scripts/wrapper/wrapper.py
M       trusted-firmware-m/platform/ext/target/nordic_nrf/nrf5340dk_nrf5340_cpuapp/RTE_Device.h
=== updating mbedtls (modules/crypto/mbedtls):
```

Best Regards, 
Nano

closed 2022-06-07 by mbolivar-nordic

marc-hb, 2022-06-07:
You may be interested in these:
- https://docs.zephyrproject.org/latest/develop/west/built-in.html#west-update
- https://github.com/zephyrproject-rtos/west/issues/381

Nano-HK, 2022-06-08:
Cheers guys! 

What do you think about having a west command that reset hard all modules? I think the `west forall -c 'git reset --hard HEAD` was neat, could it be sufficient to have a west command such as `west update reset all` that runs that underneath?

marc-hb, 2022-06-09:
Déjà vu?
- #337

mbolivar-nordic, 2022-06-24:
I have no plans to add those types of features myself, as I tend to work on west when I have a mandate to add a feature that's blocking me in some way. But the project is open to contributions.