zephyrproject-rtos/west #586: ManifestImportFailed object has no attribute 'filename'

metc, 2022-06-15:
Here is the error I get when running `west status` :

```
west status
Traceback (most recent call last):
  File "runpy.py", line 193, in _run_module_as_main
  File "runpy.py", line 86, in _run_code
  File "c:\ncs\toolchains\v2.0.0\opt\bin\Scripts\west.exe\__main__.py", line 7, in <module>
  File "c:\ncs\toolchains\v2.0.0\opt\bin\Lib\site-packages\west\app\main.py", line 786, in main
    app.run(argv or sys.argv[1:])
  File "c:\ncs\toolchains\v2.0.0\opt\bin\Lib\site-packages\west\app\main.py", line 109, in run
    self.run_command(argv)
  File "c:\ncs\toolchains\v2.0.0\opt\bin\Lib\site-packages\west\app\main.py", line 337, in run_command
    self.handle_builtin_manifest_load_err(args)
  File "c:\ncs\toolchains\v2.0.0\opt\bin\Lib\site-packages\west\app\main.py", line 195, in handle_builtin_manifest_load_err
    p, f = self.mle.project, self.mle.filename
AttributeError: 'ManifestImportFailed' object has no attribute 'filename'
```

Same issue when issuing the `west list` command for instance. I expected to get a friendly error message instead. I am using West version: v0.13.1.

marc-hb, 2022-06-15:
Maybe a duplicate of #579?

metc, 2022-06-20:
Please note that I have installed the toolchain using the nRF Connect Toolchain Manager (fully automated automation with default settings). I am running `west` from the toolchain folder using Git Bash on Windows.

```
MINGW64 /c/ncs/v2.0.0
$ which west
/opt/bin/Scripts/west
$ west --version
West version: v0.13.1
```

It is maybe related to file permissions and user id, but I am not using Docker here.

metc, 2022-06-20:
Same issue when running `west` from a freshly created Python env:

```sh
$ python --version
Python 3.10.5

$ which python
/c/Users/miaou/.pyenv/pyenv-win/versions/3.10.5/python

$ pip list --local
Package          Version
---------------- -------
colorama         0.4.5
distlib          0.3.4
docopt           0.6.2
filelock         3.7.1
packaging        21.3
pip              22.1.2
platformdirs     2.5.2
pykwalify        1.8.0
pyparsing        3.0.9
python-dateutil  2.8.2
PyYAML           6.0
ruamel.yaml      0.17.21
ruamel.yaml.clib 0.2.6
setuptools       58.1.0
six              1.16.0
virtualenv       20.14.1
west             0.13.1
```

The error I get:

```sh
$ west topdir
C:/ncs/dev/bkbx

$ pwd
/c/ncs/dev/bkbx

$ west config -l
manifest.path=app
manifest.file=west.yml
zephyr.base=ncs/zephyr

$ west list
Traceback (most recent call last):
  File "C:\Users\miaou\.pyenv\pyenv-win\versions\3.10.5\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\miaou\.pyenv\pyenv-win\versions\3.10.5\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\miaou\.pyenv\pyenv-win\versions\3.10.5\Scripts\west.exe\__main__.py", line 7, in <module>
  File "C:\Users\miaou\.pyenv\pyenv-win\versions\3.10.5\lib\site-packages\west\app\main.py", line 786, in main
    app.run(argv or sys.argv[1:])
  File "C:\Users\miaou\.pyenv\pyenv-win\versions\3.10.5\lib\site-packages\west\app\main.py", line 109, in run
    self.run_command(argv)
  File "C:\Users\miaou\.pyenv\pyenv-win\versions\3.10.5\lib\site-packages\west\app\main.py", line 337, in run_command
    self.handle_builtin_manifest_load_err(args)
  File "C:\Users\miaou\.pyenv\pyenv-win\versions\3.10.5\lib\site-packages\west\app\main.py", line 195, in handle_builtin_manifest_load_err
    p, f = self.mle.project, self.mle.filename
AttributeError: 'ManifestImportFailed' object has no attribute 'filename'
```

Can anyone help me?

marc-hb, 2022-06-22:
Forget Docker, it's apparently possible to configure (complex) Windows file permissions in a way that is incompatible with git.
- https://github.com/zephyrproject-rtos/west/issues/558#issuecomment-993781422

To be on the safe side can you please try this, see whether it fails:
```
cd C:\ncs\dev\bkbx
git init w558
cd w558
git remote add origin -- https://github.com/zephyrproject-rtos/zephyr.git
git fetch origin --tags -- zephyr-v2.7.0 'refs/heads/*:refs/remotes/origin/*'
del .git\objects\pack\*
```

metc, 2022-06-22:
Hello @marc-hb, thank you for your message.

I get an "Access is denied." error, same as in https://github.com/zephyrproject-rtos/west/issues/558#issuecomment-994298453.
I did this test using `cmd`, but I do use Git Bash.

marc-hb, 2022-06-22:
OK so you have a strange Windows file permission issue that makes your system incompatible with git - without `west` being involved at all. Maybe `west` could detect and report this better but there's nothing `west` can do to fix it.

If you find what this file permission issue please share it here, it will be useful for other users and it could also help `west` provide a better error message.

closed 2023-02-27 by mbolivar-nordic