zephyrproject-rtos/west #589: west init from a private repository

Machine-Hum, 2022-07-12:
Hey,

I've [forked this](https://github.com/zephyrproject-rtos/example-application) project and have moved it into a private repository. I'm now trying to clone it and am experience an issues with credentials.

```
[machinehum@whitebox projects]$ west init -m https://github.com/Machine-Hum/id01_trackpad_fw
=== Initializing in /home/machinehum/projects
--- Cloning manifest repository from https://github.com/Machine-Hum/id01_trackpad_fw
Cloning into '/home/machinehum/projects/.west/manifest-tmp'...
Username for 'https://github.com': machine-hum
Password for 'https://machine-hum@github.com': 
remote: Invalid username or password.
fatal: Authentication failed for 'https://github.com/Machine-Hum/id01_trackpad_fw/'
FATAL ERROR: command exited with status 128: git clone https://github.com/Machine-Hum/id01_trackpad_fw /home/machinehum/projects/.west/manifest-tmp
```
A normal git clone works find, as I've setup my keypairs and whatnot. Is there a way to get the same effect as west init -m by using a git clone?

marc-hb, 2022-07-12:
west does not know anything about git authentication.

Run `west -vv init -m https://github.com/Machine-Hum/id01_trackpad_fw --mr main id01_trackpad_fw`. This will fail but it will print a `git clone ...` command. Try to run that git clone command.

Machine-Hum, 2022-07-12:
I got the answer, when cloning you can't use the https://github.com - this tells git to use normal passwords. The command needs to be ` west init -m git@github.com

closed 2022-07-12 by Machine-Hum

marc-hb, 2022-07-13:
> you can't use the https://github.com/ - this tells git to use normal passwords

`.netrc` tokens work too, not just passwords `https://` only tells git to use... https://, the rest is up to libcurl and github, git does not know about it.

> git@github.com

For git this syntax means ssh (and then git still does not about authentication).