zephyrproject-rtos/west #593: Best known "de-initialization" method should be at least documented

marc-hb, 2022-07-25:
> The only way currently is to remove the .west directory. However, instead of hard-coding the location, you may wish to consider using this API to find it: https://docs.zephyrproject.org/latest/develop/west/west-apis.html#west.util.west_dir

> There's no explicit "de-" or "re-" initialization mechanism. We tried to design one during west's early days, and got into a bunch of internal arguments about what to do, then abandoned the idea until a compelling use case came along that would justify bringing it in again. Is this that time? If so, please file an issue.

_Originally posted by @mbolivar-nordic in https://github.com/thesofproject/sof/pull/6005#discussion_r929312490_

marc-hb, 2022-07-25:
I don't know how "compelling" it is but the use case in sof#6005 above is: switching from `zephyr/west.yml` to `sof/west.yml`. Ignoring the particular SOF history that led to that situation, I think anyone who wants to switch to work on a specific Zephyr module and track its active development without having to shuffle git repositories around their hard drive would end in the same "deinit" situation.

I can imagine switching from/to some "downstream" manifests is more or less the same requirement.

Anecdotal (e.g.: discord) evidence shows people currently achieve west "de-initialization" by simply deleting their `.west/` directory. Short of a (future and?) more abstract and more official `west deinit` command, this best-known-so-far `rm -rf .west/` method should be at least mentioned briefly in west's documentation.

mbolivar-nordic, 2022-08-19:
> switching from zephyr/west.yml to sof/west.yml

So is this a case where you're switching back and forth? Or permanently switching one time?

Switching back and forth is already supported. You reconfigure `manifest.path` (and, if necessary, `manifest.file`) and run `west update` again. I could be a weasel and claim that this is documented here:

https://docs.zephyrproject.org/latest/develop/west/config.html#west-config-cmd

which, while technically true, is definitely not in a place where anyone looking for the answer to this question would look for it, so we can move that somewhere else if it's helpful.

Permanently switching one time would require deleting all of the repositories in the workspace, though, so that's a bit of an `rm -rf zephyrprpoject`, not an `rm -rf zephyrproject/.west`.

marc-hb, 2022-08-19:
> So is this a case where you're switching back and forth? Or permanently switching one time?

In our case it was just once but I can imagine users wanting go switch back and forth - for instance between upstream and downstream?

> so we can move that somewhere else if it's helpful.

I think some better visibility would not hurt. A pointer to this `west config` section would be enough. Maybe here or here:
- https://docs.zephyrproject.org/latest/develop/west/manifest.html
- https://docs.zephyrproject.org/latest/search.html?q=west+manifest+change
- https://docs.zephyrproject.org/latest/search.html?q=west+manifest+switch

Funny enough the last search does find the "Configuration" page but the extract is not relevant and misleading.

> Permanently switching one time would require deleting all of the repositories in the workspace, though, so that's a bit of an rm -rf zephyrprpoject, not an rm -rf zephyrproject/.west.

I don't understand why you're making a difference between "back and forth" and "once". I believe the important difference is between "manifest switch" versus "de-initialization" (where the latter can provide an indirect solution for the former).

I also don't understand why de-initialization seems to require deleting all git repos. This is something we certainly did _not_ want to ask our users to do and everything went fine. So you're saying west does not automatically pick up git clones already there, automagically adding the missing `manifest-rev` branch? In my experience it does. Is this an "accidental" feature that could disappear in the future?

mbolivar-nordic, 2022-08-26:
> So you're saying west does not automatically pick up git clones already there, automagically adding the missing `manifest-rev` branch?

No, I'm saying I don't know anything about the sof manifest, and there are edge cases. E.g. if foo/bar/ is a git repository in zephyr/west.yml and sof defines foo/bar/baz as a project, west will happily stick baz inside of bar without removing bar.

mbolivar-nordic, 2022-08-26:
> I think some better visibility would not hurt. A pointer to this `west config` section would be enough. Maybe here or here:

I'll add this to the v0.14 docs pr I'm working on, thanks.

mbolivar-nordic, 2022-09-16:
@marc-hb I'm going to close this one since west will tell the user how to do this when circumstances require: https://github.com/zephyrproject-rtos/west/blob/ed01ab5fa1a918da3661733839be90d31ea94809/src/west/app/project.py#L192

I figured that would be better than documentation, since people are more likely to read error messages than docs. Please reopen if I didn't get your meaning.

closed 2022-09-16 by mbolivar-nordic