zephyrproject-rtos/west #594: West update returns error when submanifests directory does not exist

aborisovich, 2022-07-29:
According to documentation on [Manifest Import Details](https://docs.zephyrproject.org/latest/develop/west/manifest.html#manifest-import-details)

> Import keys are optional. If any of import-1, ..., import-N are missing, west will not import additional manifest data from that project.

As far as my understanding goes this behavior also works when `import` statement is inside `self`  statement.
Like so:
[image]

However, `west update` command raises following error when **submanifests** directory does not exist:
[image]

On the other hand, `west update` command does not rise any error when directory exists but does not have any files.

### Reproduction steps
1. In <enter_your_project_path> create west.yml manifest with following content:
[image]

2. Execute command `west init -l <enter_your_project_path>`
3. Execute command `west update`

### Actual result
`west update` command results in error (ignore my setup paths):
[image]

### Expected result
Command `west update` performs update of found projects.

### Work-around
In your <enter_your_project_path> create empty (or with some junk files) directory _submanifests_.

marc-hb, 2022-07-29:
Good catch

git has an "interesting" design where it does not track directories. git only creates directories "lazily" only when it has to create files in them.

As you mentioned the workaround is to git add `submanifests/.empty_placeholder`. Adding them to git makes sure no one experiences the issue.

mbolivar-nordic, 2022-08-17:
>> Import keys are optional. If any of import-1, ..., import-N are missing, west will not import additional manifest data from that project.

> As far as my understanding goes this behavior also works when import statement is inside self statement.

Yes, and this is explicitly spelled out in the same documentation:

> If self-import is missing, no additional files in the manifest repository (beyond the top-level file) are imported.

However, "missing" is not about the *file* being missing on the file system. It's about the *key* being missing in the YAML file.

> Like so:
> image

So this example is not related to the documentation. If you wrote:

```yaml
manifest:
  # ...
  self:
    # no import here!
```

Then the above documentation would apply. But since the `import: submanifests` value is present in your example, west goes to look for it, and reports an error when it fails to find it. This is expected behavior. I hope this clears this up.

closed 2022-08-17 by mbolivar-nordic