zephyrproject-rtos/west #595: west build doesn't process select in Kconfig for module correctly

jeremyherbert, 2022-08-19:
Hi,

(unsure if this is a zephyr or west bug, but I figured west is more likely to be the case)

I have a west manifest repository which checks out some custom modules from git repos which are then used in custom drivers I have written. The CMakeLists for these modules and Kconfig are pretty boilerplate, below is an example:

```
if(CONFIG_MY_LIB)
  zephyr_include_directories(
    ${ZEPHYR_MY_LIB_MODULE_DIR}/inc
  )

  zephyr_library()
  zephyr_library_sources(
    ${ZEPHYR_MY_LIB_MODULE_DIR}/src/my_lib.c
    ${ZEPHYR_MY_LIB_DEMUX_MODULE_DIR}/src/my_lib2.c
  )
endif()
```

```
config ZEPHYR_MY_LIB_MODULE
	bool

menuconfig MY_LIB
	bool "My library"
	help
	  A library that I have added myself
```

If I add `CONFIG_MY_LIB=y` to the `prj.conf` file, everything builds fine and works great (the module is linked in correctly). 

However, if I remove it from `prj.conf` and instead add it to my root Kconfig as a `select` statement which causes a driver to build:

```
config ENABLE_MY_DRIVER
    bool "extra special driver"
    select MY_LIB
```

and then add `CONFIG_ENABLE_MY_DRIVER=y` to `prj.conf`, the build doesn't succeed. It proceeds to the final linker step, but then says `undefined reference to `__device_dts_ord_14'` or similar, indicating that my driver has indeed not been compiled.

My driver is being compiled just by adding the files directly in `CMakeLists.txt`:

```
...

if(CONFIG_ENABLE_MY_DRIVER)
    target_include_directories(app PRIVATE
        ${CMAKE_CURRENT_LIST_DIR}/drivers/my_driver/inc
    )

    target_sources(app PRIVATE
        ${CMAKE_CURRENT_LIST_DIR}/drivers/my_driver/src/my_driver.c
    )
endif()

...
```

Is it expected behaviour that you cannot `select` a module?

I am using west v0.13.1 and all of my manifest/zephyr stuff is just the vanilla nordic sdk-nrf at revision v2.0.2.

marc-hb, 2022-08-19:
I asked about a similar situation before and never got a complete explanation of how Kconfig and modules interact:
- https://github.com/zephyrproject-rtos/zephyr/pull/38375

Have a look at 38375 look anyway, who may find useful information there.

> However, if I remove it from prj.conf and instead add it to my root Kconfig 

Do you mean Zephyr's root Kconfig? Then looks like a circular dependency to me, I mean: your module depends on Zephyr which depends on your module. Would also break Kconfig for people not using your module.

PS: I don't think this is a west issue. AFAIK west merely provides the list of modules to CMake.

marc-hb, 2022-08-19:
I remembered about this, could provide useful pointers too:
- https://github.com/zephyrproject-rtos/zephyr/pull/14667

Sorry for the noise if not.

jeremyherbert, 2022-08-20:
Thanks for the tips. I will try to reproduce in a separate manifest repo, and I will file a bug on the zephyr repo if I am able to do so.

marc-hb, 2022-08-21:
@jeremyherbert I can re-open and transfer this to the Zephyr repo if you prefer. Entirely up to you, just ask.

closed 2026-08-29 by marc-hb