zephyrproject-rtos/west #596: Add git-patch feature in west update

DonShogun, 2022-08-22:
In addition to revision, the configuration in west.yaml should also optionally takes patch and apply it after checkout on specified branch.

```
manifest:
  ...
  projects:
    ...
    - name: test
      revision: abcd12345dcba54321
      git-patch: path/to/test-001.patch
```
`west` should prepare all needed repositories in ready-to-build state. This should be limited to simple git commands e.g.: `git apply  path/to/test-001.patch`
This change is motivated by faster bug fixing without waiting for official releases, adding custom settings and inner-project integrations.

mbolivar-nordic, 2022-08-23:
Interesting idea; thanks for filing.

Some details that anyone who wants to work on this needs to consider:

- in my view, this would be step 1 towards reimplementing [quilt](https://www.man7.org/linux/man-pages/man1/quilt.1.html) in west; this is a potentially very large project if you take it to its logical extreme. Need to consider scope and future-proofing carefully since west is [conservative about features](https://docs.zephyrproject.org/latest/develop/west/why.html#design-constraints) and this use case can already be handled with a custom fork and some user-side scripts (or using quilt on top of west)
- probably want to either allow a list of patches or a directory containing patches; just a single patch is not going to be enough for everyone
- would need to specify the error semantics clearly
- would need to specify interactions with `west update` options like `--keep-descendants` and `--rebase` 
- `git apply`: why not `git am`? I think this is preferable because then you end up with a clean `git status` instead of a dirty tree

marc-hb, 2022-08-24:
tl;dr: why not just use a git branch? `.patch` files are using the filesystem to avoid using proper version control.

quilt-like tools managing `.patch` files have a lot of value when having to deal with ancient version control systems that don't let you rewrite history. However `git` supports rewriting the history of "volatile" branches and `west` is hardcoded to git.

`git` made most features of quilt-like tools redundant and `git` keeps getting new features like `range-diff` that keep making quilt-like tools even more redundant. For the very last quilt features that can't easily be implemented using git branches, I highly recommend tools like https://github.com/git-series/git-series/

cc: @joshtriplett, @jwb99

DonShogun, 2022-08-24:
> tl;dr: why not just use a git branch? `.patch` files are using the filesystem to avoid using proper version control.

I can easily answer for this one. Very often you don't have a permission to specific project. Forking and branching is bad when you want to keep as much as you can with the main branch. Patch often means a temporary situation to fix the problem without nasty hacks.

marc-hb, 2022-08-24:
Possibly the main reason behind git (and Github's) success is you do NOT need any permission to clone or fork (basically the same thing). I actually taught myself git for the first time so I could use `git-svn` and use version control without any permission.

> when you want to keep as much as you can with the main branch. Patch often means a temporary situation to fix the problem without nasty hacks.

You can keep rebasing a branch on top of the main branch, this is at least as easy to automate as applying `.patch` files. It also tracks the history of the rebases easier, for instance you can go back to any older rebase with a simple `git checkout`.

mbolivar-nordic, 2022-08-24:
> In bigger project this is a LOT to do.

For the record, I don't really see why this is a lot to do from a technical perspective. Forks are cheap, and you will have to keep the patch files in some git repository in the workspace anyway (or west won't be able to find it when you initialize a workspace from scratch).

From an organizational perspective where you have to get authorization to create a repository, I could see some more roadblocks for this, but I'm not sure that's west's problem to solve.

marc-hb, 2022-08-24:
> First you need to fork and then create new branch. After all you are maintainer.

That's... not what I wrote. Possibly the main reason behind git (and Github's) success is that ANYONE can clone or fork; no need to be a "maintainer" or anything.

> In bigger project this is a LOT to do.

Don't point the manifest at the "bigger project", point it to your fork/clone.

DonShogun, 2022-08-24:
> ANYONE can clone or fork; no need to be a "maintainer" or anything.

Sorry, it was about responsibilities(not roles/accounts/access) during and after fork - maintenance :)

marc-hb, 2022-08-25:
Do you mean rebasing your private `test-001` branch regularly? Looks like a small effort to me and worth it from a version tracking perspective.

Note `west update` also supports a `--rebase` option if late conflict detection is OK for you.

mbolivar-nordic, 2022-08-25:
> Sorry, it was about responsibilities(not roles/accounts/access) during and after fork - maintenance :)

Can you break this down in a bit more detail?

If it's not about roles/accounts, then I would like to know more about what responsibilities you think would be made much easier with this feature. Like @marc-hb, I don't see the difficulty here at first glance.

- If you have `revision: <SOME_SHA>` for the project you want to patch, you'll need to update this regularly and manually yourself as you integrate changes in the project you want to patch. With this feature, that's all you have to do. Without this feature, you have an additional step of needing to do a rebase (or merge, etc.) into your fork. I would not call this "a LOT to do". Updating a branch is not a lot to do. The hard part is testing the result. 
- If you have `revision: <SOME_BRANCH>` instead, then with this feature, you wouldn't have to do anything, but you'd lose determinism and the patches you wanted to apply to the project would just randomly break at some point in time; have fun with that. Without this feature, you have to manually update the branch in your fork (rebase/merge/etc.), giving you an opportunity to test the changes at least.

@DonShogun if you can please explain what I'm missing here and what the massive amount of extra work you have to do without this feature is, I'm definitely open to hearing more.

Otherwise I think I will close this issue with the above comment as an explanation why I do not think the juice is worth the squeeze here.

DonShogun, 2022-08-30:
Of course. When you can't just simple create branch because you don't have permission or it needs to be accepted by many people (time is money). You are using local mirrors instead of worldwide links (fork on your Github/Gitlab account is not available). You just want to check if simple change can fix you issue. To introduce new features when some other libraries are using different settings. Just simple to fix and test something. In above situations you must write email to architects and other project/teams, wait for response and at the end remove forks/branches 2 months after release, because they are no longer needed. This is just idea for some improvements. Using yocto bitbake I know how dangerous is that but I also know how helpful can be.

marc-hb, 2022-08-30:
>  You are using local mirrors instead of worldwide links

A .patch file is local too.

> (fork on your Github/Gitlab account is not available). 

Forking is the defining feature of Github and Gitlab.

Even if someone finds a way to disable that then it's still futile because you can always `git clone`, change something and upload somewhere else. Forking is only a shortcut.

> When you can't just simple create branch because you don't have permission or it needs to be accepted by many people (time is money).

While this is frequently controlled in central locations (for good reasons) you can always create branches locally as easily as creating a .patch file. In fact `git clone` is already creating branches. The days of version control red tape are over.

marc-hb, 2022-08-30:
> Do you mean rebasing your private test-001 branch regularly? Looks like a small effort to me and worth it from a version tracking perspective.

Actually zero effort if you make your `.patch` file a Pull Request because Github continuously updates `git fetch origin pull/12345/merge`, that's done for you.

marc-hb, 2022-08-30:
In the past I've used Yocto and bitbake too. I even discussed its patching approach with one of its main developers. The problems Yocto is trying to solve are very far beyond `west`'s objectives and again Yocto layers have to support non-git projects whereas `west` is hardcoded to git. I believe Yocto relies on .patch files as a "necessary evil", not as a desired design choice. Having no better choice, developing in Yocto requires cumbersome indirections like `devtool finish`. This is not an example to follow unless there is really no other option.

https://docs.yoctoproject.org/overview-manual/concepts.html#patching
https://docs.yoctoproject.org/sdk-manual/extensible.html#use-devtool-add-to-add-an-application

mbolivar-nordic, 2022-09-14:
@DonShogun all of your examples are about roles, accounts, and access, not technical difficulties, so I'm going to close this as I said I would.