zephyrproject-rtos/west #600: west config does not sanitize option key

desowin, 2022-09-07:
The expected use case works correctly, i.e.:
```
[tmon@tmon-laptop ~]$ west config --global build.board
nrf52840dk_nrf52840
```

However, when the option does not contain period, west terminates with an unhandled exception. 
```
[tmon@tmon-laptop ~]$ west config --global board
Traceback (most recent call last):
  File "/usr/bin/west", line 33, in <module>
    sys.exit(load_entry_point('west==0.14.0', 'console_scripts', 'west')())
  File "/usr/lib/python3.10/site-packages/west/app/main.py", line 793, in main
    app.run(argv or sys.argv[1:])
  File "/usr/lib/python3.10/site-packages/west/app/main.py", line 109, in run
    self.run_command(argv)
  File "/usr/lib/python3.10/site-packages/west/app/main.py", line 342, in run_command
    cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "/usr/lib/python3.10/site-packages/west/commands.py", line 135, in run
    self.do_run(args, unknown)
  File "/usr/lib/python3.10/site-packages/west/app/config.py", line 150, in do_run
    self.read(args)
  File "/usr/lib/python3.10/site-packages/west/app/config.py", line 181, in read
    value = self.config.get(args.name, configfile=args.configfile or ALL)
  File "/usr/lib/python3.10/site-packages/west/configuration.py", line 173, in get
    return self._get(lambda cf: cf.get(option), default, configfile)
  File "/usr/lib/python3.10/site-packages/west/configuration.py", line 216, in _get
    return getter(cf)
  File "/usr/lib/python3.10/site-packages/west/configuration.py", line 173, in <lambda>
    return self._get(lambda cf: cf.get(option), default, configfile)
  File "/usr/lib/python3.10/site-packages/west/configuration.py", line 77, in get
    return self._get(option, self.cp.get)
  File "/usr/lib/python3.10/site-packages/west/configuration.py", line 89, in _get
    section, key = option.split('.', 1)
ValueError: not enough values to unpack (expected 2, got 1)
```

It would be better to display some actual error message similar to what `git config` does:
```
[tmon@tmon-laptop ~]$ git config --global email
error: key does not contain a section: email
```

marc-hb, 2022-09-07:
I cannot reproduce, which `west` version is this?

desowin, 2022-09-07:
> I cannot reproduce, which `west` version is this?

0.14.0

I tried it on clean setup, and it turns out that in order to reproduce you have to have `~/.westconfig` with at least one entry. The config can be created with e.g. `west config --global build.board nrf52840dk_nrf52840`.

marc-hb, 2022-09-08:
Issue number 600! I could not resist a tentative quick fix: #603

PR #604 (2022-09-13): config: validate option name
Verify that the user-provided option name contains a '.' before passing it off to API functions that require that.

Fixes: #600
commit ed01ab5f (2022-09-13): config: validate option name

Verify that the user-provided option name contains a '.' before
passing it off to API functions that require that.

Fixes: #600
PR #603 (2022-09-21): configuration: fix error handling of keys missing a '.'

Fixes #600

De-duplicate the `option.split('.', 1)` "parsing" code into a new,
minimalistic and centralized `_InternalCF.key_parse` function. Add two
lines of error handling in this new function in order to provide a
meaningful error message instead of this cryptic and context free exception
when using a "build_board" key name without a dot:

```
    ...
    File "/usr/lib/python3.10/site-packages/west/configuration.py", in _get
    section, key = option.split('.', 1)
    ValueError: not enough values to unpack (expected 2, got 1)
```

T [... cut]
commit 63569cef (2022-09-21): configuration: fix error handling of keys missing a '.'

Fixes #600

De-duplicate the `option.split('.', 1)` "parsing" code into a new,
minimalistic and centralized `_InternalCF.key_parse` function. Add two
lines of error handling in this new function in order to provide a
meaningful error message instead of this cryptic and context free exception
when using a "build_board" key name without a dot:

```
    ...
    File "/usr/lib/python3.10/site-packages/west/configuration.py", in _get
    section, key = option.split('.', 1)
    ValueError: not enough values to unpack (expected 2, got 1)
```

T [... cut]
closed 2022-09-13 by mbolivar-nordic