zephyrproject-rtos/west #608: Provide a command line option for skipping unwanted portions of manifest

omkar3141, 2022-10-20:
Doing a west update can consume a significant amount of time if there are many dependencies specified in the west.yml file. It will be really great to have a command line option to skip updates of certain repositories.

For example:
`west update --skip "matter,homekit,liblc3codec,hal_st,other-unwanted-stuff"`

Such an option should apply the filter to all nested manifest files (if any) and not just the top-level west.yml file.

Such commands can then be converted to shorthand aliases to quickly swap your checked-out repos depending on the projects being worked on. This can also help save some amount of disk space.

marc-hb, 2022-10-20:
Feels like a duplicate of
- https://github.com/zephyrproject-rtos/zephyr/issues/46801

See also
- #519

mbolivar-nordic, 2022-10-24:
> TL;DR the whole thing would need to be thought over very carefully and a lot of tests would need to be written to validate the design

... or we would need to limit the scope of the feature, e.g. "you can't skip a project update if the project has imports", something like that. Even then I'm not totally sure it's trivial.