zephyrproject-rtos/west #609: Provide different SSH credentials for git repositories in west manifest files.

blake-bit, 2022-11-02:


marc-hb, 2022-11-02:
Seconding @henrikbrixandersen: @blake-bit can you explain what more specifically cannot be achieved with `.ssh/config` and why not? Stateless build systems (e.g. Docker) can change `.ssh/config` at build time if needed (paying special attention to file permissions; ssh is very strict about that); seems much easier to script than starting ssh-agent which is as you mentioned best for interactive use.

mbolivar-nordic, 2022-11-02:
I am also curious about why SSH_ASKPASS is not usable: https://git-scm.com/docs/gitcredentials/#_requesting_credentials

marc-hb, 2022-11-02:
> On the command line can be worked around manually by temporarily invoking a new ssh-agent with a given unique key,

Probably off-topic but... `ssh-agent` is not a "work-around" and its main feature is the ability to store _multiple_ keys for the whole session.

mbolivar-nordic, 2023-02-27:
closing due to lack of response

closed 2023-02-27 by mbolivar-nordic