zephyrproject-rtos/west #611: issue in west install.

harshvsingh-eaton, 2022-11-09:
Tried installing west using this command "python3 -m pip install -U west==0.12.0" which failed with this error " ##[error]DEPRECATION: docopt is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559 "
I tried using alternative solution of using the flag " --use-pep517 " which suppresses the error, here is the document page for this "https://pypi.org/project/pep517/" which also gives the alternative for the flag, that is to install the "build" but even after try it gave us the same error.

Can anyone suggest me how i can install west without using flag --use-pep517"

marc-hb, 2022-11-14:
> "python3 -m pip install -U west==0.12.0" 

What happens if you try to install the latest `west` version?

```
python3 -m pip install -U west
```

mbolivar-nordic, 2022-11-14:
> Can anyone suggest me how i can install west without using flag --use-pep517"

Did you try installing wheel?

harshvsingh-eaton, 2022-11-15:
After installing the wheel, i am able to run setup without using flag --use-pep517.

closed 2022-11-15 by harshvsingh-eaton