zephyrproject-rtos/west #616: Is there anyway to tell 'west' a different path to find board file

fidus-kai, 2022-12-21:
I have installed Nordic nRF SDK at C:\ncs drive, which has zephyr SDK and board folder.  My app project is located at different path, say D drive. when I run 'west build -b board-name',  the tool is automatically to go C:\ncs to find existing board-name, what if the customized board-name is within my project folder at D drive? I can't find a command/configure to define the board folder path.

marc-hb, 2022-12-21:
I believe `west` is merely passing the board name to CMake so this is technically not a west issue but a zephyr issue.

Does everything work if you temporarily move your board folder next to the existing boards?

Have you followed all the instructions here?
https://docs.zephyrproject.org/latest/hardware/porting/board_porting.html#create-your-board-directory

fidus-kai, 2022-12-21:
it works if I move the new-board to zephyr's boards folder.   Still looking for a way to point the new-board at different path.

jobasto, 2022-12-21:
IIRC `west` looks in the top project and all imported projects for the `boards` folder. https://github.com/zephyrproject-rtos/example-application is such a top level project that imports other projects via its `west.yml`. See the [boards/arm/custom_plank](https://github.com/zephyrproject-rtos/example-application/tree/main/boards/arm/custom_plank) folder.

One thing I had problems with was when the top level project was not checked out into the folder that was given in `path` (see [west.yml](https://github.com/zephyrproject-rtos/example-application/blob/96aff8a4eef2270fcffa591b94beb470580bdfc9/west.yml#L6)).
```yaml
manifest:
  self:
    path: example-application
```
So make sure that you check out the project into the right folder or don't put `path` into your `west.yml` at all. You can also [add additional modules](https://docs.zephyrproject.org/3.1.0/develop/modules.html#integrate-modules-in-zephyr-build-system) to your build system by some CMake variables (e.g. ZEPHYR_EXTRA_MODULES).

marc-hb, 2022-12-21:
You are of course using `-DBOARD_ROOT=`?

jobasto, 2022-12-21:
> You are of course using `-DBOARD_ROOT=`?

That's not necessary if one structures ones project following the zephyr directory layout and makes sure the the project is considered a ZEPHYR_MODULE (via ZEPHYR_EXTRA_MODULES or by configuring west.yml correctly).

The advantage is that the board directory is then not only what one has given via `-DBOARD_ROOT=` but the set of all boards directories of all referenced projects.

marc-hb, 2022-12-21:
> That's not necessary if...

Sounds great but I'm urging @fidus-kai to _at least temporarily_ try `-DBOARD_ROOT=` and see is that helps.

fidus-kai, 2022-12-21:
Just noticed that there is -DBOARD_ROOT config option at following link https://docs.zephyrproject.org/latest/develop/application/index.html#boards

While I originally only checked https://docs.zephyrproject.org/latest/develop/application/index.html#boards, which doesn't mention the BOARD_ROOT.

Thanks.

Kai Zhang | Senior Firmware Developer | 519.591.6093 | fidus.com

fidus-kai, 2022-12-21:
Thanks.

Just noticed that there is -DBOARD_ROOT config option at following link https://docs.zephyrproject.org/latest/develop/application/index.html#boards

While I originally only checked https://docs.zephyrproject.org/latest/hardware/porting/board_porting.html#create-your-board-directory
, which doesn't mention the BOARD_ROOT.

Kai Zhang | Senior Firmware Developer | 519.591.6093 | fidus.com

closed 2022-12-21 by fidus-kai