zephyrproject-rtos/west #622: Improvements for managing untracked projects

joerchan, 2023-01-14:
Whenever the manifest file is modified by removing entries in the manifest or by moving the manifest entry path the project folder may end up containing folders that are no longer needed or duplicated entries.

An example of this is the TF-M repository split, where the folder name was changed from `tfm` to `tf-m`.
Users doing `west update` after this change will end up with duplicated entries in their folder structure when the new project is imported by the old one still remains.

Usually what the user wants to do at this point is to remove the not-used project as it now has duplicated entries and this causes problems when searching in the projects for symbols.

Since the folders usually are git repositories cleaning this up should not be handled automatically by the tool as the user my still have branches in the old folder that they wish to clean up first.

Providing the user with the information about these old folders with by useful so that they can do the appropriate action to get rid of them.

Listing the untracked folders in a similar way as git does when doing `west status` would be useful.
West should skip folders that are inside existing projects, as this would otherwise be listed by git.

Example:

```
west status
=== status of manifest (zephyr):
On branch main
Your branch is up to date with 'upstream/main'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	build/

nothing added to commit but untracked files present (use "git add" to track)
=== 

Untracked folders:
	modules/tee/tfm

Add files to west.yml to track
```

joerchan, 2023-01-16:
> I would keep it simple and report only the top level directories (e.g.: modules/ here). Maybe top level files too for consistency with git?

But `modules/` and `modules/tee` contain projects that are tracked, i.e. `modules/tee/tf-m/trusted-firmware-m` for example in this case.

marc-hb, 2023-01-17:
Sorry I wrote too fast. I meant: if there is a `modules/foo/bar/` directory, then it should only report `modules/foo/` but not `modules/foo/bar` to keep the output short (assuming there is no west project anywhere below `modules/foo/`).

Different git clients report the above situation with different verbosity.

On the other hand, if there _is_ a west project `modules/foo/project/`, then it should report `modules/foo/bar/`.

So the pseudo-code should probably look like this:

```python
def west_find_untracked(parent):
   for child in parent:
        if child is a west module:
             continue # nothing to report
        if no west module below child:
             report child
             continue
        else:
             west_find_untracked(child)

west_find_untracked(west_top)
```

Nested projects should not matter: anything that `git status --ignored` can report already should not be reported a second time by west.

marc-hb, 2025-03-17:
>  However, I will take a very simplified approach

I don't know whether it works in every possible situations, but the pseudo-code I wrote above does not look very complex to me :-)

> anything under the folder of any west project is ignored.

So what about this:

```
west_project/two/levels/module_present/.git
west_project/two/levels/old_module_gone/.git
```

... where old_module_gone is not a _direct_ child of west_project. Will it be reported?

PR #786 (2025-04-01): app: manifest: Add the ability to display untracked files and directories
    This new command-line switch implements something similar to `git
    status`'s untracked files display.
    
    The code examines the workspace and prints out all files and folders
    that are not inside of a west project.

Fixes: #622
Supersedes #702
closed 2025-04-01 by pdgendt