zephyrproject-rtos/west #633: west update should exit with none-zero code if update failed

Hoohaha, 2023-02-21:
Hi there,

`west update` command always exit zero code even if there have projects update failed.
It should return none-zero code if something wrong.
This is useful in CI build system to check the result of update.

## Reproduce Steps (windows)

1. Run command:
```bash
west update
```
2. Press ctrl+c
3. Check return code
```bash
ehco %ERRORLEVEL%
```

marc-hb, 2023-02-21:
I'm not sure Ctrl-C on Windows lets `west` decide what the return code is.

Please try the following:

- Do not use Ctrl-C, corrupt the SHA1 of one project in the `west.yml` manifest and run west update. Share what the `%ERRORLEVEL%` is. I justed tested that on Linux and it's not zero. You can also try corrupting a git remote.

- On Windows, use Ctrl-C with other programs (sleep?) and share what the %ERRORLEVEL% is.

Hoohaha, 2023-02-21:
> * Do not use Ctrl-C, corrupt the SHA1 of one project in the `west.yml` manifest and run west update. Share what the `%ERRORLEVEL%` is. I justed tested that on Linux and it's not zero. You can also try corrupting a git remote.

I did the testing as you described, west returned 1 as expected.
My bad, it's not an issue.
But no idea why west returned with CTRL-C.

Anyway, thank you Marc

closed 2023-02-21 by Hoohaha

marc-hb, 2023-03-01:
> But no idea why west returned with CTRL-C.

Have a look at new #636

Hoohaha, 2023-03-07:
@marc-hb Thank you, tested with latest commit, returncode is none-zero if ctrl+c