zephyrproject-rtos/west #634: Command `west build -t rom_report` results in FATAL ERROR

IvanVnucec, 2023-02-23:
I'm using nRF Connect SDK v2.0.2.
Machine is Windows 10.
Before calling `west` commands, I've ran `C:\ncs\v2.0.2\zephyr\zephyr-env.cmd` with console to set needed environment variables.
Then I change directory to the root folder of my app and run `west build -b <some_board>`.
After succesfull build, I've ran `west build -t rom_report` which, after 15 seconds or so, results in FATAL ERROR shown below:
```sh
-- west build: running target rom_report
[0/1] cmd.exe /C "cd /D C:\Users\Korisnik\Desktop\ivan\projekti\art_nr1amp\build && C:\ncs\toolchains\v2.0.2\opt\bin\python.exe C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report -k C:/project/build/zephyr/zephyr.elf -z C:/ncs/v2.0.2/zephyr -o C:/project/build --workspace=C:/ncs/v2.0.2 -d 99 rom"
Path                                                                                             Size    %
==============================================================================================================
Root                                                                                           212424 100.00%
Traceback (most recent call last):
  File "C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report", line 803, in <module>
    main()
  File "C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report", line 792, in main
    print_any_tree(root, symsize, args.depth)
  File "C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report", line 697, in print_any_tree
    print(f"{row.pre}{cc}{row.node.name} {s} {cr}{Fore.BLUE}{percent:6.2f}%{Fore.RESET}")
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 41, in write
    self.__convertor.write(text)
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 162, in write
    self.write_and_convert(text)
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 187, in write_and_convert
    self.write_plain_text(text, cursor, start)
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 195, in write_plain_text
    self.wrapped.write(text[start:end])
  File "encodings\cp1252.py", line 19, in encode
UnicodeEncodeError: 'charmap' codec can't encode characters in position 0-2: character maps to <undefined>
FAILED: zephyr/cmake/reports/CMakeFiles/rom_report
cmd.exe /C "cd /D C:\project\build && C:\ncs\toolchains\v2.0.2\opt\bin\python.exe C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report -k C:/project/build/zephyr/zephyr.elf -z C:/ncs/v2.0.2/zephyr -o C:/project/build --workspace=C:/ncs/v2.0.2 -d 99 rom"
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\ncs\toolchains\v2.0.2\opt\bin\cmake.EXE' --build 'C:\project\build' --target rom_report
```

marc-hb, 2023-02-23:
I suspect west is not involved, can you please try to run this (long!) command:
> cmd.exe /C "cd /D C:\Users\Korisnik\Desktop\ivan\projekti\art_nr1amp\build && C:\ncs\toolchains\v2.0.2\opt\bin\python.exe C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report -k C:/project/build/zephyr/zephyr.elf -z C:/ncs/v2.0.2/zephyr -o C:/project/build --workspace=C:/ncs/v2.0.2 -d 99 rom" 

And then this in two steps

> cd /D C:\Users\Korisnik\Desktop\ivan\projekti\art_nr1amp\build

> C:\ncs\toolchains\v2.0.2\opt\bin\python.exe C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report -k C:/project/build/zephyr/zephyr.elf -z C:/ncs/v2.0.2/zephyr -o C:/project/build --workspace=C:/ncs/v2.0.2 -d 99 rom

Finally, try all these commands in powershell

IvanVnucec, 2023-02-24:
> I suspect west is not involved, can you please try to run this (long!) command:
> 
> > cmd.exe /C "cd /D C:\Users\Korisnik\Desktop\ivan\projekti\art_nr1amp\build && C:\ncs\toolchains\v2.0.2\opt\bin\python.exe C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report -k C:/project/build/zephyr/zephyr.elf -z C:/ncs/v2.0.2/zephyr -o C:/project/build --workspace=C:/ncs/v2.0.2 -d 99 rom"
> 
> And then this in two separate steps:
> 
> > cd /D C:\Users\Korisnik\Desktop\ivan\projekti\art_nr1amp\build
> 
> > C:\ncs\toolchains\v2.0.2\opt\bin\python.exe C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report -k C:/project/build/zephyr/zephyr.elf -z C:/ncs/v2.0.2/zephyr -o C:/project/build --workspace=C:/ncs/v2.0.2 -d 99 rom
> 
> Finally, try all these commands in powershell

I think that the west is involved because if I run `west build -t rom_report` command in a modified repo, the project builds the changes and then the ROM report command fails.

I've tried to run the long command you've provided but the same error occurrs:
```sh
$ cmd.exe /C "cd /D C:\Users\Korisnik\Desktop\ivan\projekti\art_nr1amp\build && C:\ncs\toolchains\v2.0.2\opt\bin\python.exe C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report -k C:/Users/Korisnik/Desktop/ivan/projekti/art_nr1amp/build/zephyr/zephyr.elf -z C:/ncs/v2.0.2/zephyr -o C:/Users/Korisnik/Desktop/ivan/projekti/art_nr1amp/build --workspace=C:/ncs/v2.0.2 -d 99 rom"
Path                                                                                             Size    %
==============================================================================================================
Root                                                                                           204648 100.00%
Traceback (most recent call last):
  File "C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report", line 803, in <module>
    main()
  File "C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report", line 792, in main
    print_any_tree(root, symsize, args.depth)
  File "C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report", line 697, in print_any_tree
    print(f"{row.pre}{cc}{row.node.name} {s} {cr}{Fore.BLUE}{percent:6.2f}%{Fore.RESET}")
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 41, in write
    self.__convertor.write(text)
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 162, in write
    self.write_and_convert(text)
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 187, in write_and_convert
    self.write_plain_text(text, cursor, start)
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 195, in write_plain_text
    self.wrapped.write(text[start:end])
  File "encodings\cp1252.py", line 19, in encode
UnicodeEncodeError: 'charmap' codec can't encode characters in position 0-2: character maps to <undefined>
```

Then I've tried to change directory to `build` and from there run this long command as you've mentioned, but the problem still occurs:
```sh
$ cmd.exe /C "C:\ncs\toolchains\v2.0.2\opt\bin\python.exe C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report -k C:/Users/Korisnik/Desktop/ivan/projekti/art_nr1amp/build/zephyr/zephyr.elf -z C:/ncs/v2.0.2/zephyr -o C:/Users/Korisnik/Desktop/ivan/projekti/art_nr1amp/build --workspace=C:/ncs/v2.0.2 -d 99 rom"
Path                                                                                             Size    %
==============================================================================================================
Root                                                                                           204648 100.00%
Traceback (most recent call last):
  File "C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report", line 803, in <module>
    main()
  File "C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report", line 792, in main
    print_any_tree(root, symsize, args.depth)
  File "C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report", line 697, in print_any_tree
    print(f"{row.pre}{cc}{row.node.name} {s} {cr}{Fore.BLUE}{percent:6.2f}%{Fore.RESET}")
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 41, in write
    self.__convertor.write(text)
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 162, in write
    self.write_and_convert(text)
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 187, in write_and_convert
    self.write_plain_text(text, cursor, start)
  File "C:\ncs\toolchains\v2.0.2\opt\bin\lib\site-packages\colorama\ansitowin32.py", line 195, in write_plain_text
    self.wrapped.write(text[start:end])
  File "encodings\cp1252.py", line 19, in encode
UnicodeEncodeError: 'charmap' codec can't encode characters in position 0-2: character maps to <undefined>
```

Then I've tried with Windows PowerShell. I've again ran `C:\ncs\v2.0.2\zephyr\zephyr-env.cmd` within PowerShell, then changed directory into `build` directory of my project and ran this command `C:\ncs\toolchains\v2.0.2\opt\bin\python.exe C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report -k C:/Users/Korisnik/Desktop/ivan/projekti/art_nr1amp/build/zephyr/zephyr.elf -z C:/ncs/v2.0.2/zephyr -o C:/Users/Korisnik/Desktop/ivan/projekti/art_nr1amp/build --workspace=C:/ncs/v2.0.2 -d 99 rom`.
And it **worked**.

IvanVnucec, 2023-02-24:
When I try to run `$ cmd.exe /C "C:\ncs\toolchains\v2.0.2\opt\bin\python.exe C:/ncs/v2.0.2/zephyr/scripts/footprint/size_report -k C:/Users/Korisnik/Desktop/ivan/projekti/art_nr1amp/build/zephyr/zephyr.elf -z C:/ncs/v2.0.2/zephyr -o C:/Users/Korisnik/Desktop/ivan/projekti/art_nr1amp/build --workspace=C:/ncs/v2.0.2 -d 99 rom"
` command in MINGW64, the project fails with the same error.

marc-hb, 2023-02-24:
Thanks @IvanVnucec for the detailed reproduction steps, very useful.

As a workaround I recommend you switch to Powershell if possible, cmd.exe is long obsolete anyway.

Since you reproduced the same with `size_report` directly this failure is unrelated to west. I'm transferring it.