zephyrproject-rtos/west #642: west status to provide also info about all repos which are not in the manifest revision

aescolar, 2023-03-10:
Similarly to how `repo status` works, it would be very nice if `west status` also provided information about all projects which are off the manifest revision; not just with uncommitted changes, but also in another sha/branch/tag.

This would facilitate working with several local changes to sub-projects projects.

aescolar, 2023-03-10:
(That other issue is all over the place, so let me be a bit clearer about what I would hope, which would cover my needs)

For each project:
1. If the project has been moved to a different commit (different that the one west last took it to(^1) in the last update), to print a line saying so (^2).
2. If the project has an unclean git status, print the output of git status
3. Otherwise no output whatsoever for that project (not even the project name)

(^1) I understad this is manifest-rev. So the condition would be if that project is not in manifest-rev right now.
(^2) Repo even tells you if you checked out a different branch but are still in the same commit. The benefit of this is marginal. So telling or not about this case is not so important.

You can be even nicer and provide a return code > 0 to the shell if any project hit 1. or 2.

The command can be called whatever.
The command should not change anything locally or in the .west or .git folders, or do anything smarter than that.

If the user moved manifest-rev on his own, that is not something this command should protect against.

marc-hb, 2023-03-10:
> (That other issue is all over the place, so let me be a bit clearer about what I would hope, which would cover my needs)

There was indeed a lot of alternatives discussed in #548. However I think the simplest one and the one I defended the most was exactly what you're asking here.

AFAIK it was closed by @mbolivar-nordic like this, verbatim:

> west status, as a wrapper around git status, reports status relative to HEAD, not manifest-rev

I hope I'm wrong but I'm not holding my breath.

marc-hb, 2023-03-10:
> With plain bash, something like this would do what I was hoping:

I highly recommend the excellent tool `apt/dnf install shellcheck` :-)

Off-topic sorry.

mbolivar-nordic, 2023-03-10:
@aescolar thanks for the clear and concise issue description.

Indeed I don't think it is a good idea to change the semantics of `west status`: the west commands that are just wrappers for git commands with the same names should be as simple and stupid as possible IMO, so `west status` should work like `git status`, which compares worktree+index with HEAD.

> The command can be called whatever.

I'm proposing `west compare` here: https://github.com/zephyrproject-rtos/west/pull/643

Please test it and provide feedback. If it matches your expectations, I will write test cases and try to get it merged.

commit c0de9835 (2023-04-12): compare: add command

This command prints information about manifest-rev and "git status"
output for a project if (and only if) at least one of the following is
true:

1. its checked out commit is different than the commit checked
   out by the most recent successful "west update"
2. its working tree is not clean (i.e. there are local uncommitted
   changes)
3. it has a local checked out branch (unless overridden via
   the compare.ignore-branches config option or the --ignore-branches
   command line option)

It does something similar for the manifest repository.

The purpose of this is to al [... cut]
closed 2023-04-12 by mbolivar-nordic