zephyrproject-rtos/west #644: Provide option for 'west init` or `west update` to fetch all submanifests

aescolar, 2023-03-24:
It would be nice having an option/command line switch in `west init` that would make it fetch all repos which have sub-manifests (projects with import) at init time.
Alternatively it could be a new command line switch to `west update` which instructs it to fetch those same repos (and nothing else).
Either of these would not (need to) find projects with sub-manifests that are affected by group filters (filtered out).

The aim would be to be able to have a workspace in which west can already work without erroring out, but without the need to pull all repositories content.

An example use-case is the solution to the problem described here:
https://github.com/zephyrproject-rtos/zephyr/pull/55696#issuecomment-1481530833
in which after `west init`, as the main manifest has a project with a sub-manifest/import, west errors out if one tries to use quite a bit of its functionality (for ex. anything related to manifest, list or status).
A solution to this particular problem, is to either do a full `west update` pulling all projects (which may be a quite a bit of unnecessary overhead), OR, to do a `west update <missing_repos_with_imports>` which requires knowledge of which repos are the ones with sub-manifests.

aescolar, 2023-03-24:
CC @stephanosio

tomi-font, 2024-04-22:
This issue seems somewhat related to a problem I faced (and is one of the issues I stumbled upon in my search for solution), so commenting here.

Basically my problem was that I needed the `tf-m-tests` repo (which is defined in [submanifests/optional.yaml](https://github.com/zephyrproject-rtos/zephyr/blob/main/submanifests/optional.yaml)) to be updated to the manifest revision because I had changed it and was building a sample that made use of it.
Because it was somehow already present (I don't remember myself doing anything to fetch those repos) I assumed that the simple `west update` did update those repos as well. But turns out not, and this did cost some time. (At least the documentation?)

I eventually found the command that does the job I needed: `west update --gf +optional`
But it was hard to find.
I think some thing(s) could be improved with regards to this, but I don't quite know what exactly.

marc-hb, 2024-04-26:
`west list | grep tf-m` ?

tomi-font, 2024-04-26:
With `west config manifest.group-filter +optional`:

> psa-arch-tests modules/tee/tf-m/psa-arch-tests 2cadb02a72eacda7042505dcbdd492371e8ce024 https://github.com/zephyrproject-rtos/psa-arch-tests
tf-m-tests   modules/tee/tf-m/tf-m-tests  08a3158f0623a4205608a52d880b17ae394e31d2 https://github.com/zephyrproject-rtos/tf-m-tests
trusted-firmware-m modules/tee/tf-m/trusted-firmware-m 0b898c9b72171b0a260c0bb64a92ea4713f9e931 https://github.com/zephyrproject-rtos/trusted-firmware-m

With `west config manifest.group-filter ''`:
> trusted-firmware-m modules/tee/tf-m/trusted-firmware-m 0b898c9b72171b0a260c0bb64a92ea4713f9e931 https://github.com/zephyrproject-rtos/trusted-firmware-m