zephyrproject-rtos/west #647: How to get Zephyr project application .elf filepath from a custom west command?

IvanVnucec, 2023-04-07:
I'm developing a custom Zephyr app using the [Zephyr example application](https://github.com/zephyrproject-rtos/example-application).
I've created `debug_ozone.py` file in the `scripts` folder which will start the Segger Ozone debugger over command line. Command line call needs to have the path to the app .elf file.

- How do I get the .elf path? I see that the `west flash` command uses the `get_build_dir` command [here](https://github.com/zephyrproject-rtos/zephyr/blob/38d9da726644c8e82b3d8390ef85a175d63804ec/scripts/west_commands/flash.py#L30), but I'm not sure how do I import `get_build_dir` command because it's defined in the `zephyr/scripts/west_commands /run_common.py` file
- Follow-up question, how can I assure that the custom west command will (re)build the project if it needs to (just how `west debug` command does it)?

marc-hb, 2023-04-09:
> but I'm not sure how do I import get_build_dir command because it's defined in the zephyr/scripts/west_commands /run_common.py file

Why not just `from run_common import add_parser_common, do_run_common, get_build_dir` as in the source you pointed at?

> Follow-up question, how can I assure that the custom west command will (re)build the project if it needs to (just how west debug command does it)?

I googled "do_run_common" that I found at the link you shared and I quickly found its source. It's main purpose seems to invoke some `rebuild()` function. 

While they do not seem to help in this particular case, keep in mind that there is some west documentation at https://docs.zephyrproject.org/latest/develop/west/west-apis.html

marc-hb, 2023-04-11:
There is some PYTHONPATH / sys.path magic you need to figure out.

mbolivar-nordic, 2023-04-12:
Hi! I will try to answer your questions, but this is not an issue for the west repository since it is about extension commands that are defined in the zephyr repository, so I'm also going to close it. Please feel free to open an issue or discussion in the zephyr repository if follow-up is needed.

> * How do I get the .elf path? I see that the `west flash` command uses the `get_build_dir` command [...]

These are implementation details and you should not be relying on them. See cmake/modules/kernel.cmake for the rules that set KERNEL_ELF_NAME from CONFIG_KERNEL_BIN_NAME. You can load the .config and figure out the .elf path from the build directory.

> Follow-up question, how can I assure that the custom west command will (re)build the project if it needs to (just how west debug command does it)?

Run cmake in build tool mode: `cmake --build <build directory>`. This is all that west does; see run_build() in zcmake.py

closed 2023-04-12 by mbolivar-nordic

IvanVnucec, 2023-04-12:
Thank you very much. Maybe I'll post a solution, once I figure it out, for the others that might stumble on this issue.