zephyrproject-rtos/west #655: Infinite recursion

botbotty, 2023-04-28:
https://github.com/zephyrproject-rtos/west/blob/51ad8866fbb508e2fc42cf8d3cce24d888ee363f/src/west/commands.py#L280-L289

`self.die()` checks config `color.ui`, which will call `self._get_config()` again, but `self._config` is still `None`.

marc-hb, 2023-04-28:
Looks like a good catch. For the record can you please share the west version and some basic reproduction steps?

botbotty, 2023-04-29:
west version is v1.0.0

reproduction steps:
```sh
west init -m https://github.com/zephyrproject-rtos/example-application reproduction
cd reproduction/
west update

west example-west-command --help # so far so good

cat > patch.diff << EOF
--- a/example-application/scripts/example_west_command.py
+++ b/example-application/scripts/example_west_command.py
@@ -24,6 +24,7 @@ reflowed for you. You can also pass
 formatter_class=argparse.RawDescriptionHelpFormatter when calling
 parser_adder.add_parser() below if you want to keep your line
 endings.''')
+        self.manifest # Try to access manifest

     def do_add_parser(self, parser_adder):
         # This is a bit of boilerplate, which allows you full control over the
EOF
patch -p1 < patch.diff

west example-west-command --help # this will fail
```

mbolivar-nordic, 2023-05-05:
Hmm, I believe we fixed this in 354b0e8ee6f7719e491a0c3f8685ac44cbc3a9c4. @botbotty the steps to reproduce you've posted are using `self.manifest` outside of `WestCommand.do_run()`, which is not legal. From the docstring for this property:

```
        '''Property for the manifest which was passed to run().

        If `do_run` was given a *manifest* kwarg, it is returned.
        Otherwise, a fatal error occurs.
        '''
```

Reading this property from the constructor is an attempt to read it before do_run() was called, so an error is not unexpected.

Can you make this happen in another way?

mbolivar-nordic, 2023-05-16:
> Can you make this happen in another way?

Please reopen with more details if so, thanks!