zephyrproject-rtos/west #658: Add an option for setup a local mirror just "repo init --mirror"?

swordligit, 2023-05-06:
For android repo tool, there is an option 'repo init --mirror' to setup a local 
mirror, but with west,  have to git clone --bare all the projects one by one.

So, is it possible to add this option?
Br,
Yingchun

mbolivar-nordic, 2023-05-06:
This seems like a valid way to set up a cache you could use with `west update --name-cache`. But since `west init` only clones the manifest repository, what would the commands that you would like to type be, and what would the desired outcomes be?

swordligit, 2023-05-08:
In my local server, I want to clone all the repository with the bare mode,
then all my members can clone whole projects from my local server.
not from the upsteam.

Here is my simple steps now:
```
west init -m https://github.com/zephyrproject-rtos/zephyr

west list | awk '{print $4}' > projects.list
mkdir -p /path/of/mirror/; cd /path/of/mirror

for p in $(cat projects.list); do git clone --bare $p; done
```
After that, I can change the upstream url in west.yaml and download the projects
from my mirror.

I thought this is almost the android `repo init --mirror ` does.

marc-hb, 2023-05-09:
```shell
west list -f '{url}' | while read -r loc; do echo git clone "$loc"; done
```

Related:
- #615

swordligit, 2023-05-11:
seems there is a real requirement,can this be implemented? using an extra script is
not so convenient.

marc-hb, 2023-05-11:
Requirement funded by whom?

marc-hb, 2023-05-11:
There two ways to manage valid but unfunded requests in issue trackers.

1. The "project manager" way: close as wontfix because not funded.
2. The "developer" way: keep open because valid.

Most open-source projects work in the second way  because there is rarely any inter-company project management.

mbolivar-nordic, 2023-05-11:
Agree, this is a valid enhancement and I would take a patch adding a new `west mirror` command that accomplished this. This doesn't belong in `west init` since that command's job is to set up a new workspace with just a manifest repository in it.