zephyrproject-rtos/west #666: [RFC] New `inactive-by-default:` project key

mbolivar-nordic, 2023-06-01:
This issue tracks a potential extension to the manifest schema. It is not a proposal at this point, just a placeholder for discussion.

Proposed syntax:

```yaml
manifest:
  projects:
    - name: foo
      inactive-by-default: <true|false>
```

Proposed semantics:

- if key is missing, no behavior changes
- if key is `false`, no behavior changes
- if key is `true`, new behavior applies to whether project is active or inactive

The new behavior that applies if and only if the key is `true` is:

- if `manifest.project-filter` applies to the project, it decides whether the project is active or inactive; `inactive-by-default` has **no effect** in this case
- otherwise, the project is inactive

This applies regardless of where the project is defined in the import hierarchy.

Concerns:

- this can no longer be overridden in an importing manifest, so users will always have to take explicit action to enable such projects
- in the case of zephyr/west.yml, `west init` + `west update` will not result in a manifest that can be resolved

mbolivar-nordic, 2023-06-01:
Discussion among @carlescufi @tejlmand @aescolar @mbolivar-nordic :

Fact that this cannot be overridden in an importing manifest is a serious problem. E.g. in NCS we could not write a manifest that imports zephyr and makes bsim active by default.

mbolivar-nordic, 2023-06-01:
Potential way forward, **not vetted, just an RFC**

```
manifest:
  projects:
    - name: zephyr
      import:
        name-allowlist:
          - bsim
          - ...
        activate:
          - bsim
```

marc-hb, 2025-04-23:
> Discussion among @carlescufi @tejlmand @aescolar @mbolivar-nordic ...

Please review #808, thanks!