zephyrproject-rtos/west #674: west autocompletion in CLI is missing directories

ericsunplus, 2023-07-04:
I am using west version v1.0.0 in bash. bash has builtin path auto completion, but when I execute a west command line, the bash auto completion seems to be bypassed.

Suppose I have the following directory structure
```
|
+--- .west
|
+--- main_repo
        |
        +--- west.yml
        |
        +--- repoSet
                |
                +--- repo1
                |
                +--- repo2
                |
                +--- ...
```
I am in directory `main_repo` and try to see information of a sub repo
```
west list repoSet/repo1
```
When I type the command, `repoSet/repo1` do not have autocomplete. But when I use other bash commands, the path have auto complete.

marc-hb, 2023-07-05:
What happens if you run `west list repoSet/repo1`?

I'm afraid you're confusing module name and module path.

ericsunplus, 2023-07-07:
> What happens if you run `west list repoSet/repo1`?
> 
> I'm afraid you're confusing module name and module path.

`west list` have the ability to both module name and module path. Sometimes when there is too many repos in west.yml, it is hard to remember the module name, path autocompletion will help here

marc-hb, 2023-07-07:
> west list have the ability to handle both module name and module path.

Oops, you're right sorry: I tried `west list modules/lib/lz4` but that does not work for me, I must do `west list ../modules/lib/lz4` instead and now it works. That's not what I expected but I understand it now (of course `west list lz4` also works)

So back to the auto completion issue you raised, you're also correct. I just tried https://github.com/zephyrproject-rtos/zephyr/blame/main/scripts/west_commands/completion/west-completion.bash for the first time and it auto completes on module names only, it does not auto complete on paths. That's very common actually, it's not specific to west and it is why I usually disable "advanced"/programmable auto-completion. I don't know if it's even possible to performed "advanced" auto completion for `west list lz4` AND "basic", path-based auto completion for `west list ../modules/lib/lz4` _at the same time_, is it? @carlescufi do you know? Have you ever seen autocompletion for any command than west that can do both at the same time? If yes please share the example.

If you want to temporarily switch to the latter you can use `complete -r west` as a workaround. It will affect only the current shell and nothing else.

In any case https://github.com/zephyrproject-rtos/zephyr/blame/main/scripts/west_commands/completion/west-completion.bash is part of the zephyr repo now so I'm transferring the issue there.