zephyrproject-rtos/west #684: west init access denied on windows

lovettchris, 2023-09-08:
Repro:

1. pip install west  (gave me west version 1.1.0)
2. git clone https://github.com/nrfconnect/sdk-nrf.git
3. cd sdk-nrf
4. west init
5. west update

Actual:

=== Initializing in F:\git\nordic\sdk-nrf
--- Cloning manifest repository from https://github.com/zephyrproject-rtos/zephyr
Cloning into 'F:\git\nordic\sdk-nrf\.west\manifest-tmp'...
remote: Enumerating objects: 919248, done.
remote: Counting objects: 100% (2/2), done.
remote: Total 919248 (delta 1), reused 1 (delta 1), pack-reused 919246
Receiving objects: 100% (919248/919248), 590.38 MiB | 12.22 MiB/s, done.
Resolving deltas: 100% (698100/698100), done.
Updating files: 100% (28988/28988), done.
Traceback (most recent call last):
  File "F:\miniconda3\envs\sr\lib\shutil.py", line 816, in move
    os.rename(src, real_dst)
PermissionError: [WinError 5] Access is denied: 'F:\\git\\nordic\\sdk-nrf\\.west\\manifest-tmp' -> 'F:\\git\\nordic\\sdk-nrf\\zephyr\\manifest-tmp'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "F:\miniconda3\envs\sr\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "F:\miniconda3\envs\sr\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "F:\miniconda3\envs\sr\Scripts\west.exe\__main__.py", line 7, in <module>
  File "F:\miniconda3\envs\sr\lib\site-packages\west\app\main.py", line 1085, in main
    app.run(argv or sys.argv[1:])
  File "F:\miniconda3\envs\sr\lib\site-packages\west\app\main.py", line 244, in run
    self.run_command(argv, early_args)
  File "F:\miniconda3\envs\sr\lib\site-packages\west\app\main.py", line 503, in run_command
    self.run_builtin(args, unknown)
  File "F:\miniconda3\envs\sr\lib\site-packages\west\app\main.py", line 611, in run_builtin
    self.cmd.run(args, unknown, self.topdir,
  File "F:\miniconda3\envs\sr\lib\site-packages\west\commands.py", line 194, in run
    self.do_run(args, unknown)
  File "F:\miniconda3\envs\sr\lib\site-packages\west\app\project.py", line 224, in do_run
    topdir = self.bootstrap(args)
  File "F:\miniconda3\envs\sr\lib\site-packages\west\app\project.py", line 330, in bootstrap
    shutil.move(os.fspath(tempdir), os.fspath(manifest_abspath))
  File "F:\miniconda3\envs\sr\lib\shutil.py", line 834, in move
    rmtree(src)
  File "F:\miniconda3\envs\sr\lib\shutil.py", line 750, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "F:\miniconda3\envs\sr\lib\shutil.py", line 615, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "F:\miniconda3\envs\sr\lib\shutil.py", line 615, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "F:\miniconda3\envs\sr\lib\shutil.py", line 615, in _rmtree_unsafe
    _rmtree_unsafe(fullname, onerror)
  File "F:\miniconda3\envs\sr\lib\shutil.py", line 620, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "F:\miniconda3\envs\sr\lib\shutil.py", line 618, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 5] Access is denied: 'F:\\git\\nordic\\sdk-nrf\\.west\\manifest-tmp\\.git\\objects\\pack\\pack-409852465dd22b365374e95e4d0eedb41d6856bf.idx'

If I create the "ncs" folder manually it works, but then "west update" fails:

(sr) F:\git\nordic\sdk-nrf>west update
Traceback (most recent call last):
  File "F:\miniconda3\envs\sr\lib\site-packages\west\manifest.py", line 1874, in _top_level_init
    current_data = current_abspath.read_text(encoding='utf-8')
  File "F:\miniconda3\envs\sr\lib\pathlib.py", line 1134, in read_text
    with self.open(mode='r', encoding=encoding, errors=errors) as f:
  File "F:\miniconda3\envs\sr\lib\pathlib.py", line 1119, in open
    return self._accessor.open(self, mode, buffering, encoding, errors,
FileNotFoundError: [Errno 2] No such file or directory: 'F:\\git\\nordic\\sdk-nrf\\zephyr\\west.yml'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "F:\miniconda3\envs\sr\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "F:\miniconda3\envs\sr\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "F:\miniconda3\envs\sr\Scripts\west.exe\__main__.py", line 7, in <module>
  File "F:\miniconda3\envs\sr\lib\site-packages\west\app\main.py", line 1085, in main
    app.run(argv or sys.argv[1:])
  File "F:\miniconda3\envs\sr\lib\site-packages\west\app\main.py", line 244, in run
    self.run_command(argv, early_args)
  File "F:\miniconda3\envs\sr\lib\site-packages\west\app\main.py", line 503, in run_command
    self.run_builtin(args, unknown)
  File "F:\miniconda3\envs\sr\lib\site-packages\west\app\main.py", line 611, in run_builtin
    self.cmd.run(args, unknown, self.topdir,
  File "F:\miniconda3\envs\sr\lib\site-packages\west\commands.py", line 194, in run
    self.do_run(args, unknown)
  File "F:\miniconda3\envs\sr\lib\site-packages\west\app\project.py", line 944, in do_run
    self.update_all()
  File "F:\miniconda3\envs\sr\lib\site-packages\west\app\project.py", line 996, in update_all
    self.manifest = Manifest.from_file(
  File "F:\miniconda3\envs\sr\lib\site-packages\west\manifest.py", line 1263, in from_file
    return Manifest(topdir=topdir, config=config,
  File "F:\miniconda3\envs\sr\lib\site-packages\west\manifest.py", line 1468, in __init__
    self._ctx = self._top_level_init(source_data,
  File "F:\miniconda3\envs\sr\lib\site-packages\west\manifest.py", line 1876, in _top_level_init
    raise MalformedConfig(
west.configuration.MalformedConfig: file not found: manifest file F:\git\nordic\sdk-nrf\zephyr\west.yml (from configuration options manifest.path="zephyr", manifest.file="west.yml")

marc-hb, 2023-09-08:
> west init

You meant `west init -l .`, not `west init`. See `west help init`.

Thanks to #676,  the next `west version` will show an error message much shorter and a bit less confusing: `FATAL ERROR: target directory already exists`

lovettchris, 2023-09-08:
ok, thanks, yeah better errors are always good.

closed 2023-09-08 by lovettchris