zephyrproject-rtos/west #694: menuconfig aborting due to Kconfig warnings

adoyle-h, 2023-10-20:
When it has warnings, I cannot open the menuconfig to re-configure the symbols.

```
> west build -t menuconfig
-- west build: running target menuconfig
[0/1] Re-running CMake...
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /Users/adoyle/Workspace/zephyr/esp32/zephyr/samples/hello_world
-- CMake version: 3.27.6
-- Cache files will be written to: /Users/adoyle/Library/Caches/zephyr
-- Zephyr version: 3.5.0-rc3 (/Users/adoyle/Workspace/zephyr/esp32/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: esp32_devkitc_wroom
-- Found host-tools: zephyr 0.16.3 (/Users/adoyle/.local/zephyr-sdk-0.16.3)
-- Found toolchain: zephyr 0.16.3 (/Users/adoyle/.local/zephyr-sdk-0.16.3)
-- Found BOARD.dts: /Users/adoyle/Workspace/zephyr/esp32/zephyr/boards/xtensa/esp32_devkitc_wroom/esp32_devkitc_wroom.dts
-- Generated zephyr.dts: /Users/adoyle/Workspace/zephyr/esp32/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /Users/adoyle/Workspace/zephyr/esp32/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /Users/adoyle/Workspace/zephyr/esp32/zephyr/build/zephyr/dts.cmake

warning: NET_L2_BT_MGMT (defined at subsys/net/l2/Kconfig:55) has direct dependencies NET_L2_BT && NETWORKING with value n, but is currently being y-selected by the following symbols:
 - NET_L2_BT_SHELL (defined at subsys/net/l2/Kconfig:61), with value y, direct dependencies NETWORKING (value: y), and select condition
 NETWORKING (value: y)
Parsing /Users/adoyle/Workspace/zephyr/esp32/zephyr/Kconfig
Loaded configuration '/Users/adoyle/Workspace/zephyr/esp32/zephyr/build/zephyr/.config'

error: Aborting due to Kconfig warnings

CMake Error at /Users/adoyle/Workspace/zephyr/esp32/zephyr/cmake/modules/kconfig.cmake:355 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /Users/adoyle/Workspace/zephyr/esp32/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /Users/adoyle/Workspace/zephyr/esp32/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /Users/adoyle/Workspace/zephyr/esp32/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FAILED: build.ninja
/opt/homebrew/Cellar/cmake/3.27.6/bin/cmake --regenerate-during-build -S/Users/adoyle/Workspace/zephyr/esp32/zephyr/samples/hello_world -B/Users/adoyle/Workspace/zephyr/esp32/zephyr/build
ninja: error: rebuilding 'build.ninja': subcommand failed
FATAL ERROR: command exited with status 1: /opt/homebrew/bin/cmake --build /Users/adoyle/Workspace/zephyr/esp32/zephyr/build --target menuconfig
```

marc-hb, 2023-10-27:
Does it work when you call `ninja -C /Users/adoyle/Workspace/zephyr/esp32/zephyr/build  menuconfig`?

adoyle-h, 2023-11-06:
Sorry for the late reply.

>  Does it work when you call ninja -C /Users/adoyle/Workspace/zephyr/esp32/zephyr/build  menuconfig?

No work. It has same error to `west build -t menuconfig`.

> Does it work if you rename build/ to build.old/ and run west build -t menuconfig again?

Yes, it works.

closed 2023-11-06 by adoyle-h

marc-hb, 2023-11-06:
This rang a bell, so I spent 2 minutes searching and I found this 2018 duplicate which is still open:
- https://github.com/zephyrproject-rtos/zephyr/issues/9573

So this is unrelated to west but it is a real bug.