zephyrproject-rtos/west #704: `pip install` fails on MSYS2

gon1332, 2024-01-29:
Hi everyone,

I want to install west via PyPI under my MSYS2/MINGW environment on Windows.

I face this issue where the `ruamel.yaml.clib` cannot be built:

```
XXX@YYY MINGW64 ~ % pip3 install -U west
Collecting west
  Using cached west-1.2.0-py3-none-any.whl (95 kB)
Requirement already satisfied: colorama in c:/msys64/mingw64/lib/python3.10/site-packages (from west) (0.4.6)
Requirement already satisfied: PyYAML>=5.1 in c:/msys64/mingw64/lib/python3.10/site-packages (from west) (6.0)
Collecting pykwalify (from west)
  Using cached pykwalify-1.8.0-py2.py3-none-any.whl (24 kB)
Requirement already satisfied: setuptools in c:/msys64/mingw64/lib/python3.10/site-packages (from west) (57.0.0)
Requirement already satisfied: packaging in c:/msys64/mingw64/lib/python3.10/site-packages (from west) (23.0)
Collecting docopt>=0.6.2 (from pykwalify->west)
  Using cached docopt-0.6.2-py2.py3-none-any.whl
Collecting python-dateutil>=2.8.0 (from pykwalify->west)
  Using cached python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
Collecting ruamel.yaml>=0.16.0 (from pykwalify->west)
  Using cached ruamel.yaml-0.18.5-py3-none-any.whl (116 kB)
Requirement already satisfied: six>=1.5 in c:/msys64/mingw64/lib/python3.10/site-packages (from python-dateutil>=2.8.0->pykwalify->west) (1.16.0)
Collecting ruamel.yaml.clib>=0.2.7 (from ruamel.yaml>=0.16.0->pykwalify->west)
  Using cached ruamel.yaml.clib-0.2.8.tar.gz (213 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: ruamel.yaml.clib
  Building wheel for ruamel.yaml.clib (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for ruamel.yaml.clib (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [14 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.mingw_x86_64-cpython-310
      creating build/lib.mingw_x86_64-cpython-310/ruamel
      creating build/lib.mingw_x86_64-cpython-310/ruamel/yaml
      creating build/lib.mingw_x86_64-cpython-310/ruamel/yaml/clib
      copying ./setup.py -> build/lib.mingw_x86_64-cpython-310/ruamel/yaml/clib
      copying ./__init__.py -> build/lib.mingw_x86_64-cpython-310/ruamel/yaml/clib
      copying ./LICENSE -> build/lib.mingw_x86_64-cpython-310/ruamel/yaml/clib
      running build_ext
      building '_ruamel_yaml' extension
      error: --plat-name must be one of ('win32', 'win-amd64', 'win-arm32', 'win-arm64')
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for ruamel.yaml.clib
Failed to build ruamel.yaml.clib
ERROR: Could not build wheels for ruamel.yaml.clib, which is required to install pyproject.toml-based projects
```

I've tried building it from source, but it still fails with the same error.

Any ideas on this?

marc-hb, 2024-01-29:
This really looks like it has nothing to do with `west`... what happens if you `pip install ruamel.yaml`?

MINGW has always been a bit of a hack.

marc-hb, 2024-01-29:
Thanks, so this is not related to `west`, I'm closing this. Please get in touch with MSYS2 and/or ruaml.

closed 2024-04-10 by marc-hb