zephyrproject-rtos/west #705: Infinite loop when building in a moved directory

celinakalus, 2024-02-13:
When trying to incrementally build an application with `-d <build-dir>` from an existing build folder that has been moved from `./build` to `./<something-else>`, west gets stuck in an infinite loop trying to re-run CMake. In this process, the old path of the folder is printed on stdout. The log output looks like this, looped over and over again:
```
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/ckalus/zephyrproject/zephyr/samples/subsys/ipc/ipc_service/icmsg/remote
-- CMake version: 3.25.1
-- Cache files will be written to: /home/ckalus/.cache/zephyr
-- Zephyr version: 3.6.0-rc1 (/home/ckalus/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: stm32h747i_disco_m4
-- Found host-tools: zephyr 0.16.5 (/home/ckalus/.local/opt/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/home/ckalus/.local/opt/zephyr-sdk-0.16.5)
-- Found BOARD.dts: /home/ckalus/zephyrproject/zephyr/boards/arm/stm32h747i_disco/stm32h747i_disco_m4.dts
-- Found devicetree overlay: /home/ckalus/zephyrproject/zephyr/samples/subsys/ipc/ipc_service/icmsg/remote/boards/stm32h747i_disco_m4.overlay
-- Generated zephyr.dts: /home/ckalus/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/ckalus/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/ckalus/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/ckalus/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/ckalus/zephyrproject/zephyr/build/zephyr/.config'
No change to configuration in '/home/ckalus/zephyrproject/zephyr/build/zephyr/.config'
No change to Kconfig header in '/home/ckalus/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Using ccache: /usr/bin/ccache
CMake Warning at /home/ckalus/zephyrproject/zephyr/CMakeLists.txt:1969 (message):
  __ASSERT() statements are globally ENABLED

-- Configuring done
-- Generating done
-- Build files have been written to: /home/ckalus/zephyrproject/zephyr/build
[0/1] Re-running CMake...
```
To work around this, one can delete all build directories and build again cleanly.

**Expected behaviour:** The build process terminates and stdout looks like this:
```
Loading Zephyr default modules (Zephyr base).
-- Application: /home/ckalus/zephyrproject/zephyr/samples/subsys/ipc/ipc_service/icmsg/remote
-- CMake version: 3.25.1
-- Found Python3: /home/ckalus/zephyrproject/.venv/bin/python3 (found suitable version "3.11.2", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/ckalus/.cache/zephyr
-- Zephyr version: 3.6.0-rc1 (/home/ckalus/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: stm32h747i_disco_m4
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (/home/ckalus/.local/opt/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/home/ckalus/.local/opt/zephyr-sdk-0.16.5)
-- Found Dtc: /home/ckalus/.local/opt/zephyr-sdk-0.16.5/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/ckalus/zephyrproject/zephyr/boards/arm/stm32h747i_disco/stm32h747i_disco_m4.dts
-- Found devicetree overlay: /home/ckalus/zephyrproject/zephyr/samples/subsys/ipc/ipc_service/icmsg/remote/boards/stm32h747i_disco_m4.overlay
-- Generated zephyr.dts: /home/ckalus/zephyrproject/zephyr/build-icmsg-m4/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/ckalus/zephyrproject/zephyr/build-icmsg-m4/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/ckalus/zephyrproject/zephyr/build-icmsg-m4/zephyr/dts.cmake
Parsing /home/ckalus/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/ckalus/zephyrproject/zephyr/boards/arm/stm32h747i_disco/stm32h747i_disco_m4_defconfig'
Merged configuration '/home/ckalus/zephyrproject/zephyr/samples/subsys/ipc/ipc_service/icmsg/remote/prj.conf'
Merged configuration '/home/ckalus/zephyrproject/zephyr/samples/subsys/ipc/ipc_service/icmsg/remote/boards/stm32h747i_disco_m4.conf'
Configuration saved to '/home/ckalus/zephyrproject/zephyr/build-icmsg-m4/zephyr/.config'
Kconfig header saved to '/home/ckalus/zephyrproject/zephyr/build-icmsg-m4/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /home/ckalus/.local/opt/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ckalus/.local/opt/zephyr-sdk-0.16.5/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Using ccache: /usr/bin/ccache
CMake Warning at /home/ckalus/zephyrproject/zephyr/CMakeLists.txt:1969 (message):
  __ASSERT() statements are globally ENABLED

-- Configuring done
-- Generating done
-- Build files have been written to: /home/ckalus/zephyrproject/zephyr/build-icmsg-m4
-- west build: building application
[1/138] Preparing syscall dependency handling

[8/138] Generating include/generated/version.h
-- Zephyr version: 3.6.0-rc1 (/home/ckalus/zephyrproject/zephyr), build: e2388b96c493
[137/138] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       42828 B         1 MB      4.08%
             RAM:       10624 B       128 KB      8.11%
            QSPI:          0 GB       256 MB      0.00%
           SRAM1:          0 GB       128 KB      0.00%
           SRAM2:          0 GB       128 KB      0.00%
           SRAM3:          0 GB        32 KB      0.00%
           SRAM4:          0 GB        64 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from /home/ckalus/zephyrproject/zephyr/build-icmsg-m4/zephyr/zephyr.elf for board: stm32h747i_disco_m4
```

**Note:** When I encountered this error, there existed both a `./build` folder AND a `./<something-else>` folder, and the `./<something-else>` folder was just an older version of the `./build` that has been moved to this new location.

**Suggested solution:** Before building, check whether path given by `-d <build-dir>` and path saved in build files in `<build-dir>` are the same, and use one of the paths consistently.

marc-hb, 2024-02-13:
Thanks for the report!

No software should ever go into an infinite loop for any sort of input.

On the other hand, you're never supposed to rename build directories. CMake uses a million of absolute paths there.

marc-hb, 2024-02-13:
I cannot reproduce, I get the expected error:

> CMake Error: The current CMakeCache.txt directory zproj/zephyr/bd2/CMakeCache.txt is different than the directory zproj/zephyr/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place.

Please provide detailed, step by step commands to reproduce this as well as the west and zephyr versions used.

Use this template as "inspiration":
https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

celinakalus, 2024-02-14:
Well, today, as it turns out, I am unable to reproduce this as well. Not sure what was borked about my setup there. Retracting this bug report until I am able to reproduce it. Sorry for the trouble!

closed 2024-02-14 by celinakalus

marc-hb, 2024-02-14:
No worries, please just re-open this if you can reproduce.

Note even if you can reproduce I would consider this as valid but low priority because:

>  you're never supposed to rename build directories. CMake uses a million of absolute paths there.