zephyrproject-rtos/west #706: `die_if_no_git` not listed under west API

Nick-Munnich, 2024-03-28:
`die_if_no_git` is a public method of WestCommand, but is not listed under https://docs.zephyrproject.org/latest/develop/west/west-apis.html#westcommand.

marc-hb, 2024-03-29:
It's "public" because it's used in `project.py`. This does not imply it is part of the official, supported West APIs.

What is the actual problem you're experiencing?

Nick-Munnich, 2024-03-29:
I wanted to use said method while writing a west extension. The positioning of the method in the source code implies (to me at least) that it was intended to be included in the API, as it is "grouped" with 3 other methods which are all part of the API under a comment reading "other public methods". 

My suggestion would be to include another comment or move said method to avoid this ambiguity/source of confusion.