zephyrproject-rtos/west #709: fatal: fetch-pack: invalid index-pack output

kenkit, 2024-05-04:
[image]

kenkit, 2024-05-04:
I've tried changing git configs even switching networks. 
But that is the error I get, is there a way to manually clone the repo.

marc-hb, 2024-05-06:
When happens when you use git directly? Try:
```
git clone https://github.com/zephyrproject-rtos/zephyr
```

PS: avoid screenshots when possible, use copy/paste instead: https://docs.zephyrproject.org/latest/develop/getting_started/index.html#use-copy-paste

kenkit, 2024-05-07:
Or you can make it possible to only clone upto x depth so as to avoid this issue

closed 2024-05-07 by kenkit

marc-hb, 2024-05-07:
I have no idea what the issue was.

kenkit, 2024-05-13:
I guess being able to use repo not initialized by west as I seemed to be having an issue cloning.

marc-hb, 2024-05-13:
`west` is "only" a git user just like you and me. Anything `west` does with git, you can do it without `west` too. Even after `west` has (successfully!) used git, you can still use most git commands in repos cloned by `west`, `west` is not exclusive/"jealous".

But if you have issues cloning then it's not going to work for you and it's not going to work for `west`, there's no difference and no magic there. Git must work, period.

How much time did you spend looking at https://docs.zephyrproject.org/latest/develop/west/index.html so far?